zephyrproject-rtos/zephyr #83864: tests: benchmarks: thread_metric: error sending/receiving messages

golowanow, 2025-01-12:
**Describe the bug**
`benchmark.thread_metric.message` on `up_squared` always returns zero metric value and `ERROR: Invalid counter value(s). Error sending/receiving messages!`
The problem observed at least on two up_squared boards. 
Other thread_metric benchmarks usually have no problems on the same devices.

**To Reproduce**
Run the thread_metric benchmark test suite  on a host with up_squared connected, e.g.
`twister -p up_squared/apollo_lake -T tests/benchmarks/thread_metric`

**Expected behavior**
The benchmark returns its time period values without errors.

**Logs and console output**
```
**** Thread-Metric Message Processing Test **** Relative Time: 30
ERROR: Invalid counter value(s). Error sending/receiving messages!
Time Period Total:  0

**** Thread-Metric Message Processing Test **** Relative Time: 60
ERROR: Invalid counter value(s). Error sending/receiving messages!
Time Period Total:  0

**** Thread-Metric Message Processing Test **** Relative Time: 90
ERROR: Invalid counter value(s). Error sending/receiving messages!
Time Period Total:  0
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain zephyr SDK 0.17.0
 - Commit SHA f9e4bc3

peter-mitsis, 2025-01-30:
Apparently this is replicable on qemu_x86_64 as well.

peter-mitsis, 2025-01-31:
I think I have it figured out. The problem occurs on systems where 'unsigned long' is 8 bytes, not 4 bytes. This violates an assumption made by the benchmark.

PR #85003 (2025-02-05): tests: Fix thread_metric message processing
Changes the type of both the tm_message_sent and tm_message_received arrays from 'unsigned long' to 'unsigned int'. The test expects those arrays to be 16 bytes long. This was a problem on 64-bit platforms as 'unsigned long' is 8 bytes, which made the arrays 32 bytes long. On both our supported 32-bit and 64-bit platforms, 'unsigned int' works out to be 4 bytes long, thereby giving us the requisite 16 byte buffer.

Although a case could be made for using 'uint32_t', this was not chosen simply to keep the structure as close as practical to the origina [... cut]
commit 11f69fb6 (2025-02-05): tests: Fix thread_metric message processing

Changes the type of both the tm_message_sent and tm_message_received
arrays from 'unsigned long' to 'unsigned int'. The test expects those
arrays to be 16 bytes long. This was a problem on 64-bit platforms
as 'unsigned long' is 8 bytes, which made the arrays 32 bytes long.
On both our supported 32-bit and 64-bit platforms, 'unsigned int'
works out to be 4 bytes long, thereby giving us the requisite 16
byte buffer.

Although a case could be made for using 'uint32_t', this was not
chosen simply to keep the structure as close as practical to the
origin [... cut]
closed 2025-02-05 by kartben