zephyrproject-rtos/zephyr #83885: subsys: llext: tests llext.* does not works on Nordic boards

katgiadla, 2025-01-13:
**Describe the bug**
Tests:
- `llext.readonly`
- `llext.readonly_mpu`
- `llext.writable`
- `llext.writable_relocatable`
- `llext.writable_relocatable_slid_linking`
- `llext.writable_slid_linking`

fails.

On boards:
- `nrf54l15dk/nrf54l05/cpuapp`
- `nrf54l15dk/nrf54l10/cpuapp`
- `nrf54l15dk/nrf54l15/cpuapp`
- `nrf54h20dk@0.9.0/nrf54h20/cpuapp`
- `nrf54h20dk@0.9.0/nrf54h20/cpurad`

reports timeout.

Expected that, overflow is reported on `nrf9160dk@0.14.0/nrf9160` and `nrf54l15dk/nrf54l15/cpuflpr`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/llext -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**

Logs for `nrf54l15dk/nrf54l{05 | 10 | 15}/cpuapp`:
```
*** Booting Zephyr OS build v4.0.0-3190-g99a63a776980 ***
Running TESTSUITE llext
===================================================================
START - test_ext_syscall_fail
 PASS - test_ext_syscall_fail in 0.001 seconds
===================================================================
START - test_load_unload_hello_world
D: Loading ELF data...
D: Loading relocatable ELF
D: Finding ELF tables...
D: section 0 at 0x0: name 0, type 0, flags 0x0, addr 0x0, size 0, link 0, info 0
D: section 1 at 0x34: name 31, type 1, flags 0x6, addr 0x0, size 232, link 0, info 0
D: section 2 at 0x4ac: name 27, type 9, flags 0x40, addr 0x0, size 112, link 10, info 1
D: section 3 at 0x11c: name 37, type 1, flags 0x3, addr 0x0, size 0, link 0, info 0
D: section 4 at 0x11c: name 43, type 8, flags 0x3, addr 0x0, size 8, link 0, info 0
D: section 5 at 0x11c: name 48, type 1, flags 0x2, addr 0x0, size 231, link 0, info 0
D: section 6 at 0x204: name 60, type 1, flags 0x2, addr 0x0, size 8, link 0, info 0
D: section 7 at 0x51c: name 56, type 9, flags 0x40, addr 0x0, size 16, link 10, info 6
D: section 8 at 0x20c: name 74, type 1, flags 0x30, addr 0x0, size 33, link 0, info 0
D: section 9 at 0x22d: name 83, type 1879048195, flags 0x0, addr 0x0, size 54, link 0, info 0
D: section 10 at 0x264: name 1, type 2, flags 0x0, addr 0x0, size 432, link 11, info 22
D: symtab at 10
D: section 11 at 0x414: name 9, type 3, flags 0x0, addr 0x0, size 152, link 0, info 0
D: strtab at 11
D: section 12 at 0x52c: name 17, type 3, flags 0x0, addr 0x0, size 99, link 0, info 0
D: shstrtab at 12
D: Allocate and copy strings...
D: region 7: start 0x20000c00, size 128
D: region 6: start 0x20000d00, size 256
D: Mapping ELF sections...
D: section 0 name  skipped
[... 94 lines cut ...]
D: 2 20000fe0 20001100 
D: relocation 2:13 info 0x802 (type 2, sym 8) offset 228 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x20000fe4 addr 0x20001100
D: 2 20000fe4 20001100 
D: relocation section .rel.exported_sym (7) acting on section 6 has 2 relocations
D: relocation 7:0 info 0x802 (type 2, sym 8) offset 0 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x20000ca0 addr 0x20001100
D: 2 20000ca0 20001100 
D: relocation 7:1 info 0x1a02 (type 2, sym 26) offset 4 sym_name test_entry sym_type 2 sym_bind 1 sym_ndx 1
I: writing relocation symbol test_entry type 2 sym 26 at addr 0x20000ca4 addr 0x20000f71
D: 2 20000ca4 20000f71 test_entry
```

Logs for  `nrf9160dk@0.14.0/nrf9160`:
```
/opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `noinit' will not fit in region `RAM'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld.bfd: region `RAM' overflowed by 34712 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA or Version used:  `v4.0.0-2912-g8aa4a`

teburd, 2025-01-14:
Why am I being assigned a nordic specific issue? If the tests are passing in CI on qemu, this implies this is a platform specific problem to me and should be assigned to the platform maintainers

katgiadla, 2025-02-11:
Fixes #85078 on the all platforms except `nrf54h20dk@0.9.0/nrf54h20/cpuapp` for below scenarios:

- `tests/subsys/llext/llext.readonly_mpu`
- `tests/subsys/llext/llext.writable`
- `tests/subsys/llext/llext.writable_relocatable`
- `tests/subsys/llext/llext.writable_relocatable_slid_linking`
- `tests/subsys/llext/llext.writable_slid_linking`

katgiadla, 2025-03-24:
Scenario `tests/subsys/llext/llext.readonly` on `nrf54h20dk@0.9.0/nrf54h20/cpuapp` reports timeout:
```
...
I: writing relocation type 2 at 0x2f015018 with symbol test_entry (0x2f014ea9)
D: 2 2f015018 2f014ea9 test_entry
D: sym 0x2f014ea9 name test_entry
D: Loaded llext: 1052 bytes in heap, .text at 0x2f014c04, .rodata at 0xe0ac430
I: Loaded extension threads_kernel_
D: calling entry function 0x2f014ea9(0)
Give semaphore from main thread
Creating thread
Starting thread
Joining thread
Take semaphore from test thread
E: ***** BUS FAULT *****
E:   Instruction bus error
E: r0/a1:  0x00000001  r1/a2:  0x00000000  r2/a3:  0x2f012570
E: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x0e09e17b
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x2f014dc0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2f012570 (unknown)
E: Halting system
```

nordic-piks, 2025-08-21:
Still fails at H20:
```
START - test_load_unload_threads_kernel_objects
E: Undefined symbol with no entry in symbol table __device_dts_ord_200, offset 632, link section 10
E: Could not find symbol __device_dts_ord_200!
E: Failed to link, ret -61

    Assertion failed at ../../../../../../../../zephyr/tests/subsys/llext/src/test_llext.c:133: load_call_unload: (res is non-zero)
load should succeed
 FAIL - test_load_unload_threads_kernel_objects in 0.030 seconds
```
@carlescufi  @bjarki-andreasen  any input?

closed 2025-11-04 by github-actions[bot]