zephyrproject-rtos/zephyr #83886: [Kconfig] Build warning that a kconfig symbol is defined with a prompt outside the choice

ParyKI, 2025-01-13:
**Describe the bug**
Kconfig build warning interrupted the configure process.

`warning: the choice symbol NRF_802154_SOURCE_HAL_NORDIC (defined at C:/repos/zephyr/zephyrproject/zephyr/modules/hal_nordic/Kconfig:16, modules\hal_nordic\Kconfig:16) is defined with a prompt outside the choice`

`warning: the choice symbol NRF_802154_SOURCE_HAL_NORDIC (defined at C:/repos/zephyr/zephyrproject/zephyr/modules/hal_nordic/Kconfig:16, modules\hal_nordic\Kconfig:16) is defined with a prompt outside the choice`

**To Reproduce**
1. remove in **NRF_802154_SOURCE** https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/hal_nordic/Kconfig#L13, 
2. input command to build: `west build -p always -b nrf51dk/nrf51822 ./samples/drivers/adc/adc_dt -d ./build`

**Expected behavior**
should not show this warning.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Latest main branch

ParyKI, 2025-01-13:
Hello @tejlmand, sorry to I'd like to bother you, but i believe you may have the experience on this part. I'd like to consult you why choice in the module folder without name caused this warning, in some other places outside the module folder will not cause this warning?

ParyKI, 2025-01-13:
I noticed that modules/hal_nordic/Kconfig loaded twice time
[image]

nordicjm, 2025-01-29:
I don't follow what you mean to remove, if you removed the config from the choice then you can't have it there

tejlmand, 2025-02-20:
> I noticed that modules/hal_nordic/Kconfig loaded twice time

Correct.
https://github.com/zephyrproject-rtos/zephyr/blob/7e18f8060af99a5483ab83ab3eeeb151d4f16925/modules/Kconfig#L126-L130

tejlmand, 2025-02-20:
Not a bug, work as intended.

User should not remove the setting name of a Kconfig choice.

closed 2025-02-20 by tejlmand