zephyrproject-rtos/zephyr #83889: drivers: counter: counter_basic_api: drivers.counter.basic_api fails

katgiadla, 2025-01-13:
**Describe the bug**
The test `tests/drivers/counter/counter_basic_api/drivers.counter.basic_api` fails.

Observed for:
- `nrf54l15dk/nrf54l15/cpuapp`
- `nrf54l15dk/nrf54l10/cpuapp`
- `nrf54l15dk/nrf54l05/cpuapp`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -s drivers.counter.basic_api -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-3067-g910ec595a055 ***
Running TESTSUITE counter_basic
===================================================================
START - test_all_channels
Testing timer@55000
Testing timer@85000
Testing timer@ca000
Testing timer@cb000
Testing timer@cc000
Testing timer@cd000
Testing timer@ce000
 PASS - test_all_channels in 0.925 seconds
===================================================================
START - test_cancelled_alarm_does_not_expire
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:1058: test_cancelled_alarm_does_not_expire_instance: (0 not equal to cnt)
timer@85000: Expected 0 callbacks, got 1 (i:0)

 FAIL - test_cancelled_alarm_does_not_expire in 1.011 seconds
===================================================================
START - test_late_alarm
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:810: test_late_alarm_instance: (-ETIME not equal to err)
timer@85000: Unexpected error (0)
 FAIL - test_late_alarm in 0.123 seconds
===================================================================
START - test_late_alarm_error
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:861: test_late_alarm_error_instance: (-ETIME not equal to err)
timer@85000: Failed to detect late setting (err: -16)
 FAIL - test_late_alarm_error in 0.125 seconds
===================================================================
START - test_multiple_alarms
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:578: test_multiple_alarms_instance: (0 not equal to err)
timer@85000: Counter failed to set top value
 FAIL - test_multiple_alarms in 0.178 seconds
===================================================================
START - test_set_top_value_with_alarm
Testing timer@55000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:297: test_set_top_value_with_alarm_instance: (top_handler_cnt == 5U is false)
timer@55000: Unexpected number of turnarounds (41).
 FAIL - test_set_top_value_with_alarm in 0.134 seconds
===================================================================
START - test_short_relative_alarm
Testing timer@55000
Skipped for timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:188: counter_tear_down_instance: ((err == 0) || (err == -ENOTSUP) is false)
timer@85000: Setting top value to default failed
 FAIL - test_short_relative_alarm in 0.131 seconds
===================================================================
START - test_single_shot_alarm_notop
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:446: test_single_shot_alarm_instance: (0 not equal to err)
timer@85000: Counter set alarm failed (err: -16)
 FAIL - test_single_shot_alarm_notop in 0.195 seconds
===================================================================
START - test_single_shot_alarm_top
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:433: test_single_shot_alarm_instance: (0 not equal to err)
timer@85000: Counter failed to set top value
 FAIL - test_single_shot_alarm_top in 0.194 seconds
===================================================================
START - test_valid_function_without_alarm
Testing timer@55000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:744: test_valid_function_without_alarm: (ticks not between ticks_expected > ticks_tol ? ticks_expected - ticks_tol : 0 and ticks_expected + ticks_tol inclusive)
timer@55000: counter ticks not in tolerance
 FAIL - test_valid_function_without_alarm in 0.035 seconds
===================================================================
TESTSUITE counter_basic failed.
Running TESTSUITE counter_no_callback
===================================================================
START - test_set_top_value_without_alarm
Testing timer@55000
Testing timer@85000

    Assertion failed at zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:339: test_set_top_value_without_alarm_instance: (0 not equal to err)
timer@85000: Counter failed to set top value (err: -16)
 FAIL - test_set_top_value_without_alarm in 0.136 seconds
===================================================================
TESTSUITE counter_no_callback failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  10.00% [counter_basic]: pass = 1, fail = 9, skip = 0, total = 10 duration = 3.051 seconds
 - PASS - [counter_basic.test_all_channels] duration = 0.925 seconds
 - FAIL - [counter_basic.test_cancelled_alarm_does_not_expire] duration = 1.011 seconds
 - FAIL - [counter_basic.test_late_alarm] duration = 0.123 seconds
 - FAIL - [counter_basic.test_late_alarm_error] duration = 0.125 seconds
 - FAIL - [counter_basic.test_multiple_alarms] duration = 0.178 seconds
 - FAIL - [counter_basic.test_set_top_value_with_alarm] duration = 0.134 seconds
 - FAIL - [counter_basic.test_short_relative_alarm] duration = 0.131 seconds
 - FAIL - [counter_basic.test_single_shot_alarm_notop] duration = 0.195 seconds
 - FAIL - [counter_basic.test_single_shot_alarm_top] duration = 0.194 seconds
 - FAIL - [counter_basic.test_valid_function_without_alarm] duration = 0.035 seconds

SUITE FAIL -   0.00% [counter_no_callback]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.136 seconds
 - FAIL - [counter_no_callback.test_set_top_value_without_alarm] duration = 0.136 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA or Version used:  `v4.0.0-3067-g910ec5`

aescolar, 2025-01-15:
Raising priority,
The issue can be reproduced in main in simulation and fails in PRs CI blocking them
`twister -p nrf54l15bsim/nrf54l15/cpuapp -T tests/drivers/counter/counter_basic_api `

nordic-krch, 2025-01-16:
Fixed by #82901

PR #82901 (2025-01-17): drivers: counter: nrfx_timer: Get TIMER max frequency from clock source
Some time ago `max-frequency` property was added to `nordic,nrf-timer` which indicated what is the clock frequency that runs the TIMER (it also indicates maximum frequency of the timer). Recently we started to add clock source to peripherals and frequency of the clock source shall be the source of information about the peripheral frequency.

PR includes:
- Revert  0ecfac663d4cb45 and 413ca65d65f9b4aceb13b4609d0141d9ee60e2f2
- Include work from #82871 which adds macro for getting frequency from the clock source
- Add missin [... cut]
closed 2025-01-17 by kartben