zephyrproject-rtos/zephyr #83895: kernel: tickless: tickless_concept: kernel.tickless.concept fails

katgiadla, 2025-01-13:
**Describe the bug**
The test `tests/kernel/tickless/tickless_concept/kernel.tickless.concept` fails.

Observed for:
- `nrf52dk/nrf52832`
- `nrf52840dk/nrf52840`
- `nrf5340dk/nrf5340/cpuapp`
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk@0.14.0/nrf9160`
- `nrf9160dk@0.14.0/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -s kernel.tickless.concept -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-3190-g99a63a776980 ***
Running TESTSUITE workqueue_api
===================================================================
START - test_k_work_queue_stop
*** Booting Zephyr OS build v4.0.0-3190-g99a63a776980 ***
Running TESTSUITE tickless_concept
===================================================================
START - test_tickless_slice
elapsed slice 120, expected: <100, 110>

    Assertion failed at zephyr/tests/kernel/tickless/tickless_concept/src/main.c:50: thread_tslice: t <= SLICE_SIZE_LIMIT is false

elapsed slice 10, expected: <100, 110>

    Assertion failed at zephyr/tests/kernel/tickless/tickless_concept/src/main.c:48: thread_tslice: t >= SLICE_SIZE is false

elapsed slice 20, expected: <100, 110>

    Assertion failed at zephyr/tests/kernel/tickless/tickless_concept/src/main.c:48: thread_tslice: t >= SLICE_SIZE is false

elapsed slice 20, expected: <100, 110>

    Assertion failed at zephyr/tests/kernel/tickless/tickless_concept/src/main.c:48: thread_tslice: t >= SLICE_SIZE is false

 FAIL - test_tickless_slice in 0.179 seconds
===================================================================
START - test_tickless_sysclock
time 690, 910
time 920, 1130
 PASS - test_tickless_sysclock in 0.450 seconds
===================================================================
TESTSUITE tickless_concept failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [tickless_concept]: pass = 1, fail = 1, skip = 0, total = 2 duration = 0.629 seconds
 - FAIL - [tickless_concept.test_tickless_slice] duration = 0.179 seconds
 - PASS - [tickless_concept.test_tickless_sysclock] duration = 0.450 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 6df81f15dd23a03a7816f173be941372
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA or Version used:  `v4.0.0-3067-g910ec5`

GoranWall, 2025-03-25:
Is it possible that this issue has been fixed indirectly while working on something in the vicinity? We have had `kernel.tickless.concept` on our internal quarantine list for some time due to this issue but we see it has been passing in out --quarantine-verify runs since at least March 7. We only test on nrf52840dk/nrf52840 though.

nordic-krch, 2025-03-25:
It fails due to dd5f11cb04b834584a6d851fe55a9673e25a5356 but i don't know yet why.

closed 2025-03-28 by kartben