zephyrproject-rtos/zephyr #83899: Add native support for STM32F479 chip group.

saphieron, 2025-01-13:
**Is your feature request related to a problem? Please describe.**
The Zephyr project is currently supporting several chips from the STM32F4 family, but not the STM32F479. Creating e.g. a simple blink project and defining a custom board using as soc "stm32f479xx" results in an unknown soc error, as expected.

**Describe the solution you'd like**
I'd love to have native support in Zephyr for that chip and its variants. From my understanding it requires defining or extending the soc definitions for that chip based on the F469. A (the only?) major difference seems to be in the F479 containing hardware support for cryptography.

**Describe alternatives you've considered**
None, beyond manually writing a soc definition by hand myself.

**Additional context**
I'd love to help with that process if possible, but as I am only starting to work with Zephyr my understanding of it is still rather limited.

cichiwskyj, 2025-01-14:
Hello, working on that same work project as well, had a look [this set of changed files](https://github.com/zephyrproject-rtos/zephyr/pull/83193/commits/1eebfe499d753e05f11284b330b79e056884a06e) and on dts/arm/st/h7/stm32h757.dtsi I'm unclear on what the line `interrupts = <79 0>;` refers to.
Tried to map that to the RM0399 Reference manual of the STM32H757, but so far did not find anything relating to those values.
@JarmouniA, can you give a pointer what line that refers to?

FRASTM, 2025-01-29:
Note that the stm32F479 reference manual is [RM0386](https://www.st.com/resource/en/reference_manual/rm0386-stm32f469xx-and-stm32f479xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)

cichiwskyj, 2025-04-04:
With https://github.com/zephyrproject-rtos/zephyr/pull/87969 being merged I think we can close this issue.

closed 2025-04-04 by saphieron