zephyrproject-rtos/zephyr #83904: Bluetooth: Mesh: Assert in bt_mesh_adv_unref when sending 2 messages in a row to a proxy client

PavelVPV, 2025-01-13:
**Describe the bug**
After [Bluetooth: Mesh: Introduce separate workq for ADV EXT by LingaoM · Pull Request #78914 · zephyrproject-rtos/zephyr](https://github.com/zephyrproject-rtos/zephyr/pull/78914)  the mesh device asserts in `bt_mesh_adv_unref` when sending 2 messages in a row to a proxy client: [ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/mesh/adv.c](https://devzone.nordicsemi.com/f/nordic-q-a/117593/assertion-fail-west_topdir-zephyr-subsys-bluetooth-mesh-adv-c) 

There are 2 issues here. First, `struct bt_mesh_adv::node` variable is shared among 2 FIFOs: one in `adv.c` and the other one in `proxy_msg.c`. Second, is the node type should be void * : [FIFOs — Zephyr Project Documentation](https://docs.zephyrproject.org/latest/kernel/services/data_passing/fifos.html#writing-to-a-fifo), [zephyr/include/zephyr/kernel.h at 0728f5d0e27ed1d1274f1c82b2bdc0ffd31d2aee · zephyrproject-rtos/zephyr](https://github.com/zephyrproject-rtos/zephyr/blob/0728f5d0e27ed1d1274f1c82b2bdc0ffd31d2aee/include/zephyr/kernel.h#L2557-L2558) but not `sys_snode_t` (or at least `sys_sfnode_t`).

The temporary fix is to define a separate variable for the proxy FIFO.

**To Reproduce**
1. Provision a mesh device using nRF Mesh app for smartphone
2. Send any 2 messages in a row
3. Observe the assertion

**Expected behavior**
No assertion happens and all messages are sent to the proxy client.

**Impact**
Proxy feature is broken in Zephyr 4.0.0+

LingaoM, 2025-01-14:
> The temporary fix is to define a separate variable for the proxy FIFO.

``` C
bool bt_mesh_proxy_relay(struct bt_mesh_adv *adv, uint16_t dst)
{
	bool relayed = false;
	int i;

	LOG_DBG("%u bytes to dst 0x%04x", adv->b.len, dst);

	for (i = 0; i < ARRAY_SIZE(clients); i++) {
		struct bt_mesh_proxy_client *client = &clients[i];
```

For ARRAY_SIZE(clients) > 1, also have problem.

LingaoM, 2025-01-14:
> The temporary fix is to define a separate variable for the proxy FIFO.

``` C
bool bt_mesh_proxy_relay(struct bt_mesh_adv *adv, uint16_t dst)
{
	bool relayed = false;
	int i;

	LOG_DBG("%u bytes to dst 0x%04x", adv->b.len, dst);

	for (i = 0; i < ARRAY_SIZE(clients); i++) {
		struct bt_mesh_proxy_client *client = &clients[i];

		if (!client->cli) {
			continue;
		}

		if (!client_filter_match(client, dst)) {
			continue;
		}

		if (bt_mesh_proxy_relay_send(client->cli->conn, adv)) {
			continue;
		}

		relayed = true;
	}

	return relayed;
}
```

For ARRAY_SIZE(clients) > 1, also have problem.

PR #84122 (2025-02-13): [Backport v4.0-branch] Bluetooth: Mesh: Fix Assert in bt_mesh_adv_unref when messages to a proxy
Backport 6371080406eb0b621f7327be77d880d9d707800b from #83934.

Fixes #83904.
closed 2025-01-17 by kartben