zephyrproject-rtos/zephyr #83905: Add Support to Pull in NXP IW416 Bluetooth Controller Firmware Binary Blob

nirav-agrawal, 2025-01-13:
## Origin

The following binary blob header is being added: iw416/uart_iw416_se.h

## Type

- [ ] Precompiled library
- [X] Firmware image

## Module

hal_nxp: https://github.com/zephyrproject-rtos/hal_nxp

## Purpose

Precompiled firmware header supporting Bluetooth Dual-Mode LL feature on NXP IW416 SoC.

## Pull Request

WIP

## Dependencies

None

## License

https://github.com/zephyrproject-rtos/hal_nxp/blob/master/zephyr/blobs/license/LA_OPT_NXP_Software_License.txt

nirav-agrawal, 2025-01-16:
Hi @dleach02, I am working on enabling this support, once ready I will update the PR in this request. 
Thank you, 

Regards,
Nirav

dleach02, 2025-01-21:
When you are ready we can reapply the TSC label

mmahadevan108, 2025-03-28:
This PR for this request are below:

HAL: https://github.com/zephyrproject-rtos/hal_nxp/pull/512
Zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/85639

dleach02, 2025-04-24:
TSC approval on 4/23/2025

closed 2025-06-03 by nashif