zephyrproject-rtos/zephyr #83920: LoRaWAN Example Works Only Once Without Additional Configuration

felf-zhaw, 2025-01-13:
https://github.com/zephyrproject-rtos/zephyr/blob/0728f5d0e27ed1d1274f1c82b2bdc0ffd31d2aee/samples/subsys/lorawan/class_a/README.rst?plain=1#L1-L27

## Description:
The current LoRaWAN example only works the first time it is run. On subsequent attempts, the device fails to connect to the LoRaWAN network.

## Root Cause:
The issue lies with the join_cfg.otaa.dev_nonce, which is hardcoded to 0 in the example. According to the LoRaWAN specification, the dev_nonce must change for every new connection attempt. When this does not happen, the join procedure fails.

## Recommendation:
To avoid confusion for users, please update the README to include the following clarification:
1. The example works only once with the current configuration.
2. For the example to work on multiple runs, users should:
   - Enable CONFIG_LORAWAN_NVM_SETTINGS in the Kconfig to allow proper storage and reuse of configuration settings.
   - OR set join_cfg.otaa.dev_nonce to a random value programmatically for each new connection attempt.
This clarification will help users better understand the limitations of the current example and guide them toward resolving the issue.

Thank you for addressing this!

kartben, 2025-01-14:
@felf-zhaw I agree this would be a great and useful clarification in the README. Are you able to submit a pull request? :)

felf-zhaw, 2025-01-14:
#83948 :)

closed 2025-01-21 by kartben