zephyrproject-rtos/zephyr #83935: kconfig: modules kconfig processed two times.

ParyKI, 2025-01-14:
**Describe the bug**
If the **kconfig-ext** is set to true in modules.yml, then add choice option without name in the module glue code kconfig file, the kconfig warning appeared. 

**To Reproduce**
Take hal_nordic as a example, if i add a sample choice option in https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/hal_nordic/Kconfig, then build any board that zephyr supported, the kconfig warning shows.

**Expected behavior**
Choice without name is normal, should not has this warning,

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain:Zephyr SDK
 - latest main branch
 - 
**Additional context**
I noticed, the kconfig will be included two times, if the  **kconfig-ext** is set to true in modules.yml. seems this two times included are caused by:https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/Kconfig#L21 and https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/Kconfig#L130. I don't know why osource in **if 0** block take effect.

if i set **kconfig-ext** to false in modules.yml, the kconfig options in the glue kocnfig can not be processed correctly, if i remove https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/Kconfig#L130,  the kconfig options in the glue kocnfig can be processed correctly. 

I am going to add an external project to zephyr, I want to put the glue code in zephyr repo, and the project has a lot of choice options without name, I got warnings when doing build. I want to get community help, thanks!

rruuaanng, 2025-01-14:
> I am going to add an external project to zephyr

Can I see the external project you mentioned?

ParyKI, 2025-01-14:
> > I am going to add an external project to zephyr
> 
> Can I see the external project you mentioned?

actually it is a vendor HAL, it is in prototype stage, this HAL use kconfig to establish dependencies between components, and choice without name is heavily used in this project.

kartben, 2025-01-14:
triaging as low priority given this only results in warnings

tejlmand, 2025-02-20:
> I don't know why osource in if 0 block take effect.

because `if` in Kconfig is not a safeguard in the normal C style but simply means `apply if` to all settings inside.
Kconfig always sources all files.

For example:
```
if FOO

config BAR
    bool "Bar"
    default BAZ

endif
```
is equivalent to:
```
config BAR
    bool "Bar" if FOO
    default BAZ if FOO
```

so `if` in Kconfig only saves you typing, but everything is still sourced.
See also: https://www.kernel.org/doc/html/latest/kbuild/kconfig-language.html

As for the reason of `if 0`, then the comment right abover describes the reason:
https://github.com/zephyrproject-rtos/zephyr/blob/7e18f8060af99a5483ab83ab3eeeb151d4f16925/modules/Kconfig#L126-L128

Remember, a downstream manifest might decide to **not** import a given project (Zephyr module), but in order for Kconfig to still check all dependencies then the symbols must still be defined (but not available for usage) even when the module is missing.

tejlmand, 2025-02-20:
> I am going to add an external project to zephyr, I want to put the glue code in zephyr repo, and the project has a lot of choice options without name, I got warnings when doing build. I want to get community help, thanks!

Thanks, but if you want to have the glue code in the Zephyr repo, then all choices must be given a setting name.
If this is not desired, then your only options is to place the Kconfig glue code in the external project, and then use:
```
build:
    kconfig: zephyr/Kconfig
```

Like it's done in Espressif hal and Picolibc.
There the Kconfig lives in the module repo:
https://github.com/zephyrproject-rtos/hal_espressif/blob/36640428fdedcffed0574f06b6ec65c390862caf/zephyr/module.yml#L4
https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/zephyr/Kconfig

https://github.com/zephyrproject-rtos/picolibc/blob/db4d0fe5952d5ecd714781e3212d4086d970735a/zephyr/module.yml#L4
https://github.com/zephyrproject-rtos/picolibc/blob/main/zephyr/Kconfig

closed 2025-02-20 by tejlmand