zephyrproject-rtos/zephyr #83938: Cellular: LwM2M Client error   <err> net_lwm2m_engine: Cannot connect UDP (-116)

Tang-junyao, 2025-01-14:
Hi,

I followed the instruction (https://docs.nordicsemi.com/bundle/ncs-2.5.0/page/nrf/samples/cellular/lwm2m_client/sample_description.html) to build and run on my DK, but got these errors.
**[00:00:00.255,218] <err> spi_nor: Device id 00 00 00 does not match config c2 28 17
[00:18:40.046,051] <err> net_lwm2m_engine: Cannot connect UDP (-116)
[00:18:40.047,027] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-116)
[00:27:00.138,244] <err> app_lwm2m_client: LwM2M engine reported a network error.
[00:29:05.239,624] <wrn> at_monitor: No heap space for incoming notification: +CSCON: 0
[00:29:05.275,054] <err> modem_info_params: Link data not obtained: 5 -22
[00:29:05.275,054] <err> lwm2m_connmon: Unable to obtain modem parameters: -22**
[image]
[image]

Do you have any idea how to solve them?

board: nrf9160dk
SDK: v2.5.0
lwm2m server: Leshan demo server , Leshan bootstrap server
DTLS
nb-iot (China Mobile Hong Kong)
no sensor, no GNSS
config files: prj.conf , overlay-leshan-bootstrap.conf , overlay-nbiot.conf , overlay-lwm2m-1.1.conf , overlay-dtls-cid.conf , overlay-aggressive-psm.conf , overlay-lowpower.conf 

Thanks,
Liza

Tang-junyao, 2025-01-14:
[image]

kartben, 2025-01-14:
This is a Nordic specific code sample, please use Nordic forums for initially requesting assistance

closed 2025-01-14 by kartben