zephyrproject-rtos/zephyr #83949: esp32s3: System freeze when WiFi + BT are active simulteaneously

Yanok35, 2025-01-14:
**Describe the bug**

Using the bluetooth peripheral sample including WiFi configuration in `prj.conf`, it is observed the system freeze after few minutes.

This has been reproduced on 4.0.0 tag release and also on "main" branch (v4.0.0-3140-g950dcde66aa7).

**Environment**

 -  Modify `platform/zephyr/samples/bluetooth/peripheral/prj.conf` to enable WiFi in bluetooth peripheral app:

    ```
    CONFIG_WIFI=y
    CONFIG_NETWORKING=y
    CONFIG_NET_L2_ETHERNET=y
    CONFIG_NET_IPV6=n
    CONFIG_NET_IPV4=y
    CONFIG_NET_DHCPV4=y
    CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y
    CONFIG_NET_LOG=y
    CONFIG_NET_SHELL=y
    CONFIG_NET_L2_WIFI_SHELL=y
    # ---
    CONFIG_NET_ZPERF=y
    CONFIG_NET_SOCKETS_POLL_MAX=10
    # Other switch to try with (EXTRAM)
    CONFIG_ESP_SPIRAM=y
    CONFIG_ESP_BT_HEAP_RUNTIME=y
    CONFIG_ESP_WIFI_HEAP_RUNTIME=y
    ```

    Enable WiFi node in `boards/espressif/esp32s3_devkitc/esp32s3_devkitc_procpu.dts`:

    ```
    &wifi {
           status = "okay";
    };
    ```

 -  Build & Flash the sample app on the DevkitC:

    ```
    cd platform/zephyr/samples/bluetooth/peripheral/
    west build -b esp32s3_devkitc/esp32s3/procpu
    west flash
    ```

**To Reproduce**

Steps to reproduce the behavior:

1. Start the board and connect it to an access point:
        `wifi connect -s <ssid> -p <psk> -k 1`

2. On a Linux machine, start a WiFi communcation (`ping` flood test):
        ` $ sudo ping -f <esp32_ipv4>`

3. On a Linux machine, start a Bluetooth client (Python client sample in attachement):
        ` $ ./gatt-client.py <esp32_bluetooth_macaddr>`

**Issue**

After few minutes, the boards freeze and Bluetooth & WiFi links drops down.

**Bluetooth Client**

A sample Bluetooth client application is proposed at this location : [gatt-client.py](https://gist.github.com/Yanok35/95f0b781aa686a4b7c028f8892895379)

Yanok35, 2025-01-14:
Hi @sylvioalves. Once both clients are launch, you can leave as is. Issue has been reproduced whereas using ICMP (ping flood) or TCP Stream ( IPerf test session).

sylvioalves, 2025-01-15:
@Yanok35, I am still working on this, will reply back soon.

marekmatej, 2025-01-16:
Hi @Yanok35, usually the main stack is set to 2kB. The setup you are describing seems like a heavy load on the system. Can you try increasing the stack by adding this `CONFIG_MAIN_STACK_SIZE=4096` to your `prj.conf`?

Yanok35, 2025-01-17:
Hi @sylvioalves , thanks for the update.

Hi @marekmatej, the can be reproduced with bigger stack.

Here are the whole changes in `prj.conf` currently on the bench:

```
diff --git a/samples/bluetooth/peripheral/prj.conf b/samples/bluetooth/peripheral/prj.conf
index f788fa670aa..d5a95c86e40 100644
--- a/samples/bluetooth/peripheral/prj.conf
+++ b/samples/bluetooth/peripheral/prj.conf
@@ -13,7 +13,7 @@ CONFIG_BT_HRS=y
 CONFIG_BT_IAS=y
 CONFIG_BT_CTS=y
 CONFIG_BT_CTS_HELPER_API=y
-CONFIG_BT_PRIVACY=y
+#CONFIG_BT_PRIVACY=y
 CONFIG_BT_DEVICE_NAME="Zephyr Peripheral Sample Long Name"
 CONFIG_BT_DEVICE_APPEARANCE=833
 CONFIG_BT_DEVICE_NAME_DYNAMIC=y
@@ -24,4 +24,32 @@ CONFIG_BT_SETTINGS=y
 CONFIG_FLASH=y
 CONFIG_FLASH_MAP=y
 CONFIG_NVS=y
-CONFIG_SETTINGS=y
+#CONFIG_SETTINGS=y
+# ---
+CONFIG_WIFI=y
+
+CONFIG_NETWORKING=y
+CONFIG_NET_L2_ETHERNET=y
+
+CONFIG_NET_IPV6=n
+CONFIG_NET_IPV4=y
+CONFIG_NET_DHCPV4=y
+CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y
+
+CONFIG_NET_LOG=y
+
+CONFIG_NET_SHELL=y
+CONFIG_NET_L2_WIFI_SHELL=y
+# ---
+CONFIG_NET_ZPERF=y
+CONFIG_NET_TCP=y
+CONFIG_NET_UDP=y
+CONFIG_NET_SOCKETS_POLL_MAX=10
+# Other switch to try with (EXTRAM)
+CONFIG_ESP_SPIRAM=y
+CONFIG_ESP_BT_HEAP_RUNTIME=y
+CONFIG_ESP_WIFI_HEAP_RUNTIME=y
+
+CONFIG_MAIN_STACK_SIZE=4096
+CONFIG_SHELL_STACK_SIZE=6000
```

horakdavid, 2025-01-17:
We are facing similar issue as this one in our application where we use both wifi and bt. We are using wifi and bt only for scanning. Wifi is set in promiscuous mode and bt scanning is enabled. After few minutes system will freeze and watchdog timer resets the system. We found out that the system is stuck in double exception handler. We tried increasing stack sizes but it didn't help.

For serial communication with device we use USB Serial/JTAG peripheral, when using standard UART peripheral instead, it seems that the system got stuck less often.

sylvioalves, 2025-05-07:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/88982