zephyrproject-rtos/zephyr #83964: Add a section/FAQ about Cyber Resilience Act

kartben, 2025-01-14:
**Is your enhancement proposal related to a problem? Please describe.**

People using Zephyr to build products might have questions/concerns regarding compliance with the Cyber Resilience Act (CRA). They may need guidance on how Zephyr's features and processes align with the CRA requirements, and what tools (ex. SBOM generation) are available to them.

**Describe the solution you'd like**

Add some blurb (in the form of an FAQ?) to https://docs.zephyrproject.org/latest/security/standards/index.html

**Describe alternatives you've considered**

We don't do anything, but people may then assume project "doesn't care" about CRA (and maybe we don't btw, I don't know, that's likely for the Governing Board, Security WG, and TSC to decide)

**Additional context**

N/A

cc @kestewart @ceolin

d3zd3z, 2025-01-27:
Some thoughts on what to cover:
- Are there changes to Zephyr needed to comply.
- What things, Zephyr affects, needs to be covered.
- What configuration of Zephyr are needed to comply.

lindblandro, 2025-02-04:
This would be highly useful! Currently the information is pretty scattered and you need to know what to search for. A FAQ, checklist or something similar would be a great starting point.

lindblandro, 2025-02-11:
> If we'd provide some utility script with Zephyr RTOS version + [Kconfig](https://docs.zephyrproject.org/latest/build/kconfig/index.html) configuration as input and CVEs + EPSS scores as output this would be gold standard I guess.

This sounds a lot like Yocto's `cve-check` tooling, which is great and easy to use. This could maybe be integrated with SBOM generation? Getting a little sidetracked from the main issue, but great ideas nonetheless!

schuam, 2025-08-07:
> @lindblandro We can open a dedicated issue and continue idea collection there.

@fkromer Do you know, has someone opened such an issue?

I would have some questions/ideas for the SBOMs that can be generated using `west`. It seems as if some fields in those SBOMs are missing (version and supplier information in app.spdx for example). This might be intentional, as the required fields are not defined yet.

Point (24) of [Article 13 of the CRA](https://eur-lex.europa.eu/legal-content/EN/TXT/HTML/?uri=OJ:L_202402847#art_13) basically states that the commission may specify the format and elements of the SBOM. To my knowledge, there are no harmonized normes or delegated acts yet that do specify the format and elements. But NTIA has issued [The Minimum Elements For a Software Bill of Materials (SBOM)](https://www.ntia.gov/report/2021/minimum-elements-software-bill-materials-sbom) and also the German BSI has published a [techincal guideline](https://www.bsi.bund.de/DE/Themen/Unternehmen-und-Organisationen/Standards-und-Zertifizierung/Technische-Richtlinien/TR-nach-Thema-sortiert/tr03183/TR-03183_node.html) about cyber resilienz requirements. Part 2 of that guideline is about SBOMs. Both documents require version and supplier information. I would assume that that information will also be required once the commission has specified the format and elements of the SBOM.

So it would be interesting to know what the plans are: Wait until everything is specified? Or include such information pro-actively, because it is very likely that it will be required.

schuam, 2025-08-11:
> Shouldn’t SPDX and CycloneDX already support them already?

I haven't really looked in CycloneDX, but SPDX does support them.

> Version and supplier information: Sounds reasonable to consider. If formats already support it format spec wise… why not considering it already?

That's basically my point. I used the tool [ntia-conformance-checker](https://github.com/spdx/ntia-conformance-checker) to analyze the SPDX files that `west spdx (...)` generates. Only the generated SBOM `zephyr.spdx` contains version information. Non of the genreated SBOMs contains supplier information.

I looked into the Phython scripts under `scripts/west_commands/zspdx/` in the Zephyr repo. In `walker.py` one can find functions that setup the different documents (e.g. `setupBuildDocument(...)` and `setupZephyrDocument(...)`, ...). This is were the version information is added to the Zephyr document, but not to the others. I'm wondering why? Was that an intentional decision that I just don't understand right now? Or should that be added?

I can't find anything about the supplier information. For that I'm also not sure where that information should be added into a project. But I think it should be added somewhere and subsequently be added to the generated SBOMs.

kartben, 2025-08-11:
> I haven't really looked in CycloneDX, but SPDX does support them.
>
>
> That's basically my point. I used the tool [ntia-conformance-checker](https://github.com/spdx/ntia-conformance-checker) to analyze [...]

Can you please open an issue for the fields you'd like to see added, with some examples too? Thanks!

schuam, 2025-08-17:
I just found [this pull request](https://github.com/zephyrproject-rtos/zephyr/pull/91013) that might actually take care of adding version and supplier data to the generated SBOMs. I didn't see that before. I will wait for it to be merged and then check if anything else is needed.