zephyrproject-rtos/zephyr #83971: kscan_input_disable_callback() does not disable processing of key-press events

quic-bkulkarn, 2025-01-14:
**Describe the bug**
- Zephyr kscan_input_disable_callback() does not disable the processing of key-press events handled via KSCAN-INPUT driver. 
- Though, the flag "enabled" gets updated as part of the above callback, it is not consumed to disable processing of any new key-press using the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Invoke kscan_input_disable_callback() from Zephyr application supporting keyboard via KSCAN DEVICE
2. Press any key after kscan_input_disable_callback() is executed
3. Key-press still gets processed

**Expected behavior**

No key-press should be processed after kscan_input_disable_callback() is executed.

**Impact**

Impacts disabling of keyboard if required from the Zephyr application.

**Logs and console output**
None

**Environment (please complete the following information):**

 - OS: Windows with Zephyr v3.6

**Additional context**
None

kartben, 2025-01-14:
Please make sure you have a look at the Input subsystem if you haven't already for kscan is now pretty much deprecated

kartben, 2025-03-16:
Support for deprecated KSCAN has been dropped. Have you been successful switching to the input subsystem, @quic-bkulkarn?

closed 2025-03-16 by kartben