zephyrproject-rtos/zephyr #83982: RFC: Introduce Zephyr Driver Support for Peripheral Sensor Interface (PSI5) Protocol

congnguyenhuu, 2025-01-15:
## Introduction
This RFC proposes the introduction of a Zephyr driver to support the PSI5 protocol, enhancing the capabilities of the Zephyr RTOS for automotive sensor applications (https://en.wikipedia.org/wiki/Peripheral_Sensor_Interface_5).

Technical document reference: https://www.psi5.org/fileadmin/user_upload/psi5/Specification/V1.3/psi5_specification_v13_080729.pdf

### Problem description
The PSI5 protocol is widely used in automotive applications for sensor communication. Adding support for PSI5 in Zephyr will enable developers to integrate automotive sensors more efficiently, leveraging Zephyr’s real-time capabilities.

### Proposed change
Introduce new Zephyr driver support for PSI5 protocol: PR #79824.

## Detailed RFC

### Proposed change (Detailed)
1. Driver Architecture:
• The driver will be implemented as a new driver Zephyr.
• The driver will include support for PSI5 communication, including initialization, data transmission, and error handling.
• The driver will support both PSI5 transmitters and receivers.

2. Configuration:
• The driver will be configurable via Kconfig options.
• Device Tree bindings will be provided to specify hardware configurations.
3. APIs Design:
- The driver will expose APIs for initializing the PSI5 interface, sending and receiving
	- start_sync(): Start the sync pulse generator on a specific channel
	- stop_sync(): Stop the sync pulse generator on a specific channel
	- send(): Transmitting PSI5 data on a specific channel
	- add_rx_callback(): Add a callback function to handle messages received for a specific channel
- Callbacks will handle PSI5 events.
	- psi5_tx_callback_t(): Callback handler function signature for sending
	- psi5_rx_callback_t(): Callback handler function signature for receiving
4. Testing:
• sample `samples/boards/nxp/s32/psi5` will validate the driver’s functionality and show how to use the driver.

### Dependencies
Hardware: New components are required to interface with PSI5 sensors

### Concerns and Unresolved Questions

## Alternatives
Adding support for PSI5 in Zephyr will significantly benefit automotive applications, providing a robust and real-time solution for sensor communication.

congnguyenhuu, 2025-02-04:
Hello @henrikbrixandersen, I saw that my RFC is still listed under "Todo Architecture Review" status, no actions. Could you please help me expedite the review process? Thanks.

henrikbrixandersen, 2025-02-25:
Architecture WG 2025-02-18:
- This topic was deferred until 2025-03-04

carlescufi, 2025-03-04:
Architecture WG

- @Dat-NguyenDuy presents slides regarding the standard and the API itself