zephyrproject-rtos/zephyr #83983: RFC: Introduce Zephyr Driver Support for Single Edge Nibble Transmission (SENT) Protocol

congnguyenhuu, 2025-01-15:
## Introduction
This RFC proposes the introduction of a Zephyr driver to support the SENT protocol, enhancing the capabilities of the Zephyr RTOS for automotive sensor applications (https://en.wikipedia.org/wiki/SENT_(protocol)).

Technical document reference: https://www.nxp.com/docs/en/application-note/AN4219.pdf

### Problem description
The SENT protocol is widely used in automotive applications for sensor communication. Adding support for SENT in Zephyr will enable developers to integrate automotive sensors more efficiently, leveraging Zephyr’s real-time capabilities.

### Proposed change
Introduce new Zephyr driver support for SENT protocol: PR #80117.

## Detailed RFC

### Proposed change (Detailed)
1. Driver Architecture:
• The driver will be implemented as a new driver Zephyr.
• The driver will provide APIs for SENT initialization, data transmission, and error handling.
• The driver will support SENT receivers.
2. Configuration:
• The driver will be configurable via Kconfig options.
• Device Tree bindings will be provided to specify hardware configurations.
3. APIs Design:
- The driver will expose APIs for receiving SENT frames.
	- start_rx(): Enable a specific channel to start receiving from the bus
	- stop_rx(): Disable a specific channel to stop receiving from the bus
	- add_rx_callback(): Add a callback function to handle messages received for a specific channel
- Callbacks will handle SENT events 
	- sent_rx_callback_t(): Callback handler function signature for receiving
4. Testing:
• sample `samples/boards/nxp/s32/sent` will validate the driver’s functionality and show how to use the driver.

### Dependencies
Hardware: New components are required to interface with SENT sensors

### Concerns and Unresolved Questions

## Alternatives
Adding support for SENT in Zephyr will significantly benefit automotive applications, providing a robust and real-time solution for sensor communication.

congnguyenhuu, 2025-02-04:
Hello @henrikbrixandersen, I saw that my RFC is still listed under "Todo Architecture Review" status, no actions. Could you please help me expedite the review process? Thanks.

henrikbrixandersen, 2025-02-07:
> Hello @henrikbrixandersen, I saw that my RFC is still listed under "Todo Architecture Review" status, no actions. Could you please help me expedite the review process? Thanks.

That's the correct status until this has been presented at the Architecture Working Group.

henrikbrixandersen, 2025-02-25:
Architecture WG 2025-02-18:
- This topic was deferred until 2025-03-04

manuargue, 2025-02-26:
> static int sensor_sample_fetch(const struct device *dev,
        enum sensor_channel chan)
{
        sent_add_rx_callback(sent_dev, SENT_CHANNEL, rx_queue_cb, &frame_queue);
        /* To get queue and parse data frame then store internal sample buffer */
}
static int sensor_channel_get(const struct device *dev,
        enum sensor_channel chan,
        struct sensor_value *val)
{
        sent_add_rx_callback(sent_dev, SENT_CHANNEL, rx_queue_cb, &frame_queue);
        /* To get queue and parse data frame then store in the user pointer val */
}

Please look at https://docs.zephyrproject.org/latest/hardware/peripherals/sensor/index.html#implementing-fetch-and-get. 
fetch() -> should block to read sensor data and stash it
get() -> should return already stashed sensor data

> void rx_triger_cb(const struct device *dev, uint8_t channel_id, struct sent_frame *frame,
        enum sent_status status, void *user_data)
{
        if (status == SENT_RX_FAST_FRAME) {
                data->handler_drdy(dev, data->trig_drdy)
        }
}

how would be slow frames handled? If I understand correctly one complete frame is composed of 18 consecutive error free slow frames.

carlescufi, 2025-03-04:
Architecture WG

- @Dat-NguyenDuy shows slides with an introduction to the standard and the API proposed
- @decsny asks who allocates the memory required for RX, @Dat-NguyenDuy shows how it works via the callback, which is called once per received message
- @decsny suggest to look at ways of improving the throughput, perhaps by allowing the application to provide the buffers itself

decsny, 2025-03-04:
some nits:
- in zephyr , adding callback functions are usually called like "xyz_**register**_callback" 
- don't know if it is needed to namespace by the spec, it seems like that spec is trying to standardize de facto usage of SENT, maybe you can provide a config struct paramater to specify a spec if there is a different one in the future, and any other kind of config parameters that might crop up, it would be a bit more future proof  I think

congnguyenhuu, 2025-04-08:
> * [@decsny](https://github.com/decsny) suggest to look at ways of improving the throughput, perhaps by allowing the application to provide the buffers itself

This is a minimal introduction to SENT. We aim for the minimal code to be merged first. Looking ahead, we plan to enhance throughput. For example, the start_rx API could be updated to include a parameter that specifies the number of receiving buffers required before triggering a receiving callback.

> * in zephyr , adding callback functions are usually called like "xyz_**register**_callback"

I updated to rename api to `saej2716_register_callback`

> * don't know if it is needed to namespace by the spec, it seems like that spec is trying to standardize de facto usage of SENT, maybe you can provide a config struct paramater to specify a spec if [...]

The namespace 'saej2716' does not represent the specification, it refers to the name of the protocol.

Regarding the data frame structure, it is designed to allow future extensions. It includes the field 'saej2716_frame_type' to support additional frame types. For instance, to incorporate support for SENT-B in the future (frames with up to 56 bits of data and a 7-bit CRC), we can introduce an enum for this frame type and expand the data field by adding a struct element to store high-data values.

decsny, 2025-04-08:
> This is a minimal introduction to SENT. We aim for the minimal code to be merged first. Looking ahead, we plan to enhance throughput. For example, the start_rx API could be updated to include a parameter that specifies the number of receiving buffers required before triggering a receiving callback.

I'm not sure why do you think the goal here is to merge something as fast as possible. You are introducing a public API which will be used by all of  zephyr going forward. If we think there are better ways to do it then I don't know why we would not.  It sounds like you are implying that the architecture you're introducing in this PR will be deprecated later, and yet still wanting to merge it, as if it's finalized already before being approved or merged, and deal with any debt and redo it later. That doesn't make any sense. If you think this solution is not very strong and are just trying to get something done fast for your own requirement, upstream is not the place for this type of work.

dleach02, 2025-06-12:
@carlescufi by removing the arch review label does this mean the arch group approved inclusion of this?

manuargue, 2025-06-23:
> [@carlescufi](https://github.com/carlescufi) by removing the arch review label does this mean the arch group approved inclusion of this?

hi @carlescufi , @henrikbrixandersen . Is there any further discussion needed from arch WG pov? or can we close the `Architecture Review` item? Thanks

manuargue, 2025-06-27:
implementation merged in https://github.com/zephyrproject-rtos/zephyr/pull/80117

closed 2025-06-27 by manuargue