zephyrproject-rtos/zephyr #84005: GCC warning old-style-declaration on bluetooth host/att.c

tpennors, 2025-01-15:
**Describe the bug**
warning old-style-declaration on bluetooth host/att.c:137
`/zephyr/subsys/bluetooth/host/att.c:137:1: error: 'static' is not at beginning of declaration [-Werror=old-style-declaration]`

**To Reproduce**
Use arm-zephyr-eabi-gcc (Zephyr SDK 0.17.0) 12.2.0

**Expected behavior**
No warnings

**Impact**
I have to activate for my project : zephyr_compile_options("-Wno-old-style-declaration")

kartben, 2025-01-15:
@tpennors thanks for reporting this! could you please send a PR to fix this?

tpennors, 2025-01-18:
Ok I will make an attempt of PR.

Bug reproduced with the following simple modifications

`west build -b b_u585i_iot02a /zephyr/samples/bluetooth/peripheral_gatt_write -p`
with following additions in CMakeLists.txt to see the warning: 
```
zephyr_compile_options("-Wextra")
zephyr_compile_options("-Wno-unused-parameter")
zephyr_compile_options("-Wno-sign-compare")
```

tpennors, 2025-01-18:
Please check #84184

closed 2025-01-22 by kartben