zephyrproject-rtos/zephyr #84025: doc: Documentation issue in 'develop/application/index'

LysandrosAn, 2025-01-15:
**Describe the bug**

The text _Since you are using an existing Zephyr workspace, you can use west build or any other west commands to build, flash, and debug_ can be ambiguous. A custom application can be used with west commands, even if the application folder is placed outside the Zephyr workspace, right? As a matter of fact, west commands can be used irrespective of the topology, is that correct?

**Mac OS X**

* Page: `develop/application/index`
* Version: main
* SHA-1: 1db9b009a70e92cb75ef2e403f06a7a5bff022ef

kartben, 2025-01-21:
@LysandrosAn do you want to take a stab at suggesting an alternative wording? Thanks!

nashif, 2025-06-19:
7a426deae3d fixed this

closed 2025-06-19 by kartben