zephyrproject-rtos/zephyr #84034: gen_kobject_list.py is extremely slow

arbrauns, 2025-01-15:
**Describe the bug**
With `CONFIG_USERSPACE` enabled, `gen_kobject_list.py` can take an obscenely long time to run, approximately 30s in our current (decently large) project.

**To Reproduce**
1. Set `CONFIG_USERSPACE=y` in a large project
2. Build with `ninja -v`
3. Watch the jobs for `kobject_prebuilt_hash.gperf` and `kobject_hash.gperf` take a very long time

This can even be observed in the barebones `samples/userspace/hello_world_user`, where each of those takes 0.6s (though they do run in parallel).

**Expected behavior**
Generating these tables should not take up a significant part of the total build time.

**Impact**
For affected projects (large, with userspace enabled), this makes the already slow link times much, much worse. Touching a single file makes a subsequent build take 60 seconds, approximately 55 of which are spent waiting on the two invocations of `gen_kobject_list.py`.

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA or Version used: v3.7.0

kartben, 2025-01-21:
@arbrauns is this new or something you were already observing in previous versions of Zephyr and you would then describe as a regression?

ubieda, 2025-01-21:
Dropping link to related long-standing known issue: #8559

arbrauns, 2025-01-27:
> is this something you were not observing in previous versions of Zephyr and you would then describe as a regression?

No, I don't think this is particularly new, it's more that as the project grew larger it eventually got to a point where I got sick of just accepting and ignoring it.

arbrauns, 2025-04-14:
The script hasn't been touched since last year, I highly doubt it magically got faster because this issue has been sitting here.

arbrauns, 2025-07-29:
Shockingly, still no improvement from the script sitting there unmodified for a couple months longer. Maybe if we threaten to close the issue a couple more times it'll get scared into being faster? Let's try that.

arbrauns, 2025-09-29:
Hmm, no luck so far.

teburd, 2025-09-29:
@arbrauns I've been looking at replacing quite a few bits of python with rust actually, because its one of the longest parts of running twister across large swaths of boards. So for very similar reasons actually. From my side I've been doing some 🧑‍🔬 to see how difficult the task of parsing the c preprocessor, kconfig, and dts would be in a single tool. I get the kobject gen task is a bit different, but... if all these various little build parse/code_gen utilities could be bundled into a program and easily distributed perhaps there's a chance it could supplant these very slow python scripts one day.

arbrauns, 2025-09-30:
Wow, those are ambitious goals! Replacing `gen_kobject_list.py` was significantly easier because it's a standalone tool that doesn't require parsing and interpreting C code directly.

arbrauns, 2025-12-01:
I do at least want to see the issue's first anniversary before it's closed under the assumption that it probably went away on its own.