zephyrproject-rtos/zephyr #84041: I2C gyro tests aren't skipped

yishai1999, 2025-01-15:
**Describe the bug**

When running the tests/drivers/i2c/i2c_api application the two gyro tests fails instead of being skipped when there is no gyro node present.

**To Reproduce**

Add a `i2c-0` alias to the board's DTS if it doesn't have one already. Then build and flash the test:
```
west build -p auto -b xg23_rb4210a zephyr/tests/drivers/i2c/i2c_api
west flash
```
This can be done with any other board that doesn't have a gyro connected too.

**Expected behavior**

Tests should be skipped.
I also can see that in `zephyr/tests/drivers/i2c/i2c_api/testcase.yaml` there is the following line:
`dt_alias_exists("gy271")`
This lines seems like it should cause the test to be filtered on my board but it doesn't seem to do the job for some reason.

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v4.0.0-3331-g7f01e6014056 ***                                             
Running TESTSUITE i2c_gy271          
===================================================================                                   
START - test_i2c_burst_gy271                       
I2C get_config failed                              

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/i2c/i2c_api/src/test_i2c.c:199: i2c_gy271_test_i2c_burst_gy271: test_burst_gy271() == TC_PASS is false

 FAIL - test_i2c_burst_gy271 in 0.017 seconds      
===================================================================                                   
START - test_i2c_gy271                             
I2C get_config failed                              

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/i2c/i2c_api/src/test_i2c.c:194: i2c_gy271_test_i2c_gy271: test_gy271() == TC_PASS is false

 FAIL - test_i2c_gy271 in 0.016 seconds            
===================================================================                                   
TESTSUITE i2c_gy271 failed.                        

------ TESTSUITE SUMMARY START ------              

SUITE FAIL -   0.00% [i2c_gy271]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.033 seconds    
 - FAIL - [i2c_gy271.test_i2c_burst_gy271] duration = 0.017 seconds                                   
 - FAIL - [i2c_gy271.test_i2c_gy271] duration = 0.016 seconds                                         

------ TESTSUITE SUMMARY END ------                

===================================================================                                   
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK

**Additional context**

teburd, 2025-01-15:
testcase.yml is used by twister, west build has no visibility of testcase.yml

teburd, 2025-01-16:
Closing as I think this was a misunderstanding of what west build vs west twister (or running twister directly) can see. Please re-open if I'm wrong.

closed 2025-01-16 by teburd