zephyrproject-rtos/zephyr #84049: [RFC] Enhancements to Zephyr Power Management

ceolin, 2025-01-15:
# Proposal for Enhancements to Zephyr Power Management

## Introduction

Zephyr’s current power management framework provides mechanisms for managing device power states. However, we propose a refinement to focus exclusively on device runtime power management for better control, simplicity, and efficiency. Additionally, we will simplify power management build options and provide clear guidelines for device behavior during power state transitions.

## Proposed Changes

### Device Power Management via Runtime API Only
* Eliminate calls to `pm_device_action_run` for managing device power states.
* Power transitions will be exclusively managed through the runtime power management API.

### Application Responsibility for Device Power Management
* Applications must call `pm_device_runtime_get `to acquire a reference for a device before using it.
* Applications must release the device using `pm_device_runtime_put` when done.

### Subsystem Responsibility for Device Power Management
* Subsystems must adopt similar responsibilities as applications:
   * Acquire references to devices they use via` pm_device_runtime_get`.
   * Release references when usage ends via `pm_device_runtime_put`.

### Devices Initialized in Powered-Down State
* All devices will initialize in a powered-down state.
* Devices will transition to active states only when explicitly needed by an application or subsystem.

### Simplify Build Options

Consolidate build options to only two power management modes:

* **System Power Management**: Manages system-wide power transitions (e.g., suspend, resume).
   *  Zephyr’s System Power Management focuses on managing the **system power states** (e.g., sleep, deep sleep, suspend, or power off).
  * System state changes are primarily motivated by two factors:
    * The system attempts to enter the **lowest possible power state** based on the next scheduled event. (IDLE thread)
     * **Constraints**: Devices and applications can define wake-up latency requirements, which restrict deeper low-power states, and power state constraints.
 * **Device Power Management**: Manages individual device power states via runtime APIs.
    * When device power management is **disabled**, devices will be initialized and kept active by default.

### Device Behavior for Suspend and Turn Off

* Devices must assume the following:
   * **Suspend**: Devices retain their context (e.g., internal states, configuration, and data). Devices can quickly resume operation without reinitialization.
   * **Turn Off**: Devices lose their context, including all states and configuration. This is a consequence of the suspension of the power domain the device belongs to. The device is completely powered down and requires reinitialization when powered on.
* Application does not have control whether a device is suspended or powered off. The application only can inform if a device is needed. Turn off action is only triggered by the power domain the device belongs to.
* States X actions:

[image]

* **Power Domains/Rails Mapping:**
  * A direct consequence of this behavior distinction is the need to map and control additional power domains and rails, such as those for internal SoC devices. This ensures precise control over which parts of the system are powered during suspend or turn-off states.
  * Updates to the power management framework will include support for mapping power domains to devices, allowing finer control and enabling the platform to power down unused regions of the SoC efficiently.

### Benefits
* **Improved Efficiency**: Devices powered only when needed reduce overall power consumption.
* **Simplified Build Configuration**: Unified power management options make configuration easier for developers.
* **Consistent Device Behavior**: Clearly defined rules for suspend and turn off actions ensure device drivers behave predictably.
* **Clear Responsibilities**: Applications and subsystems explicitly manage power usage, aligning with modular design principles.
* **Scalability**: The proposed changes lay the groundwork for more advanced power management features in the future.

### Conclusion
This proposal aims to streamline and improve Zephyr’s power management framework by consolidating device power management under the runtime API, simplifying build options, and defining clear behaviors for suspend and turn-off actions. These changes will enhance power efficiency, simplify device power control, and reduce the complexity of power management configuration.

### Scope of required changes

* Implement pm device hook in all device drivers
* Update all subsystems to use pm runtime
* Samples don’t need to be updated
* Stop using pm actions directly
* SoCs must map internal devices to appropriated power domains

ceolin, 2025-01-15:
@bjarki-andreasen @erwango @fabiobaltieri @mmahadevan108 @kartben @JordanYates @henrikbrixandersen @nashif at all :)

de-nordic, 2025-01-16:
If the hook is pm_device_action_cb_t , would it be also possible to take it out of `struct pm_device_base` and put it directly to `struct device`? It then allows to do virtual devices that have no state without adding the state object. Such devices, when called to change state, would just pass the call to the parent device they are handing off.

As an example of virtual device:  Nordic has Flash over IPC device that is completely software construct - here actually PM is not needed as such, because the device is always on and takes no power, but we could also do a device like Flash Partition device or Flash to Uart device - and these would would need to pass state to a real device they actually operate on.

JordanYates, 2025-01-16:
Proposal looks good, probably because it matches how I have thought about PM for some time.
An additional action would be a pass over all drivers that removes internal PM state checks.
I have one pending for sensors, but I suspect other systems may have the same pattern: #83017

bjarki-andreasen, 2025-01-16:
> I'll need to dig into power domains, but proposal looks a priori complete and indeed simplified.
> 
> Though, IIUC, one missing bit is the S2RAM use case applied to devices suspension. In this use case, application (and global f/w) context is retained while hardware context is lost. In that case, devices should be able to reinitialize HW context based on stored configuration. Is that correct or maybe I am missing something ?

In case of S2RAM, for devices which loose their context in this state, these devices will receive PM_ACTION_OFF, which will tell the driver the hardware lost all context :) This differs from suspend in which the "power domain" is still on, so the registers retain their value :)

fabiobaltieri, 2025-01-16:
cc @gmarull

JordanYates, 2025-01-16:
> In case of S2RAM, devices will receive `PM_DEVICE_ACTION_OFF`

This is not acceptable. For drivers to be clear and reliable, actions must correspond to exactly one logical event to handle.
Preparing the hardware for loss of power (which may involve GPIO reconfiguration etc), will not involve the same sequence of events as losing the local software context (`device->data`).

ceolin, 2025-01-16:
> > In case of S2RAM, devices will receive `PM_DEVICE_ACTION_OFF`, which will tell the driver the hardware lost all context :)
> 
> This is not acceptable. For drivers to be clear and reliable, actions must correspond to exactly one logical event to handle. Preparing the hardware for loss of power (which may involve GPIO reconfiguration etc), will not involve the same sequence of events as losing the local software context (`device->data`).
> 
> You can tell the statement is not quite right because the external hardware has not lost any context at all.

I agree, sw context != hw power.

bjarki-andreasen, 2025-01-17:
> > In case of S2RAM, devices will receive `PM_DEVICE_ACTION_OFF`, which will tell the driver the hardware lost all context :)
> 

> This is not acceptable. For drivers to be clear and reliable, actions must correspond to exactly one logical event to handle.
> Preparing the hardware for loss of power (which may involve GPIO [...]

I may have misunderstood your comment, I'm referring to hardware registers loosing their register values, which they do as a result of S2RAM or any other power mode where the suspended hardware is not powered. `dev->data`, stored in RAM, which remains powered in S2RAM, is not lost. The driver just needs to know that the hardware must be reinitialized from whatever is stored in `dev->data` when it is powered again, as a result of leaving S2RAM.

These are the clear and reliable actions, starting with the driver resumed:
* Driver receives suspend action
* Driver suspends hardware

At this time, no context is lost, hardware registers nor RAM

* Driver receives power off action (this is before power is actually removed from the hardware)
* Driver stores configurations (somewhere safe, which survives S2RAM, which could just be its dev->data which is stored in RAM) for reinit ones power comes back
* Driver configures GPIOs, releases clocks, releases power domains etc.
* Hardware looses power (as result of power domain going down or S2RAM or other lower power mode being entered)

Leaving S2RAM or simply having a power domain turned on again:

* Driver receives ON action
* Driver re-initializes hardware
* Driver receives resume
* Driver resumes hardware

The only thing the driver needs to know is:
* Did my hardware loose its context/register values
* Is my hardware supposed to be resumed or not

Presuming that no power mode supported by PM results in RAM being reset (that would be power off, which requires a reset of the entire chip anyway), sw context is never lost, its only the hardware which looses context

ceolin, 2025-01-18:
> I may have misunderstood your comment, I'm referring to hardware registers loosing their register values, which they do as a result of S2RAM or any other power mode where the suspended hardware is [...]

That is indeed the workflow that we have discussed, the challenge is that we will need to have devices mapped under power domains to effectively receive turn off action.

fabiobaltieri, 2025-01-29:
> All devices will initialize in a powered-down state.

This as it works now and in my experience is a source of confusion for the application, the situation is: an application uses pm runtime, a driver gets enhanced to support power management, now on the application the driver used to initialize automatically and now it has to be initialized manually with a single `pm_device_runtime_get` to restore the previous behavior. Had that reported to me few times as I added pm runtime support on input drivers, also has the side effect of doing a suspend-resume call a init for no good reason.

Do you see any changes in that behavior? The problem is on driver that always enables PM runtime by calling `pm_device_runtime_enable` on themselves, I'm thinking that maybe that API should be eliminated and pm-runtime should only be enabled with `zephyr,pm-device-runtime-auto`, then drivers that use the PM runtime internally (which to be perfectly honest I don't think should be allowed) can enable that prop in the soc dtsi, applications can enable it on their own dts file only for the devices that actually needs it.

> Device Power Management via Runtime API Only

This means that `PM_RUNTIME` would go? I'm specifically looking into the case where PM tries to suspend all devices when the soc goes idle.

(sorry for the delay, was ooo, working on my backlog)

ceolin, 2025-01-29:
> > All devices will initialize in a powered-down state.
> 
> This as it works now and in my experience is a source of confusion for the application, the situation is: an application uses pm runtime, a driver gets enhanced to support power management, now on the application the driver used to initialize automatically and now it has to be initialized manually with a single `pm_device_runtime_get` to restore the previous behavior. Had that reported to me few times as I added pm runtime support on input drivers, also has the side effect of doing a suspend-resume call an init for no good reason.

I am not sure if I understand, what we want to do is that devices are initialized until they are needed. We want to avoid this power on / suspend initial cycle.

> Do you see any changes in that behavior? The problem is on driver that always enables PM runtime by calling `pm_device_runtime_enable` on themselves, I'm thinking that maybe that API should be [...]

The idea is that drivers won't be doing it. Who is using the driver will have to call `pm_device_runtime_get` before calling any driver api. It will probably be done directly by an application or a subsystem.

> 
> > Device Power Management via Runtime API Only
> 
> This means that `PM_RUNTIME` would go? I'm specifically looking into the case where PM tries to suspend all devices when the soc goes idle.

Yes :) to get there though we will need to map devices to power domains / rails so internal devices are properly handled when the SoC goes idle.

> 
> (sorry for the delay, was ooo, working on my backlog)

No worries at all !

fabiobaltieri, 2025-01-29:
> I am not sure if I understand, what we want to do is that devices are initialized until they are needed. We want to avoid this power on / suspend initial cycle.

Ok that's cool but so what happens when a driver gets reworked to support power management in the new system? Sounds like it'd go from always starting initialized to always starting in suspend. That's the situation we have now if a driver gets pm runtime code AND enables itself. It's fine but then every change to implement PM on a driver becomes a downstream breaking change as now the device goes from auto initializing to always need to be manually initialized.

I think we should consider a situation where a device driver an be reworked to implement device PM, but the default behavior for the application is still to start initialized until the device gets told to actually start in standby, similarly to how the `zephyr,pm-device-runtime-auto` works right now. Does it make sense?

Or we can not do that and accept that every PM driver enablement is going to be a downstream breaking change but that sucks for the user, even on a PM enable application there's some device that you never want to suspend.

JordanYates, 2025-01-29:
> also has the side effect of doing a suspend-resume call a init for no good reason.

> Or we can not do that and accept that every PM driver enablement is going to be a downstream breaking change but that sucks for the user, even on a PM enable application there's some device that you never want to suspend.

All of these concerns should be resolved by using `pm_device_driver_init` in the driver. If `PM_DEVICE_RUNTIME` isn't enabled, the driver starts in `ACTIVE`. If `zephyr,pm-device-runtime-auto` isn't on the node, the driver starts in `ACTIVE`. If `zephyr,pm-device-runtime-auto` does exist, it starts in either `SUSPENDED` or `OFF`, depending on the power domain state.

bjarki-andreasen, 2025-01-30:
Regarding whether devices use runtime or not, they will, all of them, pm_device_action_run() will be removed, along with the runtime-auto dt flag, given it will be implicit. A new flag can be added, which makes zephyr call get() once on init on a device making it active always, but that is an outlier.

pm_device_runtime_enable() is also removed, drivers only call pm driver init, and react to pm actions. If drivers are on buses, they get() them.

So, all devices use pm device runtime, even devices which are "always on", they just use an unbalanced get to remain powered.

fabiobaltieri, 2025-01-30:
> pm_device_runtime_enable() is also removed, drivers only call pm driver init, and react to pm actions. If drivers are on buses, they get() them.

Spot on, thanks.

JordanYates, 2025-01-30:
> The idea is that drivers won't be doing it. Who is using the driver will have to call pm_device_runtime_get before calling any driver api. It will probably be done directly by an application or a subsystem.

This does go against where we ended up with some APIs, e.g. flash, where the general consensus as I understood it was that calling `pm_device_runtime_get` inside the `flash_read` implementation for example is useful for easing friction for users, since enabling PM on the device would therefore not break the various users of the API which may not be power-aware, while retaining the ability for applications to optimize transitions by requesting `ACTIVE` for a longer duration manually.

To be scalable these sorts of cases need to be defined at the API level though, not in the drivers. One option to achieve that could be for the logic to live in the API wrapper. e.g.

```
static inline int z_impl_flash_read(const struct device *dev, off_t offset,
				    void *data,
				    size_t len)
{
	const struct flash_driver_api *api =
		(const struct flash_driver_api *)dev->api;
    	int rc;

    	rc = pm_device_runtime_get(dev);
    	    	if (rc < 0) {
    	    	return rc;
    	}
    	rc = api->read(dev, offset, data, len);
    	(void)pm_device_runtime_put(dev);
    	return rc;
}
```
The would make it very obvious which functions are automatically handling PM at least.

fabiobaltieri, 2025-01-30:
> This does go against where we ended up with some APIs, e.g. flash

Right that's one of the issues I originally raised, the fact that some drivers use the pm runtime apis internally to control their own state is a bit of a user nightmare, it results in a situation where the application needs to use the runtime pm for some devices but not for others, if it uses it on others (such as the flash driver that uses pm get/put on itself) the whole story falls apart. Has other side effect such as https://github.com/zephyrproject-rtos/zephyr/pull/79829.

de-nordic, 2025-01-30:
> 
>     * Avoid unnecessary resume/suspend cycles, especially relevant for sensors where this cycle is looong.
> 

Additional issue is here that you may wake device to do nothing. Of course it is mostly an  error case, but consider external flash read on spi device, where size to read is 0 (should not be an error, but anyway). In such case you will wake up device to do nothing with the device, because no read command will be issued on spi lines, except from DPD commands to wake it and put it asleep.

Note that driver itself is just a software construct and it should not wake device until it is ready to do something to it, except cases where asked by user to keep device awake.  Users may want to keep the "final" device awake, like SPI flash, to avoid delays and additional commands on bus (for example doing DPD enter/exit each time read is performed).

bjarki-andreasen, 2025-05-12:
Noting down a missing case:

Devices may be required as wakeup sources, which means they need a new state, indicating that they are in a state capable of waking the system, so not off, and not "really" suspended. In this state, the application is not allowed to actually use it, so calling uart_transmit would not be allowed in this state, and, the system is allowed to enter a state where the device is suspended, but "active" enough to react to some event and wake the system

decsny, 2025-05-28:
Am I correct in understanding the implication here seems to be that CONFIG_PM_DEVICE is *required* to be enabled in order to enter system low power modes? Due to the fact that there isn't a way to suspend the devices and if a device is active then it's domain must be active which will likely disallow some of the lowest power states due to constraints?

Also, I find the wording of saying "simplify the build options to two modes, system power management and device power management" to be confusing. From the context it seems like these two options are not mutually exclusive, so I'm not sure why they are being referred to as modes, which would imply that they are. Can you clarify this, and if one is dependent on the other or not? Also, what does CONFIG_PM do now?

bjarki-andreasen, 2025-05-28:
> Am I correct in understanding the implication here seems to be that CONFIG_PM_DEVICE is _required_ to be enabled in order to enter system low power modes? Due to the fact that there isn't a way to [...]

Modes may be a bit confusing, but "domains" was taken, so... :D

`CONFIG_PM` is **System Power Management** according to this RFC. This covers sleep states, each of which can be constrained by wakeup latencies and devices being active. These are your S2RAMs, S2IDLEs etc. **System Power Management**, contrary to today with `CONFIG_DEVICE_SYSTEM_MANAGED`, can not force devices to do anything. The lowest sleep state which allows active devices to remain active, and meets wakeup latencies defined by the `pm_policy_latency_` APIs, will be entered in a "race to the bottom" fashion.

`CONFIG_PM_DEVICE_RUNTIME` is **Device Power Management** according to this RFC. If a device is needed by the system, its requested with `pm_device_runtime_get()`, once no longer needed, `pm_device_runtime_put()` is called. Once a device is resumed, its power domain will be resumed, and sleep states which would disable it are constrained.

So, these two "modes" are definitely not mutually exclusive :) They can be enabled independently, though in pretty much all cases, both will be enabled, as not doing device power management means sleep states affecting devices will never be able to be entered if all devices are forced ACTIVE all the time :)

decsny, 2025-05-28:
To be clear about what specifically I am asking: If CONFIG_PM is enabled, but not CONFIG_PM_DEVICE, are device constraints (from DT) of all the active devices used, i.e., the system would never be able to enter a low power mode which disables any of the devices that are active (which seems to be all of them)?

bjarki-andreasen, 2025-05-28:
> To be clear about what specifically I am asking: If CONFIG_PM is enabled, but not CONFIG_PM_DEVICE, are device constraints (from DT) of all the active devices used, i.e., the system would never be [...]

Yes. It could be argued that `CONFIG_PM` brings little value if `CONFIG_PM_DEVICE` is not enabled for most SoCs

decsny, 2025-05-29:
I recently helped someone understand configuration for the new proposal by drawing for them this table with this example device:

--------------------
possible SOC power states (example): mode_0 (active), mode_1 (idle), mode_2 (standby)

example device DT:
```
dev@x {
    reg = <x>;
    compat = "some_dev";
    zephyr,disabling-states = <&mode_2>;
};
```
| `CONFIG_PM` | `CONFIG_PM_DEVICE` | Initial Device State | Lowest possible SOC mode | 
|---|---|---|---|
| n | n | Active <sup>*</sup> | mode_0 |
| y | n | Active <sup>*</sup> | mode_1 | 
| n | y | Suspend | mode_0 |
| y | y | Suspend | mode_2<sup>+</sup> |

<sup>*</sup> will never be suspended if CONFIG_PM_DEVICE=n
<sup>+</sup> can only enter mode_2 if dev is suspended (device constraint)

--------------------

My question first of all is, is the explanation accurate?

Second, if it is, can I propose that we combine the two configs into one choice config? Since the implications of this are kind of confusing otherwise to people I think. My suggestion for the choices are like this for a hypothetical `CONFIG_PM_MODE` or similar:

Old configs -> new choice
* CONFIG_PM=n, CONFIG_PM_DEVICE=n --> "disabled" (because power management is disabled, obviously)
* CONFIG_PM=y, CONFIG_PM_DEVICE=n --> "basic" (documented as the mode where no domains ever lose power but system can idle?)
* CONFIG_PM=n, CONFIG_PM_DEVICE=y --> "domains" (the domains can be powered off but the system never idles, niche)
* CONFIG_PM=y, CONFIG_PM_DEVICE=y --> "full" (this is the mode where the system design potentially gets the most complex)

Thoughts?

JordanYates, 2025-05-29:
> CONFIG_PM=n, CONFIG_PM_DEVICE=y --> "domains" (the domains can be powered off but the system never idles, niche)

This is not niche, this is the only PM configuration that makes sense for nRF SoCs, since their SoC power modes are automatically handled. Users only need to be concerned with whether individual devices are needed or not.

decsny, 2025-05-30:
> This is not niche, this is the only PM configuration that makes sense for nRF SoCs, since their SoC power modes are automatically handled. Users only need to be concerned with whether individual [...]

I removed the word niche then from my comment, but, AFAIK the system power modes are supposed to always be transparent to the user, they only transition as a result of being in the idle thread and device constraints being met. If nRF platform handles it automatically in hardware (I'm assuming you mean, no system power driver needed?), then I guess "domains" and "full" would be the same software?

bjarki-andreasen, 2025-05-30:
@decsny The table is accurate, I added it to the description, changing `CONFIG_PM` and `CONFIG_PM_DEVICE` to be **System PM** and **Device PM** match wording in description :)

But, there is one missing component, the acceptable latency :) mode_2 could be blocked by the application requesting a lower wakeup latency than what mode_2 can achieve, so CONFIG_PM=n could simplify an application, saving some RAM/ROM and execution time, if it has very high latency requirements (always lowest latency), while still getting power savings from suspending individual devices :)

carlescufi, 2025-06-02:
@ceolin @bjarki-andreasen do we need a discussion in the Arch WG meeting about this?

bjarki-andreasen, 2025-06-02:
> @ceolin @bjarki-andreasen do we need a discussion in the Arch WG meeting about this?

If time permits :) The issue has about a years worth of comments so anything and everything has probably been discussed in some form already :)

nashif, 2025-06-04:
> [@ceolin](https://github.com/ceolin) should we consider restarting the Power Management weekly meetings?

agree, we need a few of those to nail down the details. There lots here which could be considered an overhaul and we need everyone on board and a clear plan.

ceolin, 2025-06-05:
> > [@ceolin](https://github.com/ceolin) should we consider restarting the Power Management weekly meetings?
> 
> agree, we need a few of those to nail down the details. There lots here which could be considered an overhaul and we need everyone on board and a clear plan.

Let's do this. We can start bi-weekly and see if we need to make it more frequently.

bjarki-andreasen, 2025-06-23:
> I'd like to understand the consequences of this RFC better when it comes to Silicon Labs devices. As an example, I'd like to use the USART peripheral with the Serial driver (`uart_silabs_usart`).
> 
> The USART peripheral is only active down to `runtime-idle` (EM1). In `suspend-to-idle`/`standby` (EM2), the USART is clock gated and its power domain (PD1) is shut down. This means that these modes should be prevented if there is an active transfer ongoing. However: all register content from the peripheral is retained in hardware, so when the power domain is powered again, there is no need to reinitialize the peripheral. This means that modeling this as the PD1 power domain being suspended/resumed, causing the USART to get a TURN_ON/TURN_OFF callback doesn't seem to fully apply, since the hardware does the repainting autonomously. We can't do nothing in these callbacks either, since they're also used during initial init at boot.
> 
> Another aspect is what sleep modes are available. Today, the serial driver uses `pm_policy_state_lock_get/put` to prevent entering deep sleep modes while transfers are ongoing (it's not blocked in hardware). This is similar but not identical to self-`pm_device_runtime_get/put`. With this, the driver is fully functional in deep sleep modes, even with just System PM (`CONFIG_PM`).
> 
> What would the expected configuration be when this RFC is implemented? From my reading of the proposal, I see two options:
> 
> * Do _not_ set `zephyr,disabling-states` in DTS for the EM2 state, and continue using PM policy to block sleep when transfers are active in the driver. As I understand it, this would retain the current behavior, but it would not achieve the goal of centralizing all device dependency reference counting at the Device PM level.
> * _Do_ set `zephyr,disabling-states` in DTS and remove the PM policy calls from the driver, making it the user's responsibility to do `pm_device_runtime_get/put`, and live with the wasted CPU cycles spent in on/off callbacks. This would have the effect of breaking the ability to go to the EM2 deep sleep state with System PM only, since Device PM with manual reference counting would now become required to ever unblock EM2.

> However: all register content from the peripheral is retained in hardware, so when the power domain is powered again, there is no need to reinitialize the peripheral

In this case PD1 is not the power domain of the peripheral, at least as power domains are defined by zephyr. You will only need to prevent the EM2 (and deeper) system states in this case, and only need to care about the ON state for initialization (as it sounds like you are already doing) :)

> What would the expected configuration be when this RFC is implemented? From my reading of the proposal, I see two options:

Set zephyr,disabling-states in DTS for EM2 (and deeper states). PM_DEVICE_RUNTIME will automatically block EM2 using pm_policy constraints while device is resumed with `pm_device_runtime_get()`, regardless of whether the call to `pm_device_runtime_get()` is done internally in the driver where appropriate, or by the user. PM policy calls should be removed from the driver since PM_DEVICE_RUNTIME will call them based on `zephyr,disabling-states`.

With this RFC, if CONFIG_PM_DEVICE_RUNTIME=n,  CONFIG_PM will not be very helpful, since it will be "forever blocked" by the enabled device which requires the highest EM mode, that is by design.

bjarki-andreasen, 2025-06-23:
> > regardless of whether the call to pm_device_runtime_get() is done internally in the driver where appropriate, or by the user
> 
> If we can do self-get/put inside the driver, I agree that we can just switch from PM Policy to PM Device get/put, and everything will work the same. I had the impression that the intent was for self-get/put to become disallowed at some point. I see that there is a recent discussion in the subtask on this, though. My opinion is that it would be beneficial to continue allowing it, it acts as a fallback to at least get to low power modes if the user fails to update their app to always call get/put every time they touch a `struct device`.

The wish with this RFC would be to disallow it at some point, since I at least believe the inconsistent latency of APIs resulting from drivers self getting/putting will lead to really strange behavior for users which can be really hard to track down without an oscilloscope (I have already observed users complaining about "slow" SPI transactions because the self get/put took quite a while on a platform), if a user tries bit banging something via GPIO it could work just fine, and then suddenly start failing since some other part of the system happened to keep the GPIO resumed, thus ensuring consitent latency.

But, realistically it seems inline get/put will not be removed any time soon, maybe ever :)

bjarki-andreasen, 2026-08-17:
@RichardSWheatley I added the following to #89031

### Device runtime power management + Device driver self management
Device drivers call the device runtime power management APIs on themselves. This is a common pattern in event driven devices with short resume/suspend times like UARTs and other busses today, allowing calls like `uart_tx()` to internally call pm_device_runtime_get() on themselves during the transaction. This makes applications simpler, and can even be more power efficient since pm_device_runtime_get() can be pushed to just before the hardware needs to be resumed, and pm_device_runtime_put() called right after the transaction is complete.

Pros:
* Less API calls needed for the application, saving flash
* Can achieve less power consumption if devices have short enough resume/suspend times
* All the pros of Device runtime power management since users can still resume the devices to minimize latency using the same framework

Cons:
* Same cons as Device runtime power management for devices with long resume/suspend times

@ceolin I think allowing devices self-resuming is the common sentiment in the comments if this RFC, and I don't see any real downside of allowing it, do you agree, if so, could you update this RFCs description to explicitly allow it?

Regarding bi-weekly meeings, I'm also game, probably need to ask for it at TSC maybe to make it an official meeting?