zephyrproject-rtos/zephyr #84051: tests: benchmarks: invalid queue timings on mps2/an385 (qemu)

golowanow, 2025-01-15:
**Describe the bug**
Invalid queue benchmark time measurements observed on mps2/an385 (qemu simulation).

**To Reproduce**
`twister -p mps2/an385 -vv -j 1 -T tests/benchmarks/sched_queues`
`twister -p mps2/an385 -vv -j 1 -T tests/benchmarks/wait_queues`
similar results for `mps2/an521/cpu0`

**Expected behavior**
correct time measurements and benchmark results.

**Impact**
* invalid benchmark values.
* possible incorrect timings on the platform(s)

**Logs and console output**
`twister -p mps2/an385 -vv -j 1 -T tests/benchmarks/sched_queues`
```
twister - DEBUG - run test: mps2/an385/zephyr/tests/benchmarks/sched_queues/benchmark.sched_queues.multiq
twister - DEBUG - Reset instance status from 'passed' to None before run.
twister - DEBUG - Spawning QEMUHandler Thread for mps2/an385/zephyr/tests/benchmarks/sched_queues/benchmark.sched_queues.multiq
twister - DEBUG - Running mps2/an385/zephyr/tests/benchmarks/sched_queues/benchmark.sched_queues.multiq (qemu)
twister - DEBUG - Spawning QEMUHandler Thread for mps2/an385/zephyr/tests/benchmarks/sched_queues/benchmark.sched_queues.multiq
twister - DEBUG - QEMU (1103361): *** Booting Zephyr OS build v4.0.0-3379-g0df6736bb9f6 ***
twister - DEBUG - QEMU (1103361): Time Measurements for multiq sched queues
twister - DEBUG - QEMU (1103361): Timing results: Clock frequency: 25 MHz
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_tail.min         - Add threads of decreasing priority, min.           :     393 cycles ,   15720 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_tail.max         - Add threads of decreasing priority, max.           : 18446744073693168 cycles , 3607117184 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_tail.avg         - Add threads of decreasing priority, avg.           :   39820 cycles , 1592800 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_tail.stddev      - Add threads of decreasing priority, stddev.        : 290614440 cycles , 3034643008 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_head.min      - Remove threads of decreasing priority, min.        :     352 cycles ,   14080 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_head.max      - Remove threads of decreasing priority, max.        : 18446744073676388 cycles , 3606445984 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_head.avg      - Remove threads of decreasing priority, avg.        : 184467440735439 cycles , 3944431704 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_head.stddev   - Remove threads of decreasing priority, stddev.     : 416111575 cycles , 3759561112 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_head.min         - Add threads of increasing priority, min.           :     393 cycles ,   15720 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_head.max         - Add threads of increasing priority, max.           : 18446744073676391 cycles , 3606446104 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_head.avg         - Add threads of increasing priority, avg.           : 184467440733462 cycles , 3944352624 ns :
twister - DEBUG - QEMU (1103361): REC: sched.add.thread.ReadyQ_head.stddev      - Add threads of increasing priority, stddev.        : 377361135 cycles , 2209543512 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_tail.min      - Remove threads of increasing priority, min.        :     352 cycles ,   14080 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_tail.max      - Remove threads of increasing priority, max.        : 18446744073676349 cycles , 3606444424 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_tail.avg      - Remove threads of increasing priority, avg.        : 184467440734768 cycles , 3944404864 ns :
twister - DEBUG - QEMU (1103361): REC: sched.remove.thread.ReadyQ_tail.stddev   - Remove threads of increasing priority, stddev.     : 424670937 cycles , 4101935592 ns :
twister - DEBUG - QEMU (1103361): ===================================================================
twister - DEBUG - QEMU (1103361): PROJECT EXECUTION SUCCESSFUL
twister - DEBUG - HARNESS:Console:EXPECTED:'PROJECT EXECUTION SUCCESSFUL'
twister - DEBUG - QEMU (1103361) complete with passed (None) after 2.4397289752960205 seconds
twister - DEBUG - No timeout, return code from QEMU (None): 0
twister - DEBUG - return code from QEMU (None): 0
twister - DEBUG - run status: mps2/an385/zephyr/tests/benchmarks/sched_queues/benchmark.sched_queues.multiq passed

```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain  Zephyr SDK 0.17.0
 - Commit SHA 0df6736

gramsay0, 2025-01-15:
The sysclock calculations overflow on mps2/an385 with `CONFIG_SYS_CLOCK_TICKS_PER_SEC=1` (see https://github.com/zephyrproject-rtos/zephyr/pull/66601).
Try with `CONFIG_SYS_CLOCK_TICKS_PER_SEC=3`

golowanow, 2025-01-15:
#84051 also seems to fix the overflow, but when the benchmark has large `BENCHMARK_NUM_ITERATIONS` e.g. 10000, the overflow reappears:
```
Spawning QEMUHandler Thread for mps2/an385/tests/benchmarks/sched_queues/benchmark.sched_queues.multiq
QEMU (1131936): *** Booting Zephyr OS build v4.0.0-2779-ge8f2e7699ee6 ***
QEMU (1131936): Time Measurements for multiq sched queues
QEMU (1131936): Timing results: Clock frequency: 25 MHz
QEMU (1131936): REC: sched.add.thread.ReadyQ_tail.min         - Add threads of decreasing priority, min.           :     390 cycles ,   15600 ns :
QEMU (1131936): REC: sched.add.thread.ReadyQ_tail.max         - Add threads of decreasing priority, max.           :    7101 cycles ,  284040 ns :
QEMU (1131936): REC: sched.add.thread.ReadyQ_tail.avg         - Add threads of decreasing priority, avg.           :    1967 cycles ,   78680 ns :
QEMU (1131936): REC: sched.add.thread.ReadyQ_tail.stddev      - Add threads of decreasing priority, stddev.        :    2322 cycles ,   92880 ns :
QEMU (1131936): REC: sched.remove.thread.ReadyQ_head.min      - Remove threads of decreasing priority, min.        :     368 cycles ,   14720 ns :
QEMU (1131936): REC: sched.remove.thread.ReadyQ_head.max      - Remove threads of decreasing priority, max.        : 1844674406943536 cycles , 773177216 ns :
QEMU (1131936): REC: sched.remove.thread.ReadyQ_head.avg      - Remove threads of decreasing priority, avg.        : 18446744071112 cycles , 2971326272 ns :
QEMU (1131936): REC: sched.remove.thread.ReadyQ_head.stddev   - Remove threads of decreasing priority, stddev.     : 374469184 cycles , 2093865472 ns :
```

PR #84050 (2025-01-17): arch: common: timing: Fix timing cycles 32bit rollover
Fix arch_timing_cycles_get() to prevent overflow on 32bit cycles rollover.

The issue is observable, for example when `tests/benchmarks/wait_queues` or `tests/benchmarks/sched_queues` are executed on qemu for `mps2/an385` and the benchmark has its iterations large enough as the default `BENCHMARK_NUM_ITERATIONS=1000`.

Fixes #84051
closed 2025-01-17 by kartben