zephyrproject-rtos/zephyr #84078: Stack protection does not work with native_sim, newlibc and posix API

erlingrj, 2025-01-16:
**Describe the bug**
The stack protection, enabled by `CONFIG_STACK_CANARIES=y` terminate trivial programs on `native_sim` when the POSIX API is used. When stepping through the code it appears to happen inside `pthread_cond_wait`.

**To Reproduce**
prj.conf
```
CONFIG_NETWORKING=y
CONFIG_POSIX_API=y
CONFIG_STACK_CANARIES=y
```
Note: For some reason just selecting `CONFIG_NEWLIB_LIBC=y` is not enough to get it compiled with newlib and get the stack protection working. It then just defaults back to picolib. But when selecting networking it selects newlib and I get this error.

main.c
```
#include <zephyr/kernel.h>

int main(void) {
  printf("Hello Before Sleep!\n");
  k_usleep(10000);
  printf("Hello After Sleep!\n");
  return 0;
}
```

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.20.0)

set(BOARD native_sim)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(blinky)

target_sources(app PRIVATE src/main.c)
```

Compile and run:
```sh
west build -p always -t run
```

**Logs and console output**

```sh
*** Booting Zephyr OS build v3.7.0 ***
Hello Before Sleep!
@ WEST_TOPDIR/deps/zephyr/kernel/compiler_stack_protect.c:39
Exiting due to fatal error

FAILED: zephyr/CMakeFiles/run_native /home/erling/dev/lf-west-template/apps/HelloZephyr/build/zephyr/CMakeFiles/run_native 
cd /home/erling/dev/lf-west-template/apps/HelloZephyr/build && /home/erling/dev/lf-west-template/apps/HelloZephyr/build/zephyr/zephyr.exe
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/erling/dev/lf-west-template/apps/HelloZephyr/build --target run
```
**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Toolchain: zephyr-sdk-0.16.8
 - zephyr v3.7.0

**Additional context**

kartben, 2025-01-21:
cc @aescolar

aescolar, 2025-01-22:
@erlingrj CONFIG_STACK_CANARIES is not supported in native_sim (in the posix architecture):
https://docs.zephyrproject.org/latest/boards/native/doc/arch_soc.html#posix-arch-unsupported

Sidetrack:
>  For some reason just selecting CONFIG_NEWLIB_LIBC=y is not enough to get it compiled with newlib

You cannot select NEWLIB_LIBC with native_sim either. That's why kconfig goes back to picolibc.

aescolar, 2025-01-22:
For clarification, there is no need to enable all the options in the bug description. It is enough to do
`cmake -GNinja -DBOARD=native_sim ../samples/hello_world/ -DCONFIG_STACK_CANARIES_ALL=y -DCONFIG_ENTROPY_GENERATOR=y` in main today.
The issue is just that CONFIG_STACK_CANARIES is not supported in the posix arch/native_sim. As it was never made to work. This is documented and as such not a bug. It could be a feature request. Maybe we could have kconfig prevent this from being set.

From a quick look, it seems the reason why this particular feature does not work in native_sim is that, the kernel init is initializing `__stack_chk_guard` "too late" in this platform from the point of view of the execution in native_sim, and therefore it looks like it just corrupted the stack guard instead of initializing it which leads to the stack check failing for the host thread that just run z_cstart().

If you want the equivalent functionality when building for native_sim you can add `-DEXTRA_CPPFLAGS=-fstack-protector-all` to your build command.

I'd close this as not an issue.

erlingrj, 2025-01-22:
Thank you for the explanation @aescolar that makes sense. It is not a problem for me to disable the stack canaries option, if it is a simple fix, then I think it would be very helpful with a warning/error from Kconfig if board is `native_sim` and `CONFIG_STACK_CANARIES=y`.

SzymonRichert, 2025-03-11:
@erlingrj can you close this one, after @aescolar explanation ?

nashif, 2025-03-11:
not a bug,

closed 2025-03-11 by nashif