zephyrproject-rtos/zephyr #84085: west: blobs fetch

WilliamGFish, 2025-01-16:
**West blobs fetch failure**

When attempting to retrieve blobs using the west command receive an error.

https://github.com/zephyrproject-rtos/mbedtls/pull/61
(https://github.com/zephyrproject-rtos/mbedtls/pull/61/files#diff-27d260b3de385946dc5c56da8181a3a9d4f5897cfab9df6dc2f5bb56bb0bbd6f)
zephyr/module.yml
```
build:
  cmake-ext: True
  kconfig-ext: True
security:
  external-references:
    - cpe:2.3:a:arm:mbed_tls:3.6.1:*:*:*:*:*:*:*
    - pkg:github/Mbed-TLS/mbedtls@V3.6.1
```

By reversing PR61
The west functionality works as expected

**To Reproduce**

```
\zephyr>west blobs fetch hal_infineon
ERROR: Malformed "build" section in file: /zephyr/modules/crypto/mbedtls/zephyr/module.yml
<SchemaError: error code 2: Schema validation failed:
 - Key 'security' was not defined. Path: ''.: Path: '/'>
```

**Expected behaviour**
Retrieval of the defined files/blobs.

**Logs and console output**

> \zephyr>west blobs fetch hal_infineon
> ERROR: Malformed "build" section in file: C:/Users/BillyBob/zephyrGoodX/modules/crypto/mbedtls/zephyr/module.yml
> <SchemaError: error code 2: Schema validation failed:
>  - Key 'security' was not defined. Path: ''.: Path: '/'>

**Environment**

 - OS: Windows
 - Toolchain Zephyr SDK

kartben, 2025-01-16:
I am guessing you're on Zephyr <= 3.6, correct? 
This is due to https://github.com/zephyrproject-rtos/zephyr/pull/66495 introducing new properties to describe CPE/PURL for modules, but the zephyr_module parser is apparently not happy when it sees keys it doesn't know about (which would be the case in Zephyr 3.6 or any version before https://github.com/zephyrproject-rtos/zephyr/pull/66495/commits/30eb588b4167a45d2d6e27fd26521f38785e3b23 went in.

WilliamGFish, 2025-01-16:
I am using the very latest version (v4.0.0-3409-g9ba1f2e1704a)...BUT

It appears that my scripts folder has been come corrupted somehow. I shall try to manually resolve.

Thanks for the help

closed 2025-01-16 by WilliamGFish