zephyrproject-rtos/zephyr #84097: samples/bluetooth/peripheral_sc_only add mbedtls psa stack overflow

zhang-wenchao, 2025-01-16:
**Describe the bug**

When I use this example, a stack overflow occurs.

**To Reproduce**

`prj.conf` additional content.
```
CONFIG_LOG=y
CONFIG_STACK_SENTINEL=y

CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_PSA_CRYPTO_C=y

# Removing these two lines will return to normal
CONFIG_PSA_WANT_ALG_SHA_512=y
CONFIG_PSA_WANT_ALG_SHA_256=y
```

`west build -p always -b esp32c3_luatos_core/esp32c3/usb && west flash && west espressif monitor`

**Expected behavior**

working normally

**Impact**

bluetooth not working properly!

**Logs and console output**

```
[00:00:00.024,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.097,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.097,000] <inf> bt_hci_core: Identity: B0:81:84:AE:C7:28 (public)
[00:00:00.097,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.097,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Bluetooth initialized
Advertising successfully started
Connected 63:C0:1F:B7:0F:4E (random)
Passkey for 63:C0:1F:B7:0F:4E (random): 802256
[00:00:11.187,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:22.087,000] <err> os:      a3: 3fc9a350    t3: 00b39c6c
--- 6 messages dropped ---
[00:00:22.087,000] <err> os:      a4: 3fc94030    t4: 00000062
[00:00:22.087,000] <err> os:      a5: f0f0f0f0    t5: 418d72e0
[00:00:22.087,000] <err> os:      a6: c97dc9b8    t6: 08686df8
[00:00:22.087,000] <err> os:      a7: 1db966f6
[00:00:22.087,000] <err> os:      sp: 3fc940b0
[00:00:22.087,000] <err> os:      ra: 403801f4
[00:00:22.087,000] <err> os:    mepc: 40389216
[00:00:22.087,000] <err> os: mstatus: 00001800
[00:00:22.087,000] <err> os: 
[00:00:22.087,000] <err> os:      s0: 3fc9a2cc    s6: 00000001
[00:00:22.087,000] <err> os:      s1: 00000000    s7: 03c00200
[00:00:22.087,000] <err> os:      s2: 6dabeaaa    s8: 00000000
[00:00:22.087,000] <err> os:      s3: 3fc943bc    s9: 00000000
[00:00:22.087,000] <err> os:      s4: 3fc94178   s10: 00000000
[00:00:22.087,000] <err> os:      s5: 00000005   s11: 00000000
[00:00:22.087,000] <err> os: 
[00:00:22.087,000] <err> os: call trace:
[00:00:22.087,000] <err> os:       0: sp: 3fc940f4 ra: 420125ac
[00:00:22.087,000] <err> os:       1: sp: 3fc94130 ra: 42013960
[00:00:22.087,000] <err> os:       2: sp: 3fc94160 ra: 42015d30
[00:00:22.087,000] <err> os:       3: sp: 3fc94170 ra: 42015e1e
[00:00:22.087,000] <err> os:       4: sp: 3fc94310 ra: 420164cc
[00:00:22.087,000] <err> os:       5: sp: 3fc94360 ra: 42010b04
[00:00:22.087,000] <err> os:       6: sp: 3fc943a0 ra: 4201080e
[00:00:22.087,000] <err> os:       7: sp: 3fc94400 ra: 42009980
[00:00:22.087,000] <err> os: 
[00:00:22.087,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:22.087,000] <err> os: Current thread: 0x3fc990b8 (unknown)
[00:00:22.143,000] <err> os: Halting system
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit 819acf18f00b58d26e1c182a08106c38e5e40b9e

zhang-wenchao, 2025-01-16:
Because this problem is too complicated, I don't know where the problem occurs, esp32c3? bt? mbedtls?

zhang-wenchao, 2025-01-16:
Help me, thanks.

zhang-wenchao, 2025-01-16:
@ceolin Hi, can you help me look into this issue? I think it is related to mbedtls.

zhang-wenchao, 2025-01-16:
I temporarily turned off `CONFIG_STACK_SENTINEL`, which is obviously not a good solution.

valeriosetti, 2025-01-16:
> [@valeriosetti](https://github.com/valeriosetti) [@ceolin](https://github.com/ceolin) I'm very sorry to bother you, can you help me look into this issue? I think it is related to mbedtls.

Since BT should already enable `CONFIG_PSA_WANT_xxx` symbols it needs, I assume you are enabling them for some other reason (ex: doing some extra crypto thing in some other thread not related to BT). So my question is: did you already tried to increase the stack size of that thread?

zhang-wenchao, 2025-01-16:
I modified these, but the problem still exists

```
CONFIG_MAIN_STACK_SIZE=10240
CONFIG_IDLE_STACK_SIZE=2048
CONFIG_ISR_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=51200
```

jhedberg, 2025-01-17:
Could you try increasing `CONFIG_BT_LONG_WQ_STACK_SIZE` as well, since that's there the ECDH computation for pairing happens?

If that doesn't help, also try increasing `CONFIG_BT_RX_STACK_SIZE`.

And when you've made it work, please gather statistics on the actual usage (`CONFIG_THREAD_ANALYZER=y`, etc), so that we can make the necessary adjustments upstream. Thanks.

zhang-wenchao, 2025-01-17:
Thanks,  `CONFIG_BT_RX_STACK_SIZE` It worked, `CONFIG_BT_LONG_WQ_STACK_SIZE` It is not a user setting, so no PR is needed for this issue.

closed 2025-01-17 by zhang-wenchao