zephyrproject-rtos/zephyr #84099: extra-args to twister don't make it to the cmake cache of on-target builds

solsbarry, 2025-01-16:
**Describe the bug**

I am writing unit tests to run on hardware, using twister. I have a need to pass extra arguments (using -x, --extra-args) to cmake at the command line when I invoke twister. The output folder twister generates seems to have two different cmake areas, one for the overall invocation of twister, and then a subfolder for each test.  However the extra arguments are only defined in the CMakeCache.txt file for the overall test folder, and they are not defined in the test specific CMakeCache.txt file (in the subfolder). And so my cmakelist.txt file which depends on these variables being available does not work right.

**To Reproduce**

1. Generate a unit test that runs on a hardware platform.
2. Add something in the cmakelist.txt file for that test that uses a variable that should come from the cmake cache (not otherwise defined). Which if its not defined causes the build to fail.
3. invoke the test with twister defining your above variable with -x=MY_VARIABLE=VALUE
4. See that the build fails, and that the variable is only defined in the overall twister cmake cache, and not in the test specific one. 

**Expected behavior**

I would expect the variable to be defined in the tests cmake cache file.

**Impact**

I can't use twister, and will probably end up creating my own test orchestration framework. 

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain: 0.16.8
 - v3.7.99-ncs2 (I know this is a Nordic version, but I don't think there are any twister specific changes in it)

**Additional context**

The same thing happens if I specify the extra-args in the testcase.yaml file.

nashif, 2025-01-20:
> The output folder twister generates seems to have two different cmake areas, one for the overall invocation of twister, and then a subfolder for each test.

not fully getting this, can you provide a more detailed example and pointers to those files your are referring to?

Does things work for you with west or cmake only, without usage of twister? i.e. build with `west build -b <board> -- -DMY_VARIABLE=VALUE`

solsbarry, 2025-01-21:
This one must have been my fault. I am not sure what I changed in my project but I no longer see the issue. Previously I was seeing twister run cmake twice. Generating a build folder within the build folder. And the extra args weren't being propagated. But now I am not seeing those nested build folders, and I am only seeing cmake run once. And now my extra args are making it to my cmakelist.txt file. So anyway this one is a false alarm, sorry about that.

closed 2025-01-21 by solsbarry