zephyrproject-rtos/zephyr #84100: net: coap: Request/Response Matching Rules (RFC 7252 section-5.3.2) are not applied correctly

Aluleichtbau, 2025-01-16:
**Describe the bug**

`coap_response_received` function is used to find (and handle) a received response in an array of expected `struct coap_reply`. RFC 7252 section-5.3.2 specifies the rules for matching the reply. zTesting my own CoAP client application I stumbled over issues:
- piggybacked ACK with mid not identical are matched anyway if the token matches.
- if the tokenlen in the received message is zero, it still can be matched to replies having a token if they have the same mid

I also wondered why `((r->id == 0U) && (r->tkl == 0U))` is a criterion to not match. In `coap_reply_next_unused` replies are freed if `r->reply == NULL` what might be the purpose of the check.

**To Reproduce**

0. Make sure you have a callback
```c
static int response_handled_cb(const struct coap_packet *response,
			     struct coap_reply *reply,
			     const struct sockaddr *from)
{
	return 0;
}
```
1. Create a reply array like this:
```c
	struct coap_reply replies[] = {
		{.id = 124, .token = "token", .tkl = 5, .reply = response_handled_cb},
		{.id = 125, .token = "other", .tkl = 5, .reply = response_handled_cb},
	};
```
3. Create a coap_packet like this:
```c
	struct coap_packet response = {0};
	coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_ACK,
			     strlen("other"), "other", COAP_RESPONSE_CODE_CONTENT, 124);
```
4. Call `coap_response_received(&response, &from, &replies[0], ARRAY_SIZE(replies))` and you will get the 2nd element from the list.
5. Create another coap_packet like this:
```c
	coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_CON,
			     0, NULL, COAP_RESPONSE_CODE_CONTENT, 124);
```
6. Call `coap_response_received(&response, &from, &replies[0], ARRAY_SIZE(replies))` and you will get the  1st element of the list.

**Expected behavior**

RFC says:
> In a piggybacked response, the Message ID of the Confirmable request and the Acknowledgement MUST match, and the tokens of the response and original request MUST match. In a separate response, just the tokens of the response and original request MUST match.

So in 4. and 6. nothing shall be called and NULL shall be returned.

**Impact**

coap implementation is not compliant to RFC 7252 section-5.3.2.

**Logs and console output**

N/A

**Environment (please complete the following information):**

 - OS: Windows 10, WSL Ubuntu
 - Toolchain Zephyr SDK
 - Commit SHA: 4e201f21c8a7f58698e68a5f1cca72433f9778ac

**Additional context**

I prepared tests and fixes for this, I will provide soon.

boaks, 2025-01-17:
> if the tokenlen in the received message is zero, it still can be matched to replies having a token if they have the same mid

Not sure, what this means exactly. If the request uses an empty token, only a response with empty tokens will match. 

[RFC 7252 5.3.1 Token](https://www.rfc-editor.org/rfc/rfc7252#section-5.3.1)

> The client SHOULD generate tokens in such a way that tokens currently
> in use for a given source/destination endpoint pair are unique.
> ...
> An empty token value is appropriate e.g., when
> no other tokens are in use to a destination, or when requests are
> made serially per destination and receive piggybacked responses.

RFC 7252 5.3.2 also applies, so a request with an empty token is matched by a response with an empty token. If it's a piggybacked response (ACK), then also the MID must match.

rlubos, 2025-01-17:
> RFC 7252 5.3.2 also applies, so a request with an empty token is matched by a response with an empty token. If it's a piggybacked response (ACK), then also the MID must match.

If I understood correctly, in the described bug only the response had on empty token. The request carried a non-empty token, yet still the response with no token is matched.

I'll look into this.

rlubos, 2025-01-17:
Please check https://github.com/zephyrproject-rtos/zephyr/pull/84160

Aluleichtbau, 2025-01-20:
Sorry for late response. There is not only piggybacked (ACK and, for robustness, CON and NON_CON) and separate responses (CON and NON_CON), but also empty ACK and RESET which do not carry tokens even though the request did.

Here is the test and the fix I prepared. I tried to tackle also:
- MID = 0 and no token
- Token-len is different, but content is equal, i.e. if comparing the shorter len, the compare results in equal despite being of different len
- MID is equal but token is not
- token is equal but MID is not

```c
ZTEST(coap, test_response_received)
{
	int r;
	struct coap_packet response = {0};
	static const char payload[] = "payload";
	uint8_t *data = data_buf[0];
	struct sockaddr from;
	struct coap_reply replies[] = {
		{.id = 0, .token = "", .tkl = 0, .reply = NULL},
		{.id = 0, .token = "", .tkl = 0, .reply = response_handled_cb},
		{.id = 0, .token = "no mid", .tkl = 6, .reply = response_handled_cb},
		{.id = 123, .token = "", .tkl = 0, .reply = response_handled_cb},
		{.id = 124, .token = "token", .tkl = 5, .reply = response_handled_cb},
		{.id = 125, .token = "token", .tkl = 5, .reply = response_handled_cb},
		{.id = 125, .token = "other", .tkl = 5, .reply = response_handled_cb},
		{.id = 126, .token = "token", .tkl = 3, .reply = response_handled_cb},
	};

	memset(data_buf[0], 0, ARRAY_SIZE(data_buf[0]));

	/* empty ACK for a reply without token*/
	r = coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_ACK, 0, NULL,
			     COAP_CODE_EMPTY, 123);
	zassert_equal(r, 0, "Could not initialize packet");
	zassert_equal_ptr(
		coap_response_received(&response, &from, &replies[0], ARRAY_SIZE(replies)),
		&replies[3], "Wrong reply is picked");

	/* empty ACK for a reply with a token*/
	r = coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_ACK, 0, NULL,
			     COAP_CODE_EMPTY, 124);
	zassert_equal(r, 0, "Could not initialize packet");
	zassert_equal_ptr(
		coap_response_received(&response, &from, &replies[0], ARRAY_SIZE(replies)),
		&replies[4], "Wrong reply is picked");

	/* empty ACK for a reply with mid = 0*/
	r = coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_ACK, 0, NULL,
			     COAP_CODE_EMPTY, 0);
	zassert_equal(r, 0, "Could not initialize packet");
	zassert_equal_ptr(
		coap_response_received(&response, &from, &replies[0], ARRAY_SIZE(replies)),
		&replies[1], "Wrong reply is picked");

	/* empty ACK could not be matched*/
	r = coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_ACK, 0, NULL,
			     COAP_CODE_EMPTY, 1);
	zassert_equal(r, 0, "Could not initialize packet");
	zassert_is_null(coap_response_received(&response, &from, &replies[0], ARRAY_SIZE(replies)),
			"A reply is picked");

	/* piggybacked ACK without token matched */
	r = coap_packet_init(&response, data, COAP_BUF_SIZE, COAP_VERSION_1, COAP_TYPE_ACK, 0, NULL,
			     COAP_RESPONSE_CODE_CONTENT, 123);
	zassert_equal(r, 0, "Could not initialize packet");

	r = coap_packet_append_payload_marker(&response);
	zassert_equal(r, 0, "Failed to set the payload marker");

	r = coap_packet_append_payload(&response, payload, sizeof(payload));
	zassert_equal(r, 0, "Failed to set the payload");
[... 356 lines cut ...]

		if (r->reply == NULL) {
			/* reply not used */
			continue;
		}

		if (coap_response_matches(response, r->id, r->tkl, &r->token[0]) == false) {
			continue;
		}

		age = coap_get_option_int(response, COAP_OPTION_OBSERVE);
		/* handle observed requests only if received in order */
		if (age == -ENOENT || coap_age_is_newer(r->age, age)) {
			r->age = age;
			if (coap_header_get_code(response) != COAP_RESPONSE_CODE_CONTINUE) {
				r->reply(response, r, from);
			}
		}

		return r;
	}

	return NULL;
}
```

rlubos, 2025-01-20:
@Aluleichtbau Please comment on https://github.com/zephyrproject-rtos/zephyr/pull/84160, there's some ongoing discussion in there already. 

> However, empty RESET shall be handled.

@boaks The RESET message bugs me, sure according to the table mentioned in https://github.com/zephyrproject-rtos/zephyr/pull/84160#discussion_r1920530374 RESET is not a valid reply, but still in other parts of the RFC (like https://www.rfc-editor.org/rfc/rfc7252#page-11):

> When a recipient is not at all able to process a Confirmable message (i.e., not even able to provide a suitable error response), it
   replies with a Reset message (RST) instead of an Acknowledgement (ACK).

> When a recipient is not able to process a Non-confirmable message, it may reply with a Reset message (RST).

So perhaps we should still try to match RESET replies with MID only and call the registered callback? Otherwise the client has no notion that the request was rejected.

boaks, 2025-01-20:
> o perhaps we should still try to match RESET replies with MID only and call the registered callback? Otherwise the client has no notion that the request was rejected.

That's right. My statement in the PR was focused on a RESET and the `piggybacked`. 

Beside of the reject, also (in rare cases) a timeout my be relevant for the client to know. 
(I use the coap message encoding /decoding and my own retransmission/matching functions. Therefore I'm not common, how that is handled in zephyr.)

boaks, 2025-01-20:
```
        case COAP_TYPE_CON:
		__fallthrough;
	case COAP_TYPE_NON_CON:
		/* Separate response must match token, if there is no token fall back on comparing
		 * id */
		if (response_tokenlen != tkl) {
			ret = false;
		} else if ((tkl != 0U) && (memcmp(&response_token[0], token, tkl) == 0)) {
			ret = true;
		} else if ((tkl == 0U) && (response_id == id)) {
			ret = true;
		}
		break;
```

The MID is not matched in the case of a separate response. Therefore

```
		} else if ((tkl == 0U) && (response_id == id)) {
```

must be

```
		} else if (tkl == 0U) {
```

(Just for completeness, the currently proposed PR handles that already. )

Aluleichtbau, 2025-01-20:
@boaks Timeout-handling on `coap_pending`s is also done by LWM2M in `/subsys/net/lib/lwm2m/lwm2m_engine.c:390`, i.e. outside of the functions of coap.c. However, LWM2M does not seem to transmit or process RESET at all (not sure wheather it actually requires to or not).

`subsys/net/lib/coap/coap_client.c` processes incoming responses in `handle_response` and does not use the `coap_response_received`. It implements RESET, but I am not sure if the matching there is better. There are probably also issues with the matching, e.g. for piggybacked messages token seems to be checked, but id is not.

Aluleichtbau, 2025-01-20:
@boaks Yeah, RFC says the token shall match in that case. But if there is no token and the ID is not checked, the response literally matches to any request without a token. But as far as I read the section 5.3.1 you are right. Thanks for the remarks. So I need to update the tests also.

boaks, 2025-01-20:
@Aluleichtbau 

> So I need to update the tests also.

Great.

> But if there is no token

[RFC 7252 5.3.1 Token](https://www.rfc-editor.org/rfc/rfc7252#section-5.3.1)

>    The Token is used to match a response with a request.  The token
>    value is a sequence of 0 to 8 bytes.  (Note that every message
>    carries a token, even if it is of zero length.)

There is just no **no token**, that's the empty token. 

>  the response literally matches to any request without a token

Again, not "without a token", but with the empty token.

>    The client SHOULD generate tokens in such a way that tokens currently
>    in use for a given source/destination endpoint pair are unique.
>    (Note that a client implementation can use the same token for any
>    request if it uses a different endpoint each time, e.g., a different
>    source port number.)  An empty token value is appropriate e.g., when
>    no other tokens are in use to a destination, or when requests are
>    made serially per destination and receive piggybacked responses.
>    There are, however, multiple possible implementation strategies to
>    fulfill this.

It the same rule for the empty token as for all others: if in use, it must be unique. Otherwise you will mix up the communication.

PR #84160 (2025-01-21): net: coap: Fix response matching algorithm 
* Add test case for matching pending replies with received responses.
Cover corner cases that are failing with the current implementation.

* Fix response matching algorithm 

Fixes #84100
closed 2025-01-21 by kartben