zephyrproject-rtos/zephyr #84112: esp32-c6-devkitc-1 west debugserver not working

aslansq, 2025-01-16:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/84110

<div type='discussions-op-text'>

<sup>Originally posted by **aslansq** January 16, 2025</sup>
Hello from Croatia as Turkish :)

I like the project a lot but I could not debug my code. Any help will be appreciated
So building works. west flash and debug works. But to efficiently debug I need gui, I am running west debugserver and it does not work. I dont know why?

Do I have to read a lot of openocd cfgs to set it up?

<pre>(.venv) <font color="#26A269"><b>aslan@thinkpad</b></font>:<font color="#12488B"><b>~/zephyrproject</b></font>$ tree -L 1 ${OPENOCD_ESP32_PATH}/bin
<font color="#12488B"><b>/home/aslan/tool/openocd-esp32-linux-amd64-0.12.0-esp32-20241016/openocd-esp32/bin</b></font>
└── <font color="#26A269"><b>openocd</b></font>

1 directory, 1 file
(.venv) <font color="#26A269"><b>aslan@thinkpad</b></font>:<font color="#12488B"><b>~/zephyrproject</b></font>$ west build --sysbuild -p -b esp32c6_devkitc zephyr/samples/hello_world -- -DOPENOCD=${OPENOCD_ESP32_PATH}/bin/openocd -DOPENOCD_DEFAULT_PATH=${OPENOCD_ESP32_PATH}/share/openocd/scripts
<font color="#33DA7A">-- west build: generating a build system</font>
Loading Zephyr module(s) (Zephyr base): sysbuild_default
-- Found Python3: /home/aslan/zephyrproject/.venv/bin/python3 (found suitable version &quot;3.10.12&quot;, minimum required is &quot;3.10&quot;) found components: Interpreter 
-- Cache files will be written to: /home/aslan/.cache/zephyr
-- Found west (found suitable version &quot;1.3.0&quot;, minimum required is &quot;0.14.0&quot;)
-- Board: esp32c6_devkitc, qualifiers: esp32c6
Parsing /home/aslan/zephyrproject/zephyr/share/sysbuild/Kconfig
Loaded configuration &apos;/home/aslan/zephyrproject/build/_sysbuild/empty.conf&apos;
Merged configuration &apos;/home/aslan/zephyrproject/build/_sysbuild/empty.conf&apos;
Configuration saved to &apos;/home/aslan/zephyrproject/build/zephyr/.config&apos;
Kconfig header saved to &apos;/home/aslan/zephyrproject/build/_sysbuild/autoconf.h&apos;
-- 
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
-- Application: /home/aslan/zephyrproject/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.22.1
-- Found Python3: /home/aslan/zephyrproject/.venv/bin/python (found suitable version &quot;3.10.12&quot;, minimum required is &quot;3.10&quot;) found components: Interpreter 
-- Cache files will be written to: /home/aslan/.cache/zephyr
-- Zephyr version: 4.0.99 (/home/aslan/zephyrproject/zephyr)
-- Found west (found suitable version &quot;1.3.0&quot;, minimum required is &quot;0.14.0&quot;)
-- Board: esp32c6_devkitc, qualifiers: esp32c6
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/home/aslan/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/aslan/zephyr-sdk-0.17.0)
-- Found Dtc: /home/aslan/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version &quot;1.6.0&quot;, minimum required is &quot;1.4.6&quot;) 
-- Found BOARD.dts: /home/aslan/zephyrproject/zephyr/boards/espressif/esp32c6_devkitc/esp32c6_devkitc.dts
-- Found devicetree overlay: /home/aslan/zephyrproject/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: /home/aslan/zephyrproject/build/mcuboot/zephyr/zephyr.dts
-- Generated pickled edt: /home/aslan/zephyrproject/build/mcuboot/zephyr/edt.pickle
-- Generated zephyr.dts: /home/aslan/zephyrproject/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/aslan/zephyrproject/build/mcuboot/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/aslan/zephyrproject/build/mcuboot/zephyr/dts.cmake

warning: BOOT_MAX_IMG_SECTORS (defined at
/home/aslan/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig:463) was assigned the value &apos;512&apos;
but got the value &apos;&apos;. Check these unsatisfied dependencies: (!BOOT_MAX_IMG_SECTORS_AUTO) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BOOT_MAX_IMG_SECTORS and/or look up
BOOT_MAX_IMG_SECTORS in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/aslan/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration &apos;/home/aslan/zephyrproject/zephyr/boards/espressif/esp32c6_devkitc/esp32c6_devkitc_defconfig&apos;
Merged configuration &apos;/home/aslan/zephyrproject/bootloader/mcuboot/boot/zephyr/prj.conf&apos;
Merged configuration &apos;/home/aslan/zephyrproject/bootloader/mcuboot/boot/zephyr/socs/esp32c6.conf&apos;
Merged configuration &apos;/home/aslan/zephyrproject/build/mcuboot/zephyr/.config.sysbuild&apos;
Configuration saved to &apos;/home/aslan/zephyrproject/build/mcuboot/zephyr/.config&apos;
Kconfig header saved to &apos;/home/aslan/zephyrproject/build/mcuboot/zephyr/include/generated/zephyr/autoconf.h&apos;
-- Found GnuLd: /home/aslan/zephyr-sdk-0.17.0/riscv64-zephyr-elf/riscv64-zephyr-elf/bin/ld.bfd (found version &quot;2.38&quot;) 
[... 206 lines cut ...]
[Inferior 1 (Remote target) detached]
(.venv) <font color="#26A269"><b>aslan@thinkpad</b></font>:<font color="#12488B"><b>~/zephyrproject</b></font>$ west debugserver
<font color="#33DA7A">-- west debugserver: rebuilding</font>
[0/6] Performing build step for &apos;hello_world&apos;
ninja: no work to do.
[1/6] Performing build step for &apos;mcuboot&apos;
ninja: no work to do.
[6/6] Completed &apos;mcuboot&apos;
<font color="#33DA7A">-- west debugserver: using runner openocd</font>
<font color="#33DA7A">-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available</font>
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting &apos;jtag&apos;
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
DEPRECATED! use &apos;tcl port&apos; not &apos;tcl_port&apos;
DEPRECATED! use &apos;gdb port&apos;, not &apos;gdb_port&apos;
invalid command name &quot;reset&quot;
<font color="#F66151">FATAL ERROR: command exited with status -11: /home/aslan/tool/openocd-esp32-linux-amd64-0.12.0-esp32-20241016/openocd-esp32/bin/openocd -s /home/aslan/zephyrproject/zephyr/boards/espressif/esp32c6_devkitc/support -s /home/aslan/tool/openocd-esp32-linux-amd64-0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts -f /home/aslan/zephyrproject/zephyr/boards/espressif/esp32c6_devkitc/support/openocd.cfg -c &apos;tcl_port 6333&apos; -c &apos;telnet_port 4444&apos; -c &apos;gdb_port 3333&apos; -c &apos;reset init&apos;</font>
(.venv) <font color="#26A269"><b>aslan@thinkpad</b></font>:<font color="#12488B"><b>~/zephyrproject</b></font>$ 
</pre>

</div>

aslansq, 2025-01-16:
[image]

CLI succeeds but it does not do anything. It does not stop in breakpoint.

erhankur, 2025-01-25:
You could use the west debug command as follows:
```
west debug --target-handle '_TARGETNAME_0' --cmd-reset-halt 'init; reset'
```

But running the above command might result in an error like:
```
Cannot access memory at address 0x4200017a
Could not insert hardware breakpoints:
```

To fix this issue;

1- Use the master branch from our OpenOCD fork. https://github.com/espressif/openocd-esp32
You can build it for your machine or get distro from the latest workflow. https://github.com/espressif/openocd-esp32/actions/runs/12951143481

2- If you are are using McuBoot
```
openocd -c 'set ESP_RTOS Zephyr' -f board/esp32c6-builtin.cfg -c 'init; reset halt; esp appimage_offset 0x20000; reset'
```

For the simple boot
```
openocd -c 'set ESP_RTOS Zephyr' -f board/esp32c6-builtin.cfg -c 'init; reset halt; esp appimage_offset 0; reset'
```

3- And run gdb in another terminal

```
riscv32-esp-elf-gdb \
  -ex 'target extended-remote :3333' \
  -ex 'symbol-file build/zephyr/zephyr.elf' \
  -ex 'mon reset halt' \
  -ex 'maintenance flush register-cache' \
  -ex 'break main' \
  -ex 'continue'
```

OpenOCD part should also work with west, but you may need to run Espressif `riscv32-esp-elf-gdb`  I didn't try zephyr gdb binary with the Espressif targets.
```
west debug --target-handle '_TARGETNAME_0' --cmd-reset-halt 'init; reset halt; esp appimage_offset 0x20000; resume'
```

aslansq, 2025-01-26:
Thanks @erhankur . It works
Kolay gelsin

closed 2025-04-11 by github-actions[bot]