zephyrproject-rtos/zephyr #84115: Zephyr UART over BLE does not work for ESP32S3

ubieda, 2025-01-16:
**Describe the bug**
As mentioned in Adafruit's blog post (https://adafruit-playground.com/u/SamBlenny/pages/zephyr-shell-over-usb-or-ble-bluefruit-connect), UART over BLE does not build for ESP32S3. Explanation in the article:

> west build -p always -b adafruit_qt_py_esp32s3/esp32s3/procpu -S nus-console samples/subsys/shell/shell_module/
> 
> When I tried it, I got this error:
> 
> FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map [...lots of stuff edited out...]
> /home/sam/code/zephyr-workspace/zephyr/soc/espressif/esp32s3/hw_init.c:53: undefined reference to `esp_console_init' 
> 
> The error is happening because [zephyrproject-rtos/zephyr/soc/espressif/esp32s3/hw_init.c](https://github.com/zephyrproject-rtos/zephyr/blob/51efc600871ac69c75d0de852d65c8cae3143288/soc/espressif/esp32s3/hw_init.c#L60-L62) unconditionally makes a call to   esp_console_init(), while the definition of esp_console_init() happens inside of an #ifdef CONFIG_ESP_CONSOLE_UART block in [zephyrproject-rtos/hal_espressif/zephyr/common/console_init.c](https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/zephyr/common/console_init.c#L36-L53). When you specify the nus-console snippet option, it turns off that definition block, so the linker can't find esp_console_init().
>

**Expected behavior**
Builds as expected.

**Impact**
ESP32S3 can't use Bluetooth over .

**Suggested Action**
Implement the fix suggested in the article, preferrably by the author himself (@samblenny)

ubieda, 2025-01-16:
Hi @samblenny, thanks for the article! Would you like to submit your fix upstream? We'd love to have you as a Zephyr contributor.

samblenny, 2025-01-16:
@ubieda, thanks for the mention. You're welcome to add the change on my behalf if you like.

PR #84532 (2025-01-30): Makes esp_console_init() calling conditional
Makes the esp_console_init() calling during hardware initialization conditioned to CONFIG_ESP_CONSOLE_UART and to
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG since the definition of esp_console_init() definition is conditioned to these both parameter.

Fixes #84115
closed 2025-01-30 by kartben