zephyrproject-rtos/zephyr #84138: intel_adsp: xt-clang: unknown warning group -Wunknown-warning-option.

gbarkadiusz, 2025-01-17:
**Describe the bug**
intel_adsp/ace15_mtpm/sim
```
INFO    - 1) tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc on intel_adsp/ace15_mtpm/sim error (Build failure)
INFO    - 2) tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc_negative_testing on intel_adsp/ace15_mtpm/sim error (Build failure)
INFO    - 3) tests/lib/mem_alloc/libraries.libc.picolibc.mem_alloc on intel_adsp/ace15_mtpm/sim error (Build failure)
INFO    - 4) tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random on intel_adsp/ace15_mtpm/sim error (Build failure)
```

intel_adsp/ace30/ptl/sim
```
INFO    - 1) tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random on intel_adsp/ace30/ptl/sim error (Build failure)
INFO    - 2) tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc on intel_adsp/ace30/ptl/sim error (Build failure)
INFO    - 3) tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc_negative_testing on intel_adsp/ace30/ptl/sim error (Build failure)
INFO    - 4) tests/lib/mem_alloc/libraries.libc.picolibc.mem_alloc on intel_adsp/ace30/ptl/sim error (Build failure)
```

intel_adsp/ace20_lnl/sim
```
INFO    - 1) tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc on intel_adsp/ace20_lnl/sim error (Build failure)
INFO    - 2) tests/lib/mem_alloc/libraries.libc.minimal.mem_alloc_negative_testing on intel_adsp/ace20_lnl/sim error (Build failure)
INFO    - 3) tests/lib/mem_alloc/libraries.libc.picolibc.mem_alloc on intel_adsp/ace20_lnl/sim error (Build failure)
INFO    - 4) tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random on intel_adsp/ace20_lnl/sim error (Build failure)
```

**To Reproduce**
`twister -p intel_adsp/ace20_lnl/sim -T  tests/lib/mem_alloc/`

**Logs and console output**
`/tests/lib/mem_alloc/src/main.c:24:32: error: unknown warning group '-Walloc-size-larger-than=', ignored [-Werror,-Wunknown-warning-option]
#pragma GCC diagnostic ignored "-Walloc-size-larger-than="`

**Environment:**

 - OS: Linux
 - Toolchain: xt-clang
 - Commit SHA or Version used: v4.0.0-3486-g4cc76e8a9ae3

**Additional context**
b9d7b1bdae9b41d3a73070536821c481f77633f1 is the first bad commit
```
commit b9d7b1bdae9b41d3a73070536821c481f77633f1
Author: Tom Hughes <tomhughes@chromium.org>
Date:   Mon Jan 13 12:16:12 2025 -0800

    cmake clang/compiler_flags.cmake: Re-enable -Wunknown-warning-option

    All warnings in the code base have been resolved.

    Signed-off-by: Tom Hughes <tomhughes@chromium.org>
```

gbarkadiusz, 2025-01-17:
@thughes FYI.

thughes, 2025-01-17:
From the error, it looks like the issue is that the `xt-clang` compiler doesn't recognize the `-Walloc-size-larger-than` warning [here](https://github.com/zephyrproject-rtos/zephyr/blob/2a6e5902cc73f464c120019292f20f28940a66bf/tests/lib/mem_alloc/src/main.c#L23-L25).

I see that the warning is listed in the [GCC warning options](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html), but not [clang diagnostic flags](https://clang.llvm.org/docs/DiagnosticsReference.html).

I have a PR that makes it easier to disable warnings only for clang or only for gcc: https://github.com/zephyrproject-rtos/zephyr/pull/84063. With that change we can make this a gcc-specific warning, which should resolve this issue. (I don't have access to the `xt-clang` compiler, so I can't test myself.)

PR #84176 (2025-03-14): cmake: compiler/xt-clang: add -Wno-unknown-warning-option
xt-clang is usually based on older version of clang, and Zephyr main targets more recent versions. Because of this, some newer compiler flags may cause warnings where twister would mark as test being failed. To workaround that, add -Wno-unknown-warning-option to suppress those warnings.

Fixes #84138
commit 15411747 (2025-03-20): everywhere: Use correct macro for gcc-specific warnings

Many warnings were disabled for all compilers, even though they are
gcc-specific warnings. Now that clang has -Wunknown-warning-option
enabled, this can cause compilation failures when building with clang
toolchains.

Use TOOLCHAIN_DISABLE_GCC_WARNING for all gcc-specific macros.

https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html
https://clang.llvm.org/docs/DiagnosticsReference.html

Fixes: #84138
closed 2025-03-14 by kartben