zephyrproject-rtos/zephyr #84146: Missing nullptr check for API call in "net/phy.h"

8t-tni, 2025-01-17:
**Describe the bug**

The function `static inline int phy_link_callback_set(const struct device *dev, phy_callback_t callback,void *user_data)` claims to return the error code "-ENOTSUP" if the API call is not supported.

But the function misses a correspondung nullptr check.

**Current implementation:**
```
static inline int phy_link_callback_set(const struct device *dev, phy_callback_t callback,
					void *user_data)
{
	const struct ethphy_driver_api *api = (const struct ethphy_driver_api *)dev->api;

	return api->link_cb_set(dev, callback, user_data);
}
```

**Posseblie fix:**
```
static inline int phy_link_callback_set(const struct device *dev, phy_callback_t callback,
					void *user_data)
{
	const struct ethphy_driver_api *api = (const struct ethphy_driver_api *)dev->api;
        if (api->link_cb_set != NULL)
	    return api->link_cb_set(dev, callback, user_data);
        else
            return -ENOTSUP;
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a little sample application using a STM32 nucleo H7
2. Try to set a callback using `phy_link_callback_set`
3. Run your sample application

**Expected behavior**
The function returns -ENOTSUP indicating that the ehternet driver does not support that API call.

**Impact**
The sample application crashes with a memory/access violation.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Windows with VsCode and WSL/docker
 - Toolchain Zephyr SDK

**Additional context**

kartben, 2025-01-20:
@8t-tni thanks for the report! Would you be interested in submitting a pull request to fix this?

sayooj-aerlync, 2025-03-11:
Hi All. I can raise the pull request for this, It looks like all the functions in the file needs a null check.

sayooj-aerlync, 2025-03-12:
Hi All,

Here is the pull request, please review: https://github.com/zephyrproject-rtos/zephyr/pull/86986

sayooj-aerlync, 2025-03-13:
Are there anymore Reviews required to merge this?

maass-hamburg, 2025-03-14:
@sayooj-aerlync you can check https://merge-list.zephyrproject.io/ If you see your PR there and its white and not grey, it will be merged soon.

sayooj-aerlync, 2025-03-14:
Understood! Thanks! @maass-hamburg

closed 2025-03-14 by kartben