zephyrproject-rtos/zephyr #84148: 3.7: Bluetooth: TBS: TBS client may return NULL instead of the empty string in callbacks

Thalley, 2025-01-17:
**Describe the bug**
If the TBS client receives an empty value when reading a string, it treated that as NULL, but since empty strings are still strings, it should simply treat it as such, and provide `"\0"` instead. The callbacks are documented to return strings. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash the https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/tmap_peripheral from the 3.7 branch
2. Connect a Android 15 phone to the sample
3. See that URI list is empty when read but `NULL` provided to the callback

**Expected behavior**
Expect the TBS client to follow its documentation and always return a string when there's no read error. 

**Impact**
Confused developers. The TMAP sample would call `strlen` and `strncpy` on a NULL value. 

**Logs and console output**
```
CCP: Discovered GTBS
CCP: Discovered remote URI: �w
CCP initialized

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 1182a4ca73abf5c30be2f09261778700135ecd79 (v3.7-branch)

**Additional context**
N/A

kartben, 2025-01-21:
triaging as low due to impact being described as "confusion"

Thalley, 2025-02-03:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/84149