zephyrproject-rtos/zephyr #84155: UART HAL is not being compiled for Apollo 3

paulozulato, 2025-01-17:
Hi,

I'm facing an issue trying to use the UART hal on Ambiq Apollo 3. Follow the report:

**Describe the bug**

I'm trying to compile an application for Ambiq Apollo 3 using Zephyr 4.0.0, but I'm getting undefined reference for `am_hal_uart_*` API and I could not find a way to include it into compilation by configuration. In the end, I explicitly added a `zephyr_library_sources` command into my app's CMakeLists.txt to explicitly include the `modules/hal/ambiq/mcu/apollo3/hal/am_hal_uart.c` source file.

**To Reproduce**

Try to compile a simple application using UART API for Ambiq Apollo3 on Zephyr 4.0.0:
```
#include "am_mcu_apollo.h"

int main(void)
{
    void *_uart_handler = NULL;
    am_hal_uart_initialize(0, &_uart_handler);
    am_hal_uart_power_control(_uart_handler, AM_HAL_SYSCTRL_WAKE, false);

    return 0;
}
```

**Expected behavior**
The application is compiled successfully.

**Impact**
I could workaround it, but it seems the UART module is missing on Apollo3's MCU lib.

**Logs and console output**

```
[5/9] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /workspaces/project/build/zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/workspaces/project/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/rtc/libdrivers__rtc.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/watchdog/libdrivers__watchdog.a modules/hal_ambiq/mcu/apollo3/lib..__modules__hal__ambiq__mcu__apollo3.a  modules/hal_ambiq/components/lib..__modules__hal__ambiq__components.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/workspaces/project/build/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF -L"/opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp" -lc -lgcc && cd /workspaces/project/build/zephyr && /usr/bin/cmake -E true
/opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `main':
/workspaces/project/app/src/main.c:6: undefined reference to `am_hal_uart_initialize'
/opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /workspaces/project/app/src/main.c:7: undefined reference to `am_hal_uart_power_control'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version: 4.0.0

**Additional context**

Follow the Apollo 3 MCU module CMakeLists (modules/hal/ambiq/mcu/apollo3/CMakeLists.txt). As one can see, the `hal/am_hal_uart.c` is missing and there is configuration to add it into compilation:
```
# Ambiq HAL
#
# Copyright (c) 2024 Ambiq Micro Inc. <www.ambiq.com>
#
# SPDX-License-Identifier: Apache-2.0

# Sources and headers necessary for every build.
# These contain definitions and implementation used mostly for
# initializing the SoC, and therefore are always required.
zephyr_library()
zephyr_library_sources(hal/am_hal_global.c)
zephyr_library_sources(hal/am_hal_pwrctrl.c)
zephyr_library_sources(hal/am_hal_queue.c)
zephyr_library_sources(hal/am_hal_cachectrl.c)
zephyr_library_sources(hal/am_hal_interrupt.c)
zephyr_library_sources(hal/am_hal_flash.c)
zephyr_library_sources(hal/am_hal_rtc.c)
zephyr_library_sources(hal/am_hal_mcuctrl.c)
zephyr_library_sources(hal/am_hal_burst.c)
zephyr_library_sources(hal/am_hal_reset.c)
zephyr_library_sources(hal/am_hal_clkgen.c)
zephyr_library_sources(hal/am_hal_sysctrl.c)
zephyr_library_sources(hal/am_hal_debug.c)

if(CONFIG_AMBIQ_HAL_USE_GPIO)
    zephyr_library_sources(hal/am_hal_gpio.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_STIMER)
    zephyr_library_sources(hal/am_hal_stimer.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_TIMER)
    zephyr_library_sources(hal/am_hal_ctimer.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_WDT)
    zephyr_library_sources(hal/am_hal_wdt.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_I2C OR CONFIG_AMBIQ_HAL_USE_SPIC OR CONFIG_AMBIQ_HAL_USE_SPI)
    zephyr_library_sources(hal/am_hal_iom.c)
    zephyr_library_sources(hal/am_hal_cmdq.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_SPID)
    zephyr_library_sources(hal/am_hal_ios.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_MSPI)
    zephyr_library_sources(hal/am_hal_mspi.c)
    zephyr_library_sources(hal/am_hal_cmdq.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_BLEIF)
    zephyr_library_sources(hal/am_hal_ble.c)
    zephyr_library_sources(hal/am_hal_ble_patch.c)
    zephyr_library_sources(hal/am_hal_ble_patch_b0.c)
endif()

if(CONFIG_AMBIQ_HAL_USE_ADC)
    zephyr_library_sources(hal/am_hal_adc.c)
endif()
```

AlessandroLuo, 2025-01-20:
@paulozulato thanks for the finding, we applied the uart_pl011 driver on ambiq MCUs, so ambiq uart hal APIs have not acctually been called in zephyr driver and not be included.
I updated the cmakelists of ambiq hal module, the manifest on zephyr main repo will be update soon as well.

paulozulato, 2025-01-20:
Thank you very much, @AlessandroLuo

kartben, 2025-01-21:
> I updated the cmakelists of ambiq hal module, the manifest on zephyr main repo will be update soon as well.

@AlessandroLuo maybe add links here in this issue to help people stumbling upon this issue to find relevant in-flight PRs.

AlessandroLuo, 2025-01-21:
> > I updated the cmakelists of ambiq hal module, the manifest on zephyr main repo will be update soon as well.
> 
> [@AlessandroLuo](https://github.com/AlessandroLuo) maybe add links here in this issue to help people stumbling upon this issue to find relevant in-flight PRs.

https://github.com/zephyrproject-rtos/zephyr/pull/83720

closed 2025-04-06 by github-actions[bot]