zephyrproject-rtos/zephyr #84164: samples/bluetooth/peripheral_sc_only ios nRF Connect error

zhang-wenchao, 2025-01-17:
**Notes**

Bluetooth connection error.

**Describe the bug**

iOS devices cannot connect to this example using nRf Connect

**To Reproduce**

```
west build -p always -b esp32c3_luatos_core/esp32c3/usb && west flash && west espressif monitor
```

**Expected behavior**

Normal connection

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v4.0.0-3426-g7c50c0231cc7 ***
[00:00:00.024,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.102,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.102,000] <inf> bt_hci_core: Identity: 34:CD:B0:53:A2:18 (public)
[00:00:00.102,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.102,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Bluetooth initialized
Advertising successfully started
Connected 67:FD:F1:19:54:6D (random)
Passkey for 67:FD:F1:19:54:6D (random): 720163
[00:00:21.128,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:36.724,000] <wrn> bt_smp: Unexpected SMP code 0x08
[00:00:36.725,000] <wrn> bt_smp: The in-progress pairing has been deleted!
Pairing Failed (9). Disconnecting.
Security failed: 67:FD:F1:19:54:6D (random) level 1 err (9)
Security changed: 67:FD:F1:19:54:6D (random) level 2
[00:00:36.726,000] <wrn> bt_smp: Unexpected SMP code 0x09
Disconnected from 67:FD:F1:19:54:6D (random), reason 0x05 
```

**Environment (please complete the following information):**

 - OS:Linux
 - Toolchain Zephyr SDK
 - v4.0.0-3426-g7c50c0231cc7

**Additional context**

zhang-wenchao, 2025-01-18:
I have both esp32c3 and nrf52840, so I did a comparison test and the problem is limited to esp32c3. I am not a bluetooth expert, and this problem needs further analysis by Espressif or the maintainers of the bluetooth subsystem.

zhang-wenchao, 2025-01-18:
@sylvioalves  I'm very sorry to bother you, can you help me look into this issue? I will provide any assistance if you need.

zhang-wenchao, 2025-01-18:
In addition, Android is normal.

zhang-wenchao, 2025-01-18:
iOS use nRF Connect error.
Linux use bluez bluetoothctl error.

zhang-wenchao, 2025-01-18:
I guess the two errors I encountered may be related

I forced sc connection at first, and only Android was available. After removing it, iOS was available but in low security state, but nrf52840 had no problem, so you can compare the test.

zhang-wenchao, 2025-01-18:
Hope this issue can be given higher priority.

sylvioalves, 2025-01-20:
@zhang-wenchao, would you retry enabling `CONFIG_BT_SEND_ECC_EMULATION=y`?

sylvioalves, 2025-01-20:
Answered in #84182.

closed 2025-01-20 by sylvioalves

zhang-wenchao, 2025-01-21:
I'm still having issues with my iOS device. to keep it consistent with you, I also changed to esp32c3-devkitm

```
*** Booting Zephyr OS build v4.0.0-3684-gb361c9589ff0 ***
[00:00:00.036,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.102,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.102,000] <inf> bt_hci_core: Identity: 7C:DF:A1:A4:9C:00 (public)
[00:00:00.102,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.102,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Bluetooth initialized
Advertising successfully started
Connected 4B:51:0A:C1:72:54 (random)
Passkey for 4B:51:0A:C1:72:54 (random): 723788
[00:00:22.496,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:37.967,000] <wrn> bt_smp: Unexpected SMP code 0x08
[00:00:37.967,000] <wrn> bt_smp: The in-progress pairing has been deleted!
Pairing Failed (9). Disconnecting.
Security failed: 4B:51:0A:C1:72:54 (random) level 1 err (9)
Security changed: 4B:51:0A:C1:72:54 (random) level 2
[00:00:37.969,000] <wrn> bt_smp: Unexpected SMP code 0x09
Disconnected from 4B:51:0A:C1:72:54 (random), reason 0x05
```

Sorry, I don't know how to log on iOS.