zephyrproject-rtos/zephyr #84174: HAL module request: hal_bouffalolab

nandojve, 2025-01-17:
## Origin

BouffaloSDK is the IOT and MCU software development kit provided by the Bouffalo Lab Team, supports all the series of Bouffalo chips. Also it is the combination of `bl_mcu_sdk` and `bl_iot_sdk`. The original content came from https://github.com/bouffalolab/bouffalo_sdk/tree/release_v1.4.5
and the content to initialize the repo is here https://github.com/nandojve/hal_bouffalolab/tree/sdk-v1.4.5

## Purpose

Reuse an existing hardware abstraction layer from BouffaloSDK to enable BL SoCs series in the Bouffalolab family. Zephyr shim drivers for all the usual device driver APIs (UART, I2C, SPI, etc.) will be built on top of headers and bare metal drivers and components provided by the HAL.

## Mode of integration

The module should be integrated as an external HAL module. The proposed name is `hal_bouffalolab`.

## Maintainership

@nandojve and @VynDragon

## Pull Request

#84173

## Description

The [BouffaloSDK](https://github.com/bouffalolab/bouffalo_sdk/tree/release_v1.4.5) contains the necessary software and tools to develop firmware for the Bouffalolab BLxxx series. This includes register files, peripheral drivers, system startup files, documentation, various utilities, IDE support files, and scripts.

The Bouffalolab HAL is a pared-down version of the BouffaloSDK that contains components needed for Zephyr; namely, register files and bare metal peripheral drivers.

## Dependencies

node

## Revision

Initial version: sdk-v1.4.5 branch

## License

* ``apache-2.0``
* additional:
  * ``bsd-new``
  * ``bsd-x11``

All the files were scanned using `license_check.yml,v4`

nashif, 2025-02-05:
@nandojve will you be able to join the TSC and talk to this request (today)?

nashif, 2025-02-05:
@nandojve  the linked HAL has more than what a HAL should have and includes library code (mbedtls?) and other things. HALs should be focused on hardware support. Is this going to be reduce to what is needed in Zephyr?

VynDragon, 2025-02-05:
Hello @nashif , I will be joining the TSC meeting today about this.

There is 2 parts: first, the 'main' branch which will be the final form it will take, it is only apache 2 headers.
Secondly, to match early versions of the code that rely on the SDK and maintain attribution to third party entities (original ATL Electronics code), the version with more than that is necessary. It is possible to reduce it and we can start removing bits that are not OK with zephyr.

nandojve, 2025-02-05:
Hi @nashif ,

> [@nandojve](https://github.com/nandojve) the linked HAL has more than what a HAL should have and includes library code (mbedtls?) and other things. HALs should be focused on hardware support. Is this going to be reduce to what is needed in Zephyr?

At moment the branch is a copy of https://github.com/bouffalolab/bouffalo_sdk/tree/release_v1.4.5 to make easy to develop.
Yes, we can reduce the content as needed.

closed 2025-03-12 by nashif