zephyrproject-rtos/zephyr #84182: samples/bluetooth/periphera bluez bluetoothctl connect error.

zhang-wenchao, 2025-01-18:
**Describe the bug**

Bluetooth connection error.

**To Reproduce**

```
west build -p always -b esp32c3_luatos_core/esp32c3/usb && west flash && west espressif monitor
```

**Expected behavior**

Normal connection

**Impact**

Bluetooth is not working

**Logs and console output**

**Zephyr**

```
*** Booting Zephyr OS build v4.0.0-3426-g7c50c0231cc7 ***
[00:00:00.045,000] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.045,000] <inf> fs_nvs: alloc wra: 0, f88
[00:00:00.045,000] <inf> fs_nvs: data wra: 0, 7c
[00:00:00.047,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.443,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.447,000] <inf> bt_hci_core: Identity: 34:CD:B0:53:A2:18 (public)
[00:00:00.447,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.447,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Advertising successfully started
Indicate VND attr 0x3c040734 (UUID 12345678-1234-5678-1234-56789abcdef1)
Updated MTU: TX: 23 RX: 23 bytes
Connected
Updated MTU: TX: 517 RX: 65 bytes
[00:00:50.097,000] <wrn> bt_smp: Received invalid public key
Disconnected, reason 0x05
```

**btmon**

```

Bluetooth monitor ver 5.79
btmon[13906]: = Note: Linux version 6.12.9-arch1-1 (x86_64)                                                                                        0.274397
btmon[13906]: = Note: Bluetooth subsystem version 2.22                                                                                             0.274399
= New Index: 20:2B:20:64:A3:BE (Primary,USB,hci0)                                                                                           [hci0] 0.274400
= Open Index: 20:2B:20:64:A3:BE                                                                                                             [hci0] 0.274401
= Index Info: 20:2B:20:64:A3:BE (MediaTek, Inc.)                                                                                            [hci0] 0.274402
bluetoothctl[12996]: @ MGMT Open: bluetoothctl version 1.23                                                                               {0x0002} 0.274403
bluetoothd[709]: @ MGMT Open: bluetoothd (privileged) version 1.23                                                                        {0x0001} 0.274404
bluetoothd[709]: @ MGMT Command: Pair Device (0x0019) plen 8                                                                      {0x0001} [hci0] 11.710684
LE Address: 51:AC:72:5E:98:51 (Resolvable)
Capability: KeyboardDisplay (0x04)
< HCI Command: LE Remove Device From Accept List (0x08|0x0012) plen 7                                                                   #1 [hci0] 11.710755
Address type: Random (0x01)
Address: 46:D2:9E:CB:7F:26 (Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                                                                             #2 [hci0] 11.812770
LE Remove Device From Accept List (0x08|0x0012) ncmd 1
Status: Success (0x00)
< HCI Command: LE Add Device To Accept List (0x08|0x0011) plen 7                                                                        #3 [hci0] 11.812841
Address type: Random (0x01)
Address: 51:AC:72:5E:98:51 (Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                                                                             #4 [hci0] 11.813820
LE Add Device To Accept List (0x08|0x0011) ncmd 1
Status: Success (0x00)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 13                                                                    #5 [hci0] 11.813894
[... 232 lines cut ...]

```
[bluetooth]# scan on
...
[bluetooth]# [NEW] Device 51:AC:72:5E:98:51 Zephyr Peripheral Sample Long
[bluetooth]# scan off
...
[bluetooth]# Discovery stopped
[bluetooth]# pair 51:AC:72:5E:98:51
Attempting to pair with 51:AC:72:5E:98:51
[Zephyr Peripheral Sample Long]# [CHG] Device 51:AC:72:5E:98:51 Connected: yes
[Zephyr Peripheral Sample Long]# Failed to pair: org.bluez.Error.AuthenticationFailed
[bluetooth]# [CHG] Device 51:AC:72:5E:98:51 Connected: no
...
[bluetooth]# exit

```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - v4.0.0-3426-g7c50c0231cc7

**Additional context**

zhang-wenchao, 2025-01-18:
@sylvioalves

#84164 

I think these two issues have similarities, and it needs to be explained that nrf52840 works properly.

zhang-wenchao, 2025-01-18:
In addition, Android iOS is normal.

sylvioalves, 2025-01-20:
@zhang-wenchao, same here.. would you retry enabling `CONFIG_BT_SEND_ECC_EMULATION=y`?

sylvioalves, 2025-01-20:
@zhang-wenchao, please, fetch latest main and check again. It should be working already. PR #80712 automatically fixed the issue after having `BT_CTLR`. That enabled proper ECDH handling. Same for #84164.

zhang-wenchao, 2025-01-21:
I'm still having issues with my iOS device. to keep it consistent with you, I also changed to esp32c3-devkitm

samples/bluetooth/peripheral_sc_only

```
*** Booting Zephyr OS build v4.0.0-3684-gb361c9589ff0 ***
[00:00:00.036,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.102,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.102,000] <inf> bt_hci_core: Identity: 7C:DF:A1:A4:9C:00 (public)
[00:00:00.102,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.102,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Bluetooth initialized
Advertising successfully started
Connected 4B:51:0A:C1:72:54 (random)
Passkey for 4B:51:0A:C1:72:54 (random): 723788
[00:00:22.496,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:37.967,000] <wrn> bt_smp: Unexpected SMP code 0x08
[00:00:37.967,000] <wrn> bt_smp: The in-progress pairing has been deleted!
Pairing Failed (9). Disconnecting.
Security failed: 4B:51:0A:C1:72:54 (random) level 1 err (9)
Security changed: 4B:51:0A:C1:72:54 (random) level 2
[00:00:37.969,000] <wrn> bt_smp: Unexpected SMP code 0x09
Disconnected from 4B:51:0A:C1:72:54 (random), reason 0x05
```

Sorry, I don't know how to log on iOS.

zhang-wenchao, 2025-01-21:
This example bluez works fine, the connection is successful and the security level is increased.

**samples/bluetooth/periphera**

```
*** Booting Zephyr OS build v4.0.0-3684-gb361c9589ff0 ***
[00:00:00.057,000] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.057,000] <inf> fs_nvs: alloc wra: 0, f88
[00:00:00.057,000] <inf> fs_nvs: data wra: 0, 7c
[00:00:00.058,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.128,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.128,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.130,000] <inf> bt_hci_core: Identity: 7C:DF:A1:A4:9C:00 (public)
[00:00:00.130,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.130,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Advertising successfully started
Indicate VND attr 0x3c040744 (UUID 12345678-1234-5678-1234-56789abcdef1)
Updated MTU: TX: 23 RX: 23 bytes
Connected
Updated MTU: TX: 517 RX: 65 bytes
Passkey for 20:2B:20:64:A3:BE (public): 910656
[00:00:37.030,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:39.417,000] <inf> bas: BAS Notifications enabled
[00:00:41.878,000] <wrn> bt_conn: Timeout discarded. No blocking in syswq.
```

```
[bluetooth]# pair 64:9D:B2:F3:55:EB
Attempting to pair with 64:9D:B2:F3:55:EB
[Zephyr Peripheral Sample Long]# [CHG] Device 64:9D:B2:F3:55:EB Connected: yes
[Zephyr Peripheral Sample Long]# Request passkey
...
Service Changed
...
Vendor specific
[agent] Enter passkey (number in 0-999999): [NEW] Characteristic (Handle 0x0036)
/org/bluez/hci0/dev_64_9D_B2_F3_55_EB/service002b/char0036
12345678-1234-5678-1234-56789abcdef4
Vendor specific
[agent] Enter passkey (number in 0-999999): 910656
...
[Zephyr Peripheral Sample Long]# Pairing successful
[Zephyr Peripheral Sample Long]#
[Zephyr Peripheral Sample Long]# exit

```

zhang-wenchao, 2025-01-21:
Now the problem mainly lies in iOS.

zhang-wenchao, 2025-01-21:
I see you don't have an iOS device, I can help you test it if possible.

jhedberg, 2025-01-21:
@zhang-wenchao @sylvioalves FYI, there are some changes coming to the ECC Kconfig options, however that shouldn't necessitate any changes, unless you've been explicitly setting the `BT_SEND_HCI_ECC_EMULATION` option (which will soon not exist anymore): #84268

sylvioalves, 2025-01-21:
@zhang-wenchao I got myself an iPad but it can't find the device when BLE is scanning (`samples/bluetooth/peripheral_sc_only`). 
`samples/bluetooth/peripheral` works fine when connecting to it (not pairing). 
As far as I get, the only issue right now would be iOS pairing with C3, correct?

jhedberg, 2025-01-21:
> [00:00:46.018,000] <wrn> net_buf: Timeout discarded. No blocking in syswq

Just to rule out that this isn't the cause, could you try with #84282?

> [00:00:57.472,000] <wrn> bt_smp: Unexpected SMP code 0x08

FWIW, this is:
```c
#define BT_SMP_CMD_IDENT_INFO			0x08
```

zhang-wenchao, 2025-01-22:
I used the changes from this PR #84282

```
*** Booting Zephyr OS build v4.0.0-3684-gb361c9589ff0 ***
[00:00:00.024,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.091,000] <inf> bt_hci_core: Identity: B0:81:84:AE:C7:28 (public)
[00:00:00.091,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.091,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Bluetooth initialized
Advertising successfully started
Connected 6E:FC:DB:88:18:E3 (random)
Passkey for 6E:FC:DB:88:18:E3 (random): 502139
[00:00:31.376,000] <wrn> bt_smp: Unexpected SMP code 0x08
[00:00:31.376,000] <wrn> bt_smp: The in-progress pairing has been deleted!
Pairing Failed (9). Disconnecting.
Security failed: 6E:FC:DB:88:18:E3 (random) level 1 err (9)
Security changed: 6E:FC:DB:88:18:E3 (random) level 2
[00:00:31.378,000] <wrn> bt_smp: Unexpected SMP code 0x09
Disconnected from 6E:FC:DB:88:18:E3 (random), reason 0x05 

```

sylvioalves, 2025-01-24:
@zhang-wenchao, please, enable `CONFIG_BT_PRIVACY=y` and you will get it working.

zhang-wenchao, 2025-01-24:
Confirm this issue is resolved. thanks for your help.

closed 2025-01-24 by zhang-wenchao