zephyrproject-rtos/zephyr #84196: The filter of samples.yaml of DFU sample seems not to work as expected

nandojve, 2025-01-19:
**Describe the bug**
The dfu sample filter seems not to work. It should select only boards when slot0_partition and slot1_partition are define.

```
filter:
    dt_label_with_parent_compat_enabled("slot0_partition", "fixed-partitions")
and dt_label_with_parent_compat_enabled("slot1_partition", "fixed-partitions") and dt_chosen_enabled("zephyr,flash-controller")
and CONFIG_FLASH_HAS_DRIVER_ENABLED

```
This means that boards that define a partition, like storage, should never been selected because the final devicetree is defined without those partitions. See below fragment from same54_xpro:

```
nvmctrl: nvmctrl@41004000 {
	compatible = "atmel,sam0-nvmctrl";
	reg = < 0x41004000 0x22 >;
	interrupts = < 0x1d 0x0 >, < 0x1e 0x0 >;
	#address-cells = < 0x1 >;
	#size-cells = < 0x1 >;
	lock-regions = < 0x20 >;
	flash0: flash@0 {
		compatible = "soc-nv-flash";
		write-block-size = < 0x8 >;
		reg = < 0x0 0x100000 >;
		partitions {
			compatible = "fixed-partitions";
			#address-cells = < 0x1 >;
			#size-cells = < 0x1 >;
			storage_partition: partition@fc000 {
				label = "storage";
				reg = < 0xfc000 0x4000 >;
			};
		};
	};
};
```

The issue was found on https://github.com/zephyrproject-rtos/zephyr/actions/runs/12832618368/job/35789498048?pr=84118

**To Reproduce**
Steps to reproduce the behavior:

`west twister -p same54_xpro/same54p20a -s sample.usb.dfu.permanent.download --no-detailed-test-id`

**Expected behavior**
Twister filter can filter the conditions. This could avoid add unrelated boards to exclusion list and keep CI sane. Otherwise changes on these platforms can never be catch.

**Impact**
CI false positives and integration delays

**Logs and console output**
Details can be found at https://github.com/zephyrproject-rtos/zephyr/actions/runs/12832618368/job/35789498048?pr=84118

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.17
 - Commit SHA: mainline

nashif, 2025-01-24:
so, the issue started with commit 1440f5c19cd2de4b1765816838bbbd7ea77099e8

57300, 2025-01-25:
> so, the issue started with commit [1440f5c](https://github.com/zephyrproject-rtos/zephyr/commit/1440f5c19cd2de4b1765816838bbbd7ea77099e8)

If a quick fix is needed, then I'm okay with this patch being reverted for now.
Another option (but not necessarily acceptable) is to change this `select` to `imply`:
https://github.com/zephyrproject-rtos/zephyr/blob/6894fce411b771cbfe9172cabdcd8145d169cb6f/modules/Kconfig.mcuboot#L15

---

For a general solution, I would propose changing how Twister evaluates filter expressions.

In the topmost example, the filter has both DT and Kconfig parts, so Twister must run `package_helper.cmake` with the `kconfig` module. This fails because of a Kconfig warning, so we can't evaluate the whole formula in one go.

However, notice that if we evaluate the DT parts by hand, then we get this for `same54_xpro/same54p20a`:
```yaml
filter: false # dt_label_with_parent_compat_enabled("slot0_partition", "fixed-partitions")
  and false # dt_label_with_parent_compat_enabled("slot1_partition", "fixed-partitions")
  and true # dt_chosen_enabled("zephyr,flash-controller")
  and CONFIG_FLASH_HAS_DRIVER_ENABLED
```
This simplifies to `false` and cancels out the Kconfig part.

Remember that the `dts` module runs before `kconfig`. Therefore, even if CMake exits before the Kconfig step, then as long as it completes the DT step, we can try this kind of partial evaluation. This may also allow other `sample.yaml` files to be cleaned up, assuming that `platform_exclude` has been used to work around similar issues before.

In short, my rough proposal would be:
1. Run package helper, but ignore the exit code
2. Load `edt.pickle` and evaluate the DT parts
3. Simplify the expression - if it turns into a simple boolean, then we're done
4. Load `.config` and evaluate the Kconfig parts

@nashif, do you have any objections?

nashif, 2025-01-27:
Let's start with reverting this change to avoid additional breakages and folks trying to fix it in the wrong way....

re twister, this will involve some major changes and IMO such filters are excessive and overkill for something like a sample. The notion that each sample should build with each platform we have in the tree using filters needs reevaluation and more thought and discussion

PR #84607 (2025-01-27): Revert "kconfig: Don't USE_DT_CODE_PARTITION if there isn't one"
This reverts commit 1440f5c19cd2de4b1765816838bbbd7ea77099e8.

Fixes #84196
commit e0a9bd73 (2025-01-27): Revert "kconfig: Don't USE_DT_CODE_PARTITION if there isn't one"

This reverts commit 1440f5c19cd2de4b1765816838bbbd7ea77099e8.

Fixes #84196
closed 2025-01-27 by kartben