zephyrproject-rtos/zephyr #84213: tests: frdm_mcxn236: all application fails to run, console no output

hakehuang, 2025-01-20:
**Describe the bug**

frdm_mcxn236 platform, all application fail to run,  

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - frdm_mcxn236
 - What have you tried to diagnose or workaround this issue?
 - bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
 - yes

```
d4d180c216d946b520cd8de3b69cc4ce31882345 is the first bad commit
commit d4d180c216d946b520cd8de3b69cc4ce31882345
Author: Tomas Galbicka <tomas.galbicka@nxp.com>
Date:   Tue Nov 26 08:45:22 2024 +0100

    dts: drivers: Add DTS MBOX entry for NXP MCXN947

    This commit adds MBOX device tree entry for MCXN947.
    Adds support for MCXN in NXP ipm and mbox drivers.

    Signed-off-by: Tomas Galbicka <tomas.galbicka@nxp.com>

 drivers/ipm/ipm_mcux.c              |  9 ++++++---
 drivers/mbox/mbox_nxp_mailbox.c     |  7 +++++--
 dts/arm/nxp/nxp_mcxn94x_common.dtsi |  9 +++++++++
 soc/nxp/mcx/mcxn/Kconfig            | 11 ++++++++---
 soc/nxp/mcx/mcxn/soc.c              | 21 +++++++++++++++++++++
 5 files changed, 49 insertions(+), 8 deletions(-)

```

**To Reproduce**

```
west update
scripts/twister --device-testing --hardware-map /home/ubuntu/nxp/frdm_mcxn236/map.yaml -T samples/hello_world/
```

**Expected behavior**

application can run

**Impact**

frdm_mcxn236 support

**Logs and console output**

hen this happens no console output.

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.0.0-3615-geabef50991c

DerekSnell, 2025-01-20:
@TomasGalbickaNXP , FYI

TomasGalbickaNXP, 2025-01-21:
Fixed provided in PR: https://github.com/zephyrproject-rtos/zephyr/pull/84281

closed 2025-01-21 by kartben