zephyrproject-rtos/zephyr #84216: Bluetooth Mesh is using the system workqueue for ECC operations

jhedberg, 2025-01-20:
**Describe the bug**

The Zephyr Bluetooth host offers what's called the "long work work queue", that's intended to be used for any long-running operations, like ECDH calculations. There are also dedicated ECDH APIs available which should be used when they satisfy the requirements. The system work queue OTOH is currently used by the host stack itself for critical functionality, and should in most cases be used pretty much like an ISR, i.e. do not block and do not do anything for a long duration.

It seems like our Bluetooth Mesh stack is violating both of the above principles.

**To Reproduce**

Example code path, starting in `mesh/provisionee.c`:

`prov_public_key()` -> `k_work_submit(&dh_gen_work)` -> `prov_dh_key_gen_handler()`

**Expected behavior**

The Mesh stack should use the long wq to perform ECDH operations.

**Impact**

Unstable system in the best case, broken system in the worst case, e.g. if ECDH hasn't been optimized (say, using HW) and the operation run for multiple seconds.

jhedberg, 2025-01-20:
@alwa-nordic FYI

PavelVPV, 2025-01-21:
This sounds reasonable, but before mesh starts using the bt long wq, in my opinion, the bt long wq API needs to be stabilized meaning that it should be public with the documentation what can run on it. Using some internal API creates a risk for mesh being broken at some point because there that internal API doesn't guarantee anything. One my attempt to use the bt long wq was denied once already, so IMHO, it should be public or mesh can't use it.

jhedberg, 2025-01-21:
> Using some internal API creates a risk for mesh being broken at some point because there that internal API doesn't guarantee anything. 

I'd like to understand the above reasoning a bit more. The Zephyr Bluetooth Host and Mesh live in the same git tree, so it should be perfectly fine for Mesh to use internal APIs because of this (e.g. we should be able to catch most breakage using upstream CI). That said, I'm actually not convinced that the long wq is necessarily even the right abstraction level for Mesh to use - couldn't it use the higher level ECDH APIs that the host provides (which underneath utilize the long wq)? If that's not possible today due to some Mesh requirement not being fulfilled, we should at least look at the feasibility of fixing that in the host ECDH APIs.

Slightly related, please note #84268 which I just created, which simplifies the way ECC is available in the host (no HCI emulation layer anymore).

PavelVPV, 2025-01-21:
I have at least 2 concerns:

1. It is not clear what bt long wq is meant to be used for. What is written in the issue is not written anywhere in the code. There is no documentation, no api description, nothing for the bt long wq. What if mesh work will block something more important that runs on that workq? Mesh could have it's own wq with preemptible priority which could interleave with work that runs on the bt long wq. I don't mean mesh needs it, this is just an immediate example that I can come up with.

2. And this also answers this:

> The Zephyr Bluetooth Host and Mesh live in the same git tree, so it should be perfectly fine for Mesh to use internal APIs because of this

https://github.com/zephyrproject-rtos/zephyr/issues/77241 specifically shows that this doesn't work. Even stable API can brake the mesh. From what I see, the recent host changes didn't concider mesh behavior at all. As long as there is no clear contract for the bt long wq API, it can't be used other than host.

jhedberg, 2025-01-21:
@PavelVPV I'm not opposed to making something more stable and public, but I don't think the long wq is what we should be looking at. It's simply a utility to allow use cases needing a preemptible context without requiring a new thread stack for each use case. If such an API was offered, it shouldn't be a Bluetooth API, since there's nothing Bluetooth specific about it. A separate ECDH API on top of the long wq could make sense as a public Bluetooth API, however. We already have `include/zephyr/bluetooth/crypto.h` which seems like an appropriate place for it.

PavelVPV, 2025-01-21:
> [@PavelVPV](https://github.com/PavelVPV) I'm not opposed to making something more stable and public, but I don't think the long wq is what we should be looking at.

I am confused as I though the issue description means using bt long wq. If not, then sure, we just need another preemptible thread to defer crypto operations.

> If such an API was offered, it shouldn't be a Bluetooth API, since there's nothing Bluetooth specific about it. 

Should perhaps Zephyr offer such workq? Similar to sysworkq but with preemptible priority.

jhedberg, 2025-01-21:
> I am confused as I though the issue description means using bt long wq. If not, then sure, we just need another preemptible thread to defer crypto operations.

The Host's ECC APIs implicitly use the long wq. I still think that's what makes sense for Mesh to use, even though it wouldn't use it directly.

> Should perhaps Zephyr offer such workq? Similar to sysworkq but with preemptible priority.

The system wq can be configured to have a preemptible priority, so I'm not sure such a proposal would be accepted. That said, this configuration isn't possible with Zephyr's Bluetooth stack at the moment, since it depends on cooperative properties of the sys wq, and even has a build assert to make sure it has a cooperative priority.

alxelax, 2025-01-21:
I agree with @PavelVPV. If anything is suggested using as common, it should be public and well documented. Right now, bt long wq even doesn't have public KConfig option to tune memory in application. It seems like an internal technological hook in Host not ready for using anywhere else apart Host.

Thalley, 2025-02-03:
I guess it also depends on whether we consider Mesh as being part of the BT host stack, or a user of it. 

If the former, using internal APIs should be OK, if the latter, a public API is a must. 
Today Mesh is mostly using the public API, but does also use some internal header files, so there's no clear split yet (same situation as LE Audio). 

If we want make it a requirement that Mesh should only use the public BT API, there's more work to be done besides the BT long WQ. Personally I'd prefer if Mesh (as well as LE Audio) could entirely depend on the public API.

tzy0002088, 2025-05-06:
ECC operations are very time-consuming. Executing them in the system workqueue can occupy the CPU for a long time, affecting other business threads in the system. I believe such operations should be handled in a preemptive thread instead. I don’t understand why they are being executed in the system workqueue.

tzy0002088, 2025-05-19:
Can ECC operations support hardware ECC? it seems that there is no ability to adapt to hardware ECC. @jhedberg @alwa-nordic

tzy0002088, 2025-05-19:
> > Can ECC operations support hardware ECC? it seems that there is no ability to adapt to hardware ECC. [@jhedberg](https://github.com/jhedberg) [@alwa-nordic](https://github.com/alwa-nordic)
> 
> I do not think Bluetooth Mesh related task is relevant place for such questions. Mesh uses crypto library API. Crypto library implementation hidden beyond API.

ok，new issue is hear : https://github.com/zephyrproject-rtos/zephyr/issues/90146