zephyrproject-rtos/zephyr #84230: Assertion always fails in bt_cts_init()

bringert, 2025-01-20:
**Describe the bug**

The assertion conditions in [bt_cts_init()](https://github.com/zephyrproject-rtos/zephyr/blob/6f6f8bc931832625ba185fe902135c76f20eccd9/subsys/bluetooth/services/cts.c#L173) in  are inverted. This means that any time this function is called with `CONFIG_ASSERT=y` set, you'll get a kernel panic.

**To Reproduce**

1. Get any code that calls `bt_cts_init()`, e.g. [samples/bluetooth/peripheral](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/peripheral)
2. Set `CONFIG_ASSERT=y` in `prj.conf`
3. Build
4. Flash it to a device
5. Kernel panic

**Expected behavior**

No kernel panic

**Impact**

This means that I have to turn off assertions if I want to use `bt_cts_init()`

**Logs and console output**

```
ASSERTION FAIL [cb == ((void *)0)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/services/cts.c:168
        Current Time service need valid `struct bt_cts_cb` callback
[00:00:03.081,146] <err> os: r0/a1:  0x00000004  r1/a2:  0x000000a8  r2/a3:  0x00000014
[00:00:03.081,176] <err> os: r3/a4:  0x00000004 r12/ip:  0x00000004 r14/lr:  0x0001ee53
[00:00:03.081,207] <err> os:  xpsr:  0x21000000
[00:00:03.081,207] <err> os: Faulting instruction address (r15/pc): 0x00042a0c
[00:00:03.081,237] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:03.081,298] <err> os: Current thread: 0x200047c0 (main)
[00:00:06.341,094] <err> os: Halting system
```

**Environment (please complete the following information):**

 - OS: OS X 15.2
 - Toolchain: NCS v2.9.0
 - Commit SHA: 6f6f8bc931832625ba185fe902135c76f20eccd9 (but the offending code is also in main)

**Additional context**

Current code:

```
int bt_cts_init(const struct bt_cts_cb *cb)
{
	__ASSERT(cb == NULL, "Current Time service need valid `struct bt_cts_cb` callback");
	__ASSERT(cb->fill_current_cts_time == NULL,
		 "`fill_current_cts_time` callback api is required for functioning of CTS");
	if (!cb || !cb->fill_current_cts_time) {
		return -EINVAL;
	}
	cts_cb = cb;
	return 0;
}
```

Fixed code:

```
int bt_cts_init(const struct bt_cts_cb *cb)
{
	__ASSERT(cb != NULL, "Current Time service need valid `struct bt_cts_cb` callback");
	__ASSERT(cb->fill_current_cts_time != NULL,
		 "`fill_current_cts_time` callback api is required for functioning of CTS");
	if (!cb || !cb->fill_current_cts_time) {
		return -EINVAL;
	}
	cts_cb = cb;
	return 0;
}
```

alwa-nordic, 2025-01-20:
You are right. Will you post a PR?

bringert, 2025-01-20:
I can, but it would take a while as I haven't contributed to Zephyr before. I'd need to go through the contribution guidelines and setup steps.

alwa-nordic, 2025-01-21:
Up to you, if you want to. I'm sure someone will pick this up soon otherwise.

smaerup, 2025-01-28:
Got hit by the same assert, so I created https://github.com/zephyrproject-rtos/zephyr/pull/84802

closed 2025-01-29 by kartben