zephyrproject-rtos/zephyr #84233: Deprecated pipe test fails for TFM non-secure board targets becuase of using deprecated functions

nordicjm, 2025-01-20:
**Describe the bug**
It seems that these tests fail because of deprecated function usage warnings which are turned into errors - see https://github.com/zephyrproject-rtos/zephyr/actions/runs/12864142908/job/35862833509?pr=82907
Haven't looked into issue so assigning twister and TF-M maintainers to it

**Expected behavior**
Tests to pass

**Impact**
Showstopper holding up unrelated PRs

**Environment (please complete the following information):**

 - Commit SHA or Version used: e658bc1b2baa4191d836a04f19a9cded819e1c1f

tomi-font, 2025-01-20:
Judging from the logs, this has nothing to do with TF-M.

Mattemagikern, 2025-01-20:
Looks like my patch is the culprit (#83283). 
And it also looks like: 
`    extra_args: CMAKE_C_FLAGS="-D__deprecated='' -D__DEPRECATED_MACRO=''"
`
in `testcase.yaml` isn't applied correctly.

nordicjm, 2025-01-23:
@Mattemagikern what is the status of this?

Mattemagikern, 2025-01-24:
I just tested now, after a west update and using the `docker.io/zephyrprojectrtos/zephyr-build:main` as a build container. 

`west build -p -b nrf5340dk/nrf5340/cpuapp/ns tests/kernel/pipe/deprecated/pipe_api -T kernel.deprecated.pipe.api` 
And it doesn't generate the same error locally for me.

nordicjm, 2025-01-24:
Use https://github.com/zephyrproject-rtos/zephyr/pull/82907 it definitely still fails

nordicjm, 2025-01-24:
Or build for any non-secure board e.g. `cmake -GNinja -DBOARD=mps3/corstone300/an547/ns ..`

Mattemagikern, 2025-01-24:
I see that you resolved it on your own 👍 
I'll close my pull-request since it isn't needed anymore.

PR #82907 (2025-01-28): boards: nordic: nrf5340dk/nrf9160dk: Use sysbuild for ns twister
Enables forcing use of sysbuild to build nrf5340dk/nrf9160dk non-secure board targets when using twister

Also fixes #84233
closed 2025-01-28 by kartben