zephyrproject-rtos/zephyr #84241: kernel: common: test kernel.common.toolchain fails on native_sim.

gbarkadiusz, 2025-01-20:
**Describe the bug**
Test `kernel.common.toolchain` fails on native_sim. ` Could not find CMAKE_C_COMPILER using the following names: clang`

**To Reproduce**
`twister -p native_sim -s kernel.common.toolchain -ll debug`

**Logs and console output**
```
- Loading Zephyr default modules (Zephyr base).
-- Application: <path_to>/tests/kernel/common
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: <path_to>.cache/zephyr
-- Zephyr version: 4.0.99 
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: native_sim, qualifiers: native
-- Found host-tools: zephyr 0.17.0 (<path_to>/toolchains/zephyr-sdk-0.17.0)
-- Found toolchain: llvm (clang/ld)
CMake Error at <path_to>/cmake/compiler/clang/generic.cmake:7 (find_program):
  Could not find CMAKE_C_COMPILER using the following names: clang
Call Stack (most recent call first):
  <path_to>/cmake/modules/FindHostTools.cmake:114 (include)
   <path_to>/cmake/modules/dts.cmake:9 (find_package)
   <path_to>/cmake/modules/zephyr_default.cmake:133 (include)
   <path_to>/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
   <path_to>/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA or Version used: 8b987d9e0293b1616ae738cdbee0bac33b71c271

**Additional context**
Bisect pointed.
```
08ed0f3282f8d29e950b8e0f5e91e840fc3025fd is the first bad commit
commit 08ed0f3282f8d29e950b8e0f5e91e840fc3025fd
Author: Anas Nashif <anas.nashif@intel.com>
Date:   Tue Dec 3 17:19:13 2024 -0500

    tests: kernel: common: use integration_toolchains
    
    Use new keyword integration_toolchains to build test using additional
    toolchains.
    
    Signed-off-by: Anas Nashif <anas.nashif@intel.com>

 tests/kernel/common/testcase.yaml | 10 +++++++++-
 1 file changed, 9 insertions(+), 1 deletion(-)
```

gbarkadiusz, 2025-01-20:
@nashif FYI.

nashif, 2025-01-20:
that is not in upstream CI, is it?
you will need to define the path to the llvm toolchain in the workflow or environment where this is being run.

gbarkadiusz, 2025-01-20:
Solved by adding path to llvm.

closed 2025-01-20 by gbarkadiusz