zephyrproject-rtos/zephyr #84244: Ambiq dts nodes are missing a compatible field on zephyr 4.0.0

fbenamrouche, 2025-01-20:
**Describe the bug**
Most of the node for the apollo3 ambiq board are missing a compatible field (i2cX, spiX etc)
This is seen in zephyr 4.0.0

**Expected behavior**
We should not have to explicitly add a compatible in an overlay

**Impact**
We must specify the correct compatible field in the overlays.

AlessandroLuo, 2025-01-21:
@fbenamrouche I guess you mean compatible fields of i2c and spi are not defined in  ambiq_apollo3_blue.dtsi? they are defined in the board level for example the apollo3_evb.dts due to some historical reason, I'll move them to dtsi.

AlessandroLuo, 2025-01-22:
https://github.com/zephyrproject-rtos/zephyr/pull/84276 this pr will fix this issue

closed 2025-01-24 by AlessandroLuo