zephyrproject-rtos/zephyr #84251: flexible-array-like struct used in the middle of another struct whose array field seems to be unused (static analyzer report)

Snape3058, 2025-01-20:
(Report of a static analyzer, may be incorrect)

struct `girq_regs` has a trailing array field of length one
https://github.com/zephyrproject-rtos/zephyr/blob/e4164963f22af072de43a6c56cbf0ad9f323ff1b/soc/microchip/mec/mec172x/reg/mec172x_ecia.h#L1129-L1136
which seems to be a flexible array, and used in the middle of other structs and arrays (struct `ecia_named_regs` and struct `ecia_regs below` this).

As mentioned in the comments above its definition, it seems not intended to be a flexible array. And the array field `RSVD1` is not used as an array (according to the grep results).

If it is intended to be a flexible array struct, it is suggested to use the unsized declaration.
If it is actually an unused placeholder or not a flexible array struct, it would be better to declare field `RSVD1` as a normal field (as suggested in https://people.kernel.org/kees/bounded-flexible-arrays-in-c).

report ids: 250106-1641:7-25

kartben, 2025-01-20:
Based on your detailed analysis (thanks!), I think it would make sense to submit a pull request replacing this field by a simple uint32_t

Snape3058, 2025-01-20:
OK. I will have a try whether it is good to do the replacement.

closed 2025-06-18 by nashif