zephyrproject-rtos/zephyr #84264: [Backport v4.0-branch] Failed to backport #83476

zephyrbot, 2025-01-20:
This issue was created automatically because the backport of #83476 to `v4.0-branch` failed.

The backport to `v4.0-branch` failed:
```
[backport-83476-to-v4.0-branch e7ece9f8160] sensor: adxl345: Add get_size_info API
 Author: Luis Ubieda <luisf@croxel.com>
 Date: Tue Dec 31 17:25:10 2024 -0500
 1 file changed, 23 insertions(+)
Auto-merging drivers/sensor/adi/adxl345/adxl345.c
Auto-merging drivers/sensor/adi/adxl345/adxl345.h
[backport-83476-to-v4.0-branch 56e48395f9f] sensor: adxl345: Fix decoder for non-streaming mode
 Author: Luis Ubieda <luisf@croxel.com>
 Date: Tue Dec 31 17:26:12 2024 -0500
 3 files changed, 67 insertions(+), 40 deletions(-)
Auto-merging drivers/sensor/adi/adxl345/adxl345.c
[backport-83476-to-v4.0-branch 7753b2a1f95] sensor: adxl345: Only enable FIFO Stream with Sensor Stream is enabled
 Author: Luis Ubieda <luisf@croxel.com>
 Date: Tue Jan 14 19:26:53 2025 -0500
 1 file changed, 2 insertions(+)
Auto-merging drivers/sensor/adi/adxl345/Kconfig
CONFLICT (content): Merge conflict in drivers/sensor/adi/adxl345/Kconfig
error: could not apply e453a0ca4de... sensor: adxl345: Disable Sensor Streaming by default
hint: After resolving the conflicts, mark them with
hint: "git add/rm <pathspec>", then run
hint: "git cherry-pick --continue".
hint: You can instead skip this commit with "git cherry-pick --skip".
hint: To abort and get back to the state before "git cherry-pick",
hint: run "git cherry-pick --abort".
hint: Disable this message with "git config advice.mergeConflict false"
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v4.0-branch v4.0-branch
# Navigate to the new working tree
cd .worktrees/backport-v4.0-branch
# Create a new branch
git switch --create backport-83476-to-v4.0-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x e453a0ca4de52573312962414e3df3c47f9c42a2~4..e453a0ca4de52573312962414e3df3c47f9c42a2
# Push it to GitHub
git push --set-upstream origin backport-83476-to-v4.0-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v4.0-branch
```
Then, create a pull request where the `base` branch is `v4.0-branch` and the `compare`/`head` branch is `backport-83476-to-v4.0-branch`.

dkalowsk, 2025-04-03:
@ubieda do you want to fix this for the forth coming 4.0.1 release?  AKA should I hold off until this is closed?

ubieda, 2025-04-03:
Apologies - This fell through the cracks. I'll get it done it now @dkalowsk

ubieda, 2025-04-03:
Done!

ubieda, 2025-08-04:
PR has been merged #88136