zephyrproject-rtos/zephyr #84286: Deferred logging causes calling thread to hang when log buffer is full

besteinm, 2025-01-21:
When I use deferred logging on a STM32 Nucleo L452RE-P board and the log buffer fills up, the thread that calls the LOG_* function hangs.

I can reproduce the behaviour with the following simple program:

```
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
#include <zephyr/drivers/gpio.h>

LOG_MODULE_REGISTER(main, CONFIG_LOG_DEFAULT_LEVEL);

// use large message size to fill up the log buffer
#define MSG_SIZE (CONFIG_LOG_BUFFER_SIZE/2)
#define LED0_NODE DT_ALIAS(led0)

int main(void)
{
	static char msg[MSG_SIZE];   // static because large buffer might not fit on stack

	for (int i = 0; i < sizeof(msg); ++i) {
		msg[i] = '.';
	}
	msg[sizeof(msg)-1] = '\0';

	const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);

	if (!gpio_is_ready_dt(&led)) {
		return 0;
	}

	if (gpio_pin_configure_dt(&led, GPIO_OUTPUT_ACTIVE) < 0) {
		return 0;
	}

	while (1) {
		gpio_pin_set_dt(&led, 1);
		LOG_INF("%s", msg);
		k_msleep(500);
		gpio_pin_set_dt(&led, 0);
		k_msleep(500);
	}

	return 0;
}
```

The test program is supposed to continuously log a large message (half the log buffer size) in order to fill up the log buffer.  It sets and clears the LED to indicate that it is still alive.  However when I run the program on the board the LED stays lit on the third iteration, indicating that it is hanging inside `LOG_INF`.

project configuration:
```
CONFIG_GPIO=y

CONFIG_LOG=y
CONFIG_LOG_MODE_DEFERRED=y
CONFIG_LOG_BUFFER_SIZE=128

#CONFIG_LOG_PRINTK=n
```

(you can ignore the `CONFIG_LOG_PRINTK=n` line for now; I disable `CONFIG_LOG_PRINTK` further down to get additional log messages from the `mpsc` module).

Console output:
```
*** Booting Zephyr OS build v4.0.0-3700-g09991d526a58 ***
[00:00:00.000,000] <inf> main: ...............................................................
[00:
```

It happens on both the latest Zephyr revision  (`09991d526a58f8a51ed9c59480b98a4f30c2053d`)  and also on Zephyr 3.6.0.

The problem does not occur when building the same project for the native_sim target:
```
*** Booting Zephyr OS build v4.0.0-3700-g09991d526a58 ***
[00:00:00.000,000] <inf> main: ...............................................................
[00:00:01.020,000] <inf> main: ...............................................................
[00:00:02.040,000] <inf> main: ...............................................................
[00:00:03.060,000] <inf> main: ...............................................................
```

Digging further into the problem suggests that there might be a bug in `mpsc_pbuf.c`, specifically in `mpsc_pbuf_alloc()`.  After enabling `MPSC_PBUF_DEBUG` and setting `CONFIG_LOG_PRINTK=n` I get the following neverending console output for the Nucleo target:

```
*** Booting Zephyr OS build v4.0.0-3700-g09991d526a58 ***
alloc 24 words, wr:0/0, rd:0/0
allocated 0x20000130, wr:0/24, rd:0/0
committed 0x20000130, wr:24/24, rd:0/0
>>claimed 24: 0x20000130, wr:24/24, rd:0/24
[00:00:00.011,000] <inf> main: ...............................................................
<<freed: 0x20000130, wr:24/24, rd:24/24
invalid claim 0: 0x20000190, wr:24/24, rd:24/24
>>claimed 0: 0, wr:24/24, rd:24/24
alloc 24 words, wr:24/24, rd:24/24
allocated 0x20000130, wr:0/24, rd:24/24
committed 0x20000130, wr:24/24, rd:24/24
>>claimed 24: 0x20000130, wr:24/24, rd:0/24
alloc 24 words, wr:24/24, rd:0/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/24
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
no space: Found busy packet 0x20000130 (len:24), wr:0/0, rd:0/0
no space: Found skip packet 8 len, wr:24/24, rd:24/24
```
... and so on

The native_sim build, on the other hand, shows the following nice output:

```
*** Booting Zephyr OS build v4.0.0-3700-g09991d526a58 ***
alloc 24 words, wr:0/0, rd:0/0
allocated 0x8057450, wr:0/24, rd:0/0
committed 0x8057450, wr:24/24, rd:0/0
>>claimed 24: 0x8057450, wr:24/24, rd:0/24
[00:00:00.000,000] <inf> main: ...............................................................
<<freed: 0x8057450, wr:24/24, rd:24/24
invalid claim 0: 0x80574b0, wr:24/24, rd:24/24
>>claimed 0: (nil), wr:24/24, rd:24/24
alloc 24 words, wr:24/24, rd:24/24
allocated 0x8057450, wr:0/24, rd:24/24
committed 0x8057450, wr:24/24, rd:24/24
>>claimed 24: 0x8057450, wr:24/24, rd:0/24
[00:00:01.020,000] <inf> main: ...............................................................
<<freed: 0x8057450, wr:24/24, rd:24/24
alloc 24 words, wr:24/24, rd:24/24
allocated 0x8057450, wr:0/24, rd:24/24
committed 0x8057450, wr:24/24, rd:24/24
>>claimed 24: 0x8057450, wr:24/24, rd:0/24
[00:00:02.040,000] <inf> main: ...............................................................
<<freed: 0x8057450, wr:24/24, rd:24/24
alloc 24 words, wr:24/24, rd:24/24
allocated 0x8057450, wr:0/24, rd:24/24
committed 0x8057450, wr:24/24, rd:24/24
```
... and so on

nordic-krch, 2025-01-21:
Unfortunately, I cannot reproduce it on any board. I've tried `nrf54l15dk`, `nucleo-l552ze_q` and few qemu's.

besteinm, 2025-01-21:
Hi Krzysztof

thanks for your reply.

I forgot to mention the SDK version I am using:  the problem happens with both version `0.16.5` and `0.17.0`.  I am also able to reproduce it on a custom board with STM32L462 MCU  (which is like the L452 with AES hardware encryption).

I have also tried the `qemu_cortex_m3` target (after having removed the LED-related stuff).  There the test program runs fine.

I will attach the `.config` file generated in the build process (file renamed to `config.txt`).

[config.txt](https://github.com/user-attachments/files/18492137/config.txt)

besteinm, 2025-01-21:
I was able to reproduce the problem on the nRF54L15-DK board  (target `nrf54l15dk/nrf54l15/cpuapp`).

Console output (with `MPSC_PBUF_DEBUG` disabled):
```
*** Booting Zephyr OS build v4.0.0-3700-g09991d526a58 ***
[00:00:00.429,828] <inf> main: ...............................................................
[
```

The LED stays lit on the third iteration (same as with the Nucleo board).

besteinm, 2025-01-23:
I have played with the durations that the main thread is sleeping in the main loop and made an interesting observation.

```
...
	while (1) {
		gpio_pin_set_dt(&led, 1);
		LOG_INF("%s", msg);
--->		k_msleep(500);
		gpio_pin_set_dt(&led, 0);
--->		k_msleep(500);
	}
...
```

If I increase the sleep duration in both `k_msleep` calls enough so that the logging thread is able to send the whole message over the UART then the program no longer hangs.  It turns out that if I set the sleep duration to 505 ms (in each of the two calls to `k_msleep`) then the program runs okay.

Here are the outputs of the programs when I compile with different sleep durations, starting at 500 ms and incrementing by 1 ms.

500 ms:
```
*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
[00:00:00.434,652] <inf> main: ...............................................................
[
```
(... hangs)

501 ms:
```
*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
[00:00:00.434,694] <inf> main: ...............................................................
[00:00:01.436,876] <i
```
(... hangs)

502 ms:
```
*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
[00:00:00.434,652] <inf> main: ...............................................................
[00:00:01.438,828] <inf> main: .............
```
(... hangs)

503 ms:
```
*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
[00:00:00.434,655] <inf> main: ...............................................................
[00:00:01.440,812] <inf> main: ....................................
```
(... hangs)

504 ms:
```
*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
[00:00:00.434,657] <inf> main: ...............................................................
[00:00:01.442,796] <inf> main: ...........................................................
```
(... hangs)

505 ms:
```
*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
[00:00:00.434,658] <inf> main: ...............................................................
[00:00:01.444,844] <inf> main: ...............................................................
[00:00:02.455,023] <inf> main: ...............................................................
[00:00:03.465,199] <inf> main: ...............................................................
[00:00:04.475,375] <inf> main: ...............................................................
[00:00:05.485,551] <inf> main: ...............................................................
[00:00:06.495,695] <inf> main: ...............................................................
[00:00:07.505,871] <inf> main: ...............................................................
[00:00:08.516,047] <inf> main: ...............................................................
```
(... keeps running)

besteinm, 2025-01-23:
So it looks like if the log process thread is able to `claim` a message from the buffer and gets preempted while it is sending the message over the UART, then it somehow is not able to `free` the message again (maybe because it does not get scheduled anymore?)

This causes the main task staying stuck in `mpsc_pbuf_alloc` (as seen by the neverending sequence of "no space: Found busy packet" and "no space: Found skip packet" messages).

besteinm, 2025-01-23:
Setting the log process thread priority to 0 (which is also the priority of the main thread) also lets the test program run okay:

```
CONFIG_LOG_PROCESS_THREAD_CUSTOM_PRIORITY=y
CONFIG_LOG_PROCESS_THREAD_PRIORITY=0
```

```
[*** Booting Zephyr OS build v4.0.0-3874-g020cb79d27d0 ***
sleep 500 / main prio 0 / log prio 0
[00:00:00.437,914] <inf> main: ...............................................................
[00:00:01.438,059] <inf> main: ...............................................................
[00:00:02.446,987] <inf> main: ...............................................................
[00:00:03.455,918] <inf> main: ...............................................................
[00:00:04.464,839] <inf> main: ...............................................................
```

nordic-krch, 2025-01-27:
I've managed to reproduce. I will investigate.

closed 2025-01-29 by kartben

besteinm, 2025-01-30:
thanks for fixing!

PR #84595 (2025-01-29): lib: os: mpsc_pbuf: Fix infinite loop during allocation
It was possible that allocation was continuously attempting to find space. It could happen if allocation interrupted consuming a packet (which is marked as busy) and there is not enough space to allocate requested packet but there would be if busy packet was freed. Algorithm in that case was continuously going through the buffer in search for packets that can be dropped to find space for the new packet.

Fixes #84286.