zephyrproject-rtos/zephyr #84291: Better to use bracket in condition check in while loop of "subsys/shell/shell.cshell.c"

wwhheerree, 2025-01-21:
**Describe the bug**
I am using QEMU to run shell_uart.  
In shell "execute" function, there is a while loop to parse each argument from "sh->ctx->cmd_buff" one at a time. 
One of the AND condition is "args_left > 0" which means there is argument needs to be parsed.
[image]

The condition was added by https://github.com/zephyrproject-rtos/zephyr/commit/512de5ecac62de202cf17676d57c7c9d466dc3e2#diff-ada658f52221d7e6c6143eaa3be265c83de76dec8e4df770395ffe9c7c259d31 
It would be better to use "(args_left > 0)" instead of "args_left > 0". 
[image]

**To Reproduce**

**Expected behavior**
It would be better to use "(args_left > 0)" instead of "args_left > 0". 
[image]

**Impact**
No Impact

**Logs and console output**
None

**Environment (please complete the following information):**

 OS: Virtual Box Ubuntu 22.04.4 LTS on Windows 10
 Toolchain Zephyr SDK 0.16.9
 Version: 3.7.0 (commit https://github.com/zephyrproject-rtos/zephyr/commit/36940db938a8f4a1e919496793ed439850a221c2)

kartben, 2025-01-21:
@wwhheerree feel free to open a PR :) Thanks!

kartben, 2025-01-21:
> @wwhheerree feel free to open a PR :) Thanks!

which you have... thanks! #84292