zephyrproject-rtos/zephyr #84317: Notification to Zephyr west project maintainers & collaborators of Zephyr 4.1 feature freeze approaching

kartben, 2025-01-21:
This issue is to notify all Zephyr west projects maintainers and collaborators that as we are approaching the feature freeze (2025/02/07) for Zephyr 4.1, you may want to consider checking if your Zephyr west project would benefit from an update to get the latest updates or fixes that may be in its upstream but are not yet in the version pointed from the zephyr west manifest.

This issue is only for notification purposes and will be created and closed immediately.

kartben, 2025-01-21:
@aescolar
@afontaine-invn
@anangl
@andy-liu-telink
@aronlander-pe
@avisconti
@Casper-Bonde-Bose
@ceolin
@cfriedt
@d3zd3z
@dcpleung
@de-nordic
@dleach02
@eriktamlin
@erwango
@faxe1008
@henrikbrixandersen
@ifyall
@ioannis-karachalios
@jhedberg
@jukkar
@jvasanth1
@keith-packard
@KhiemNguyenT
@kholia
@krish2718
@kristofer-jonsson-arm
@kv2019i
@kwd-doodling
@mah-eiSmart
@Mani-Sadhasivam
@MariuszSkamra
@MaureenHelm
@microbuilder
@mmahadevan108
@najumon1980
@nandojve
@nzmichaelh
@pdgendt
@povergoing
@rlubos

kartben, 2025-01-21:
@sachinthegreen
@sgrrzhf
@soburi
@ssekar15
@StefanHri
@stephanosio
@sylvioalves
@teburd
@vaishnavachath
@wearyzen
@XenuIsWatching
@yperess

closed 2025-01-21 by kartben

jhedberg, 2025-01-21:
> we are approaching the feature freeze (2025/02/07) for Zephyr 4.1,

Based on https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management the Feature Freeze is 2025/02/14.

kartben, 2025-01-21:
> > we are approaching the feature freeze (2025/02/07) for Zephyr 4.1,
> 
> Based on [Wiki: Release Management](https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management) the Feature Freeze is 2025/02/14.

Duh. Of course. Thanks.