zephyrproject-rtos/zephyr #84432: Driver Required for the Display Sitronix ST7305

dkjLC, 2025-01-23:
I am currently working on the project where in I require the Zephyr driver for the Sitronix ST7305 Monochromic display.

Link to Datasheet : https://v4.cecdn.yun300.cn/100001_1909185148/ST_7305_V0_2.pdf

It will be Very helpful for me if you can help me with drivers.

dkjLC, 2025-01-24:
Hii, 
We got started with the program i have attached the files below. can you check for me where i am wrong as i dont see anything on screen.

For testing i have used name as examplesensor

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------

examplesensor.c

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------

#define DT_DRV_COMPAT   zephyr_examplesensor

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(st7305, CONFIG_DISPLAY_LOG_LEVEL);

#include <string.h>
#include <zephyr/device.h>
#include <zephyr/drivers/display.h>
#include <zephyr/init.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/drivers/spi.h>
#include <zephyr/sys/byteorder.h>

#define ST7305_PANEL_WIDTH   DT_INST_PROP(0, width)
#define ST7305_PANEL_HEIGHT  DT_INST_PROP(0, height)

#define ST7305_PIXELS_PER_BYTE  8U
/* Adding 2 for the line number and dummy byte
 * line_buf format for each row.
 * +-------------------+-------------------+----------------+
 * | line num (8 bits) | data (WIDTH bits) | dummy (8 bits) |
 * +-------------------+-------------------+----------------+
 */
#define ST7305_BYTES_PER_LINE  ((ST7305_PANEL_WIDTH / ST7305_PIXELS_PER_BYTE) + 2)  // KEPT AS IT IS

// #if DT_INST_NODE_HAS_PROP(0, disp_miso_en_gpios)
// 	struct gpio_dt_spec miso_gpio;
// #endif

struct ST7305_config {
    struct spi_dt_spec bus;
#if DT_INST_NODE_HAS_PROP(0, disp_miso_en_gpios)
    struct gpio_dt_spec disp_miso_en_gpios;
#endif
};

static int ST7305_blanking_off(const struct device *dev)
{
    const struct ST7305_config *config = dev->config;
    const struct spi_dt_spec *spi = &config->bus;

    uint8_t display_on_command = 0x29; // Display ON command for ST7305

    struct spi_buf tx_buf = {
        .buf = &display_on_command,
        .len = sizeof(display_on_command),
    };

    struct spi_buf_set tx = {
        .buffers = &tx_buf,
        .count = 1,
    };

    int ret = spi_write_dt(spi, &tx);
    if (ret < 0) {
        LOG_ERR("Failed to send Display ON command over SPI: %d", ret);
        return ret;
    }

[... 754 lines cut ...]
# Add subdirectory for custom drivers if they exist
if(EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/drivers)
    add_subdirectory(drivers)
endif()

# Include directories for custom drivers
zephyr_include_directories(drivers/sensor/examplesensor)

# Link application to Zephyr
target_link_libraries(app PRIVATE zephyr)

# Optional: Suppress regeneration messages
set(CMAKE_SUPPRESS_REGENERATION TRUE)

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------

CMakeLists.txt (examplesensor)

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------

add_subdirectory_ifdef(CONFIG_EXAMPLESENSOR examplesensor)

-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Hoping to get help fast as my project is much delayed.

faxe1008, 2025-01-24:
Hi @dkjLC 

Contributions are always welcome, but please open a pull request. If you want to a new driver you do not need to create an issue beforehand. Issues are for RFCs and bugreports primarily.

If you are not sure how to do that (saw that you do not have any repo on your profile) I'd suggest on reading and practicing git. 

Happy hacking :^)