zephyrproject-rtos/zephyr #84438: Memory leak with socket() + close() when socket offloading is enabled

erlingrj, 2025-01-23:
**Describe the bug**

When socket offloading is enabled, repeated `socket()` and `close()` will eventually lead to a ENOMEM. If socket offloading is disabled, the error does not occur. When the heap size is increased, more iterations of open/close is necessary to trigger the errror.

**To Reproduce**

main.c
```c
#include <zephyr/kernel.h>
#include <sys/socket.h>

int fd;
int cnt = 0;
int main(void) {
  while (true) {
    printk("Open socket\n");
    fd = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
    if (fd < 0) {
      printf("Failed to open socket after %d openings errno=%d\n",cnt, errno);
      return -1;
    }
    printk("Close socket\n");
    if (close(fd) < 0) {
      printf("Failed to close socket. errno=%d\n", errno);
      return -1;
    }
    k_usleep(100000);
    cnt++;
  }
  return 0;
}
```
prj.conf
```
CONFIG_ETH_NATIVE_POSIX=n
CONFIG_NET_DRIVERS=y
CONFIG_NETWORKING=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV4=y
CONFIG_NET_SOCKETS=y
CONFIG_POSIX_API=y
CONFIG_NET_SOCKETS_OFFLOAD=y
CONFIG_NET_NATIVE_OFFLOADED_SOCKETS=y
CONFIG_HEAP_MEM_POOL_SIZE=1024
```

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.20.0)
set(BOARD native_sim)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(mem-leak)
target_sources(app PRIVATE src/main.c)
```
**Expected behavior**
Expected behavior is that this program can run indefinitely.

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain: v0.16.8
 - Zephyr: v3.7.0

jeffwelder-ellenbytech, 2025-01-23:
I believe myself as well as others have come across this -- a failed `zsock_connect()` call requiring cleanup via `zsock_close()`.

Please reference Robert's great reply which I believe you're experiencing:
(https://github.com/zephyrproject-rtos/zephyr/issues/41404#issuecomment-999696732)

erlingrj, 2025-01-24:
Thanks for your help. I am not so familiar with backports, does this mean I can continue using v3.7.0 and get this fix? Do I need to do anything with my `west.yml` to get the backport, once it is merged? Currently looks like:

```
manifest:
  self:
    west-commands: scripts/west-commands.yml
    path: .

  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v3.7.0
      import: 
        path-prefix: deps
```

rlubos, 2025-01-24:
@erlingrj You'd need to change the Zephyr revision to `v3.7-branch` (LTS branch), and when the next bugfix release for LTS is out, you  could switch back to tag (`v3.7.2`). I don't know if the next bugfix release is scheduled yet though.

erlingrj, 2025-01-26:
Great, thanks for your assistance!

PR #84478 (2025-01-29): [Backport v3.7-branch] net: nsos_sockets: free allocated socket on close
Backport c22233a1afcb97e55af2929c7d1416da77d96ca9 from #80468.

Fixes #84438.
closed 2025-01-26 by erlingrj